MarlinFirmware/Marlin #2904: Can't compile Marlin

Shtreban, 2016-01-16:
I guess that problem is in serial port initialization. Board's COM is 2. I've tried to set SERIAL_PORT to 2 or any else random number, but it didn't help.
Error message is:
Arduino: 1.6.5 (Windows 7),  "Sanguino, ATmega1284p 16MHz"

In file included from Marlin.h:44:0,
                 from BlinkM.cpp:5:
MarlinSerial.h: In member function 'int MarlinSerial::available()':
MarlinSerial.h:100: error: 'rx_buffer' was not declared in this scope
       return (unsigned int)(RX_BUFFER_SIZE + rx_buffer.head - rx_buffer.tail) % RX_BUFFER_SIZE;
                                              ^
MarlinSerial.h: In member function 'void MarlinSerial::write(uint8_t)':
MarlinSerial.h:46: error: 'UCSR2A' was not declared in this scope
 #define M_UCSRxA SERIAL_REGNAME(UCSR,SERIAL_PORT,A) // defines M_UCSRxA to be UCSRnA where n is the serial port number
                                 ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:46:18: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_UCSRxA SERIAL_REGNAME(UCSR,SERIAL_PORT,A) // defines M_UCSRxA to be UCSRnA where n is the serial port number
                  ^
MarlinSerial.h:105:17: note: in expansion of macro 'M_UCSRxA'
       while (!((M_UCSRxA) & (1 << M_UDREx)))
                 ^
MarlinSerial.h:51: error: 'UDRE2' was not declared in this scope
 #define M_UDREx SERIAL_REGNAME(UDRE,SERIAL_PORT,)    
                                ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:51:17: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_UDREx SERIAL_REGNAME(UDRE,SERIAL_PORT,)    
                 ^
MarlinSerial.h:105:35: note: in expansion of macro 'M_UDREx'
       while (!((M_UCSRxA) & (1 << M_UDREx)))
                                   ^
MarlinSerial.h:52: error: 'UDR2' was not declared in this scope
 #define M_UDRx SERIAL_REGNAME(UDR,SERIAL_PORT,)  
                               ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:52:16: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_UDRx SERIAL_REGNAME(UDR,SERIAL_PORT,)  
                ^
MarlinSerial.h:108:7: note: in expansion of macro 'M_UDRx'
       M_UDRx = c;
       ^
MarlinSerial.h: In member function 'void MarlinSerial::checkRx()':
MarlinSerial.h:46: error: 'UCSR2A' was not declared in this scope
 #define M_UCSRxA SERIAL_REGNAME(UCSR,SERIAL_PORT,A) // defines M_UCSRxA to be UCSRnA where n is the serial port number
                                 ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:46:18: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_UCSRxA SERIAL_REGNAME(UCSR,SERIAL_PORT,A) // defines M_UCSRxA to be UCSRnA where n is the serial port number
                  ^
MarlinSerial.h:114:11: note: in expansion of macro 'M_UCSRxA'
       if((M_UCSRxA & (1<<M_RXCx)) != 0) {
           ^
MarlinSerial.h:55: error: 'RXC2' was not declared in this scope
 #define M_RXCx SERIAL_REGNAME(RXC,SERIAL_PORT,)
                               ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:55:16: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_RXCx SERIAL_REGNAME(RXC,SERIAL_PORT,)
                ^
MarlinSerial.h:114:26: note: in expansion of macro 'M_RXCx'
       if((M_UCSRxA & (1<<M_RXCx)) != 0) {
                          ^
MarlinSerial.h:52: error: 'UDR2' was not declared in this scope
 #define M_UDRx SERIAL_REGNAME(UDR,SERIAL_PORT,)  
                               ^
MarlinSerial.h:41:61: note: in definition of macro 'SERIAL_REGNAME_INTERNAL'
 #define SERIAL_REGNAME_INTERNAL(registerbase,number,suffix) registerbase##number##suffix
                                                             ^
MarlinSerial.h:52:16: note: in expansion of macro 'SERIAL_REGNAME'
 #define M_UDRx SERIAL_REGNAME(UDR,SERIAL_PORT,)  
                ^
MarlinSerial.h:115:29: note: in expansion of macro 'M_UDRx'
         unsigned char c  =  M_UDRx;
                             ^
MarlinSerial.h:116: error: 'rx_buffer' was not declared in this scope
         int i = (unsigned int)(rx_buffer.head + 1) % RX_BUFFER_SIZE;
                                ^
'rx_buffer' was not declared in this scope

I've tried versions 1.1.0RC3, RCBugFix and 1.0.2, but none of them worked. 
My board is Sanguinololu 1.3b

fbarcenas, 2016-02-28:
try arduino 1.0.5

thinkyhead, 2016-03-01:
I don't think you can set `SERIAL_PORT` to `2`. I believe it should be set to `0`.

jbrazio, 2016-03-02:
Well I haven't dig into the code but the Arduino Mega has more than one serial port, don't know if that define controls which serial port is used by Marlin. Nevertheless using 0 as the serial port should compile flawlessly.

closed 2016-03-22 by thinkyhead

Hedobum, 2018-03-27:
I had the same issue on my GT2560 board. Changed the port to 0 (it was always `SERIAL_PORT` 7 in past compiles). Regardless, it seemed to work perfectly and compiled the sketch right away. 
Unfortunately, I'm a dunderhead and blew a fuse on my GT2560 as I was checking some wiring and now I have to wait for fuses from China before I can actually upload the sketch to my board. fml...there are forces working against me.

kpgresk, 2019-10-12:
I cannot compile MARLINSERIAL_H_,  ln:155 expected primary-expression at end of input:       
static void print(char, int = byte);
**156  static void print(unsigned char, int = byte);** 
      static void print(int, int = DEC);

**Being a newbie, and spending days on this, I haven't a clue what to do to get Marlin 2011-2018 MarlinFirmware to compile. I nneed help, please.**
***I  am under the impression that Marlin is an off-the-shelf popular working program for 3D printing..