MarlinFirmware/Marlin #2906: New controller card: MPX.3 arduino/RAMPS mashup single card pins.h

RalphSutherland, 2016-01-18:
I have a chinese single board, a Arduino Mega2560/RAMPS with 4988 stepper drivers mashup called an MPX.3.  The main difference is a fourth power MOSFET pin in addition to the normal three.  I have made a pins_MPX3.h for a ramps compatible and the following power pins and it seems to run fine.

/**
 * MPX-3 with RAMPS v1.4 pin assignments EEB + fan
 *
 *  ie RAMPS_14_EBFE (Extruder, Bed, Fan, Extruder)
 */

#define IS_RAMPS_EEB

#include "pins_RAMPS_13.h"

#undef HEATER_BED_PIN
#define HEATER_BED_PIN    8    // BED

#undef FAN_PIN
#define FAN_PIN           9    // FAN

#undef HEATER_0_PIN
#define HEATER_0_PIN     10    // E0

#undef HEATER_1_PIN
#define HEATER_1_PIN      7    // E1

KiteLab, 2016-01-18:
Do you have a link to the product?

RalphSutherland, 2016-01-18:
http://www.maxmicron.com/index.php?route=product/product&product_id=52  the dual extruder model is the only one I have seen, no idea if their other single E models use the same card.   I'll take  some closeup photos of the board and make a URL for them soon.  It is that single red board in the background of some of the photos.  It came with a version of marlin 1.0.0 already installed.

thinkyhead, 2016-03-11:
I'll put this on my to-do list for `dev` and perhaps we can toss it into `RCBugFix` as well since it's both obscure and in need of wider testing. Just recently we properly assigned most boards as derivatives of RAMPS 1.4 instead of 1.3. This MPX.3 is probably also RAMPS 1.4. (The `SERVO_0_PIN` is the only difference.)

MikeInMinnesota, 2016-04-28:
I also have this MPX.3 board.  It seems to be the same as a HICTOP controller.  I've been running it as a RAMPS 1.4, with basically Ralph's EBFE configuration.  My board came as the controller for a dual head Max Micron printer.  Attached are my Configuration.h and Configuration_adv.h for the board.  My configuration has auto bed leveling, which is an option for the Max Micron.  I'm currently running RC6.

[MPX.3-configuration.zip](https://github.com/MarlinFirmware/Marlin/files/239745/MPX.3-configuration.zip)

thinkyhead, 2016-04-29:
The pins listed in the original post are identical to the `MKS_BASE` pins, actually. So I think we should just make a note in the `boards.h` file that `MKS_BASE` is the same as HICTOP and MPX.3.

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-04 by jbrazio

MikeInMinnesota, 2016-05-08:
Yes, I'm very much looking forward to the 1.1 release.

My comment about the issue was more towards the configuration issue for different printers.  I believe the controller itself should work as a stock 1.4 board, and indeed that is how I have been using it.  I am unaware of the status for providing a tool for adjusting the configuration.h and anfiguration_adv.h parameters.  I understand Thinkyhead was working on this a while back.  Do you have any further information about this effort?   

Regards,
Mike

Note from PRGA.COM: This message has been scanned for SPAM.

Click here to mark email as junk<https://mail.quarantine.prga.com:8444/FrontController?operation=mbeu&f=00001_49_20160504_4209058.eml&chkBayesian=1&pr=1&mt=1&ma=s>.
-----------------------------------------------------------

thinkyhead, 2016-05-09:
@MikeInMinnesota A friendlier configuration tool is in the planning stage, but so far not built. In the meantime I'm continuing to work on a more complete configuration tool at https://github.com/thinkyhead/MarlinConfigurator – which needs code updates to deal with the latest config files, and to make it smarter as well.

jasonb194, 2016-09-26:
I actually got HIC to send me the source files for their Marlin Set up for the MPX.3

Their config has the board set up as a 33. I can send you any of the files if you want to see them.

RalphSutherland, 2016-09-26:
HI, 

Not a problem.  I am fine, and adding aux and servo boards anyway so non-standard now anyway. 

R

Dr Ralph Sutherland
Research School of Astronomy and Astrophysics
Australian National University

Blade893, 2016-12-15:
Hi jasonb194, can you please send me your HIC source files?

Fever2112, 2017-01-18:
@jasonb194 Can you send me the source files?   I have this board but the language is not English... I would like to reprogram the menus or get one that is the right language.

probonopd, 2017-01-21:
@jasonb194 best would be to upload the source files to a GitHub repository.

Fever2112, 2017-01-21:
Is the firmware from HIC any different that the Marlin kit at marlinfw.org?  Does the MarlinFW 1.0.x run on this MPX.3 board?

Fever2112, 2017-01-21:
and yes, please upload the files if/when you are able.
  :)

jasonb194, 2017-01-22:
Below is the firmware that HICTOP sent me for the MPX.3 board.

It's the normal Marlin Firmware, but different settings.
[Marlin_3DP08_3DP11_398.zip](https://github.com/MarlinFirmware/Marlin/files/722293/Marlin_3DP08_3DP11_398.zip)

probonopd, 2017-02-20:
Does anyone know where the serial port is located on the MPX.3? Is there a pinout? Does it provide 3.3V or 5V somewhere that would be sufficient to hook up a ESP8266? I would like to use https://github.com/luc-github/ESP3D with it to turn it into a WLAN-enabled board.

probonopd, 2017-02-22:
> Does anyone know where the serial port is located on the MPX.3? 

Turns out on the AUX-1 header, same as for the RAMPS board.

ronenst1, 2017-02-26:
Dose anyone know how can I connecte the bltouch sensor to the MPX.3 board? I need to use the servo output to raise the bltouch pin.

MikeInMinnesota, 2017-02-26:
You use D11 by the S1 button for the servo connection (3 pin) and the Z- end stop that is almost next to that for the bltouch 2 pin connector.  The Z- stop touches the EXP2 connector that goes to the display.  If we call the EXP2 connector the top right corner, then the vertical pin assignment on the servo is (top to bottom) 5V, GND, D11.  You only use 2 pins of the Z- stop.  The middle pin is GND and the bottom pin is the Z MIN signal.

You also need to rebuild the Marlin firmware.  I’d suggest RC8.

Regards,
Mike

enter801, 2017-02-27:
Also should mention that if you have your firmware set up with RAMPS 1.3 as the board (33), then it will re-assign the servo pin to 7.  Needs to be 11 in the firmware, so setting it as RAMPS 1.4 EFB  instead of 1.3 EFB should do the trick.  Just about to test as my BLTOUCH isn't responding, as I'm sure it's set to pin 7 instead of 11 (but hooked up to 11).

ronenst1, 2017-02-27:
Thank you all. I will try it this week.

ronenst1, 2017-02-27:
Mike thanks again but unfortunately my board printer (Max Micron dual head including Auto-Leveling) in the same as the HICTOP board but without that specific D11 conectors.
I can see that the name of the board is MPX PRO V3.0
Mike do you have any suggestions?

MikeInMinnesota, 2017-02-28:
That’s a good question.  It sounds like Max Micron has a slightly changed version of the MPX 3, maybe via HICTOP.  If nothing on the board is labeled D11 then the controller has been re-worked in their latest version.  I’d suggest contacting Max Micron’s  support at maxmicron@hotmail.com<mailto:maxmicron@hotmail.com>.  I don’t know if he’s still there, but they had a pretty knowledgeable guy named ZhaoMonster that was helpful for some of my previous questions about connecting sensors to various end-stops.

Regards,
Mike

ronenst1, 2017-02-28:
Thanks Mike I will try Zhao

robertch1969, 2017-03-20:
has anyone ended up getting a new marlin version for this board working ?\

i have the board on this machine https://www.aliexpress.com/item/New-3D-Printer-Large-Build-Size-ET-I3-3D-Printer-with-Dual-Extruder-Touch-Screen-and/32763624787.html?spm=2114.01010208.3.1.HrI6ZU&ws_ab_test=searchweb0_0%2Csearchweb201602_5_10065_10068_10501_10000074_10503_10000032_119_10000025_10000029_430_10000028_10060_10000067_10062_10056_10055_10000062_10054_301_10059_10099_10000022_10000013_10103_10102_10000016_10096_10000018_10000019_10000056_10000059_10052_10053_10107_10050_10106_10051_10000053_10000007_10000050_10084_10083_10000047_10080_10082_10081_10110_10111_10112_10113_10114_10115_10000041_10000044_10078_10079_10000038_10073_10000035_10121-10503_10501%2Csearchweb201603_7%2Cafswitch_4%2Csingle_sort_2_default&btsid=937ee225-c7e7-4ea7-8c74-b56fb25cd920 and am trying to install a BLtouch on it have tried to make a few different versions of marlin for it myself but am not sure how to do this so been reading and trying different things closest i have got is new firmware runs but axises are very erratic things they seem to be missing steps i revert to the old version of the firmware and it works fine

i have the source file for the original firmware i think it is 1.0.0 RC2

i want to achieve 2 things get a new more recent version of Marlin for my board and get the Bltouch connected and working

enter801, 2017-03-20:
I have the BLToich working on this board with RC8 firmware. You have to set it to ramps 1.4 as the board in the firmware then plug the three pin cable into that. If you don't have headers on D11 then you should get some JST three pin male and female plugs. You can get those on Amazon, a whole box with all sizes for cheap. You have to set up the servo and a few other things. I can send you my config and config advanced files. The BLTouch is amazing btw. Way way better than the inductive sensor. Do yourself a favor and put a PEI/Ultem sheet on your bed. No need for glass or glue or tape after that. Can print ABS without warping without an enclosure.  The BLTouch is awesome for that because it will get accurate bed leveling done on an uneven bed, which you can get with the PEI sheet being installed poorly (double sided 3m glue) like I did.

robertch1969, 2017-03-21:
yeah enter801 have the PEI sheet i can get the bltouch to move but something is going wrong with the printer movement with the new firmware i am missing something i think pretty new to this so here is the hole source
[双200x280.zip](https://github.com/MarlinFirmware/Marlin/files/856747/200x280.zip)
 file

robertch1969, 2017-03-21:
so update here have played with the config h have y and 7 moving normally and BLtouch responding to basic test commands but x just makes a noise but no movement re flash with old firmware all axis move as expected but cant get BL touch to work with it

enter801, 2017-03-21:
Here are my RC8 configuration files.  I have an e3d hot end set up as well, so you will have to match it for that, and for the bed size, offsets for where you have the BLTouch mounted with respect to the hotend.  Another thing you will have to do is figure out what your z offset is.  My firmware has the homing distance set to 5mm, so you can go down 5 mm, and then whatever past 5mm it requires to get the hotend 100 microns away(sheet of paper height) from the bed is your z offset.  You can change it in the firmware config (I believe mine is set to -1.5 in the firmware) or override it with the z offset gcode command.
[MarlinRC8_BLTouch_MPX3_Board_enter801.zip](https://github.com/MarlinFirmware/Marlin/files/857595/MarlinRC8_BLTouch_MPX3_Board_enter801.zip)

robertch1969, 2017-03-21:
thank you very much for this  will give that a go i dont know what i am doing wrong but my x axis doesn't like it when i configure for BL Touch is the Pic here how you wired the BL touch to your board ?
[image] my board has a slightly different number on it it is AS 1630

enter801, 2017-03-21:
So your servo wiring  is off.  Look at my config on how the servo is set up in the firmware.  Take your equivalent values over to RC8 that are specific to your machine, then merge my settings for the servo and BLTouch.  The wiring should be signal, 5v, then ground, signal on the inside pin of d11, 5v in middle, ground on outside pin.  So make sure you're using all 5 wires on it.  
Also, if your old firmware is working on the x-axis, and then it doesn't when you change firmware, look at the differences between the two.  Use a diff program like Beyond Compare.

robertch1969, 2017-03-21:
thanks @enter801 pretty new at this loving my printer but i am not a software guy and dont know a lot about the electronics would you be interested in doing a firmware for me for some $$

enter801, 2017-03-21:
Make sure you have RAMPS 1.4 selected as well, the stock firmware from Hictop has it on 1.3, which puts D11 on some other input.

robertch1969, 2017-03-21:
@enter801  yeah have it on 1.4 but still no joy will keep trying

ronenst1, 2017-07-27:
Thanks to all, I uploaded the new Marlin version after updating the configuration.h files and install the BLTouch on my printer. It is working perfectly with dual head. 
Link to my dual head design.
https://www.thingiverse.com/thing:2161827

probonopd, 2017-08-13:
@enter801 since the BLTouch is relatively pricey, do you think a __LJC18A3__ would work too on this board?

ronenst1, 2017-08-24:
I have tried the *LJC18A3*  and it is working fine but the performance of
the BLTouch is far away better and it is worth the price.

*Stolarski Ronen*

*System Engineer*

*Camon, Migdal Hamaym 5 street*

*Camon 20112*

*Israel*

*Tel: : 972-4-9883429*

*Cell: : 972-52-3793279*

BMW R1200GS 2005