MarlinFirmware/Marlin #2907: "FILAMENT_RUNOUT_SENSOR" do not work.

famtory, 2016-01-18:
I can't speak english. I used google translate.
======================================
marlin 1.1.0 rc3....
configuration.h
.
.
.
//===========================================================================
//========================= Filament Runout Sensor ==========================
//===========================================================================
#define FILAMENT_RUNOUT_SENSOR // Uncomment for defining a filament runout sensor such as a mechanical or opto endstop to check the existence of filament
                                 // In RAMPS uses servo pin 2. Can be changed in pins file. For other boards pin definition should be made.
                                 // It is assumed that when logic high = filament available
                                 //                    when logic  low = filament ran out
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  const bool FIL_RUNOUT_INVERTING = true;  // Should be uncommented and true or false should assigned
  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
.
.
.
I use microswitch.
.
I try D4, D5, D6 and D11.
and.
const bool FIL_RUNOUT_INVERTING = false;  // Should be uncommented and true or false should assigned
and.
//  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
.
when I microswitch on or off...does not happen anything.
.
====
I used marlin 1.0.6 and D42 ( http://forums.reprap.org/read.php?1,297350 )
It was work well.

Roxy-3DPrintBoard, 2016-01-18:
Are you using a Ramp's board?   

Did you notice the comment:

// In RAMPS uses servo pin 2. Can be changed in pins file. For other boards pin definition should be made.

KiteLab, 2016-01-18:
If it's only a simple switch
#define ENDSTOPPULLUP_FIL_RUNOUT
makes sense in about half of the possible cases.

famtory, 2016-01-19:
to. Roxy-3DPrintBoard 
Thank you.
I using Ramps 1.4
servo pin 2 = D6
I try D6
servo pin 1,3,4 = D11, D5, D4
I try also.
but not work.

to KiteLab
sorry.
I can't understand what you say.
please tell me again.

Roxy-3DPrintBoard, 2016-01-19:
@famtory   You need to KNOW what pin you have things connected.  Do you have a volt meter or a logic probe?   If you connect it to your pin, you should be able to use M42 P? S0 followed by a M42 P? S1 to see the pin toggle.

When you know a pin number you can connect to and toggle, we can help you more.

famtory, 2016-01-24:
My printer use coreXY.
so D2(axis X max pin) not use.

Mod pins_RAMPS_13.h
===========
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  // define digital pin 4 for the filament runout sensor. Use the RAMPS 1.4 digital input 4 on the servos connector
//  #define FILRUNOUT_PIN     4 
  #define FILRUNOUT_PIN     2
#endif
============

I connect sensor on D2
.

Now... It work!

Thank you.

closed 2016-02-07 by thinkyhead

373996214, 2016-05-11:
@famtory Hi, famtory! These days, I meet the same problem. I tried some methods just like change the pin for filament runout to Z_endstop_max or other unused pins, and modify the codes relating to filament runout. Unfortunately, I failed. 
The hardwares I used are arduino mege 2560 + remp 1.4. Instead of using SD card printing method, I use PC.  To be exact, I did not buy any components about SD Card. 
So, how you solve your problem? Are there any details I missed? Thank you!

373996214, 2016-05-12:
@KiteLab  Hi, KiteLab! These days, I meet the same problem. I tried some methods just like change the pin for filament runout to Z_endstop_max or other unused pins, and modify the codes relating to filament runout. Unfortunately, I failed. 
The hardwares I used are arduino mege 2560 + remp 1.4. Instead of using SD card printing method, I use PC. To be exact, I did not buy any components about SD Card. 
So, how you solve your problem? Are there any details I missed? Thank you!

Blue-Marlin, 2016-05-12:
https://github.com/MarlinFirmware/Marlin/issues/2907#issuecomment-174297577?

373996214, 2016-05-13:
@Blue-Marlin  Hi, Blue-Marlin. Yes, I met this issue. For solving it, I also used an unused pin D4. I just use a PC to print my model not SD card. The following codes are what I used. 
---------------------------------------------------------------Configuration.h
//========================= Filament Runout Sensor ==========================
//================================================== =========================
#define FILAMENT_RUNOUT_SENSOR // Uncomment for defining a filament runout sensor such as a mechanical or opto endstop to check the existence of filament
// In RAMPS uses servo pin 2. Can be changed in pins file. For other boards pin definition should be made.
// It is assumed that when logic high = filament available
// when logic low = filament ran out
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
const bool FIL_RUNOUT_INVERTING = true; // Should be uncommented and true or false should assigned
#define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
#define FILAMENT_RUNOUT_SCRIPT "M600"
#endif

-----------------------------------------------pin_RAMPS_13.h
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
// define digital pin 4 for the filament runout sensor. Use the RAMPS 1.4 digital input 4 on the servos connector
#define FILRUNOUT_PIN 4
#endif

-------------------------------------------------------------------------Marlin_main.cpp
void setup_filrunoutpin() {
#if HAS_FILRUNOUT
pinMode(FILRUNOUT_PIN, INPUT);
#if ENABLED(ENDSTOPPULLUP_FIL_RUNOUT)
WRITE(FILRUNOUT_PIN, HIGH);
#endif
#endif
}

--------------------------------------------------------------------Marlin_main.cpp
void manage_inactivity(bool ignore_stepper_queue/*=false*/) function.

#if HAS_FILRUNOUT
// if (IS_SD_PRINTING && !(READ(FILRUNOUT_PIN) ^ FIL_RUNOUT_INVERTING))
if ((READ(FILRUNOUT_PIN) ^ FIL_RUNOUT_INVERTING))
filrunout();
#endif

How do you think these codes? Am I right or wrong? Thank you!

Blue-Marlin, 2016-05-13:
If it works for you it's ok, if not not.
The main problem seems to be to fin a __unused__ pin.
Alone in the provided standard configuration D4 can be used by the FILRUNOUT_PIN, the FAN_PIN (for EEB) and SERVO3_PIN. If using D2 pay attention on it is not used for x_max. And so on.

373996214, 2016-05-13:
@Blue-Marlin  Thank you!
Today, I changed D4 to Z_Max_endstop. Although I did this, the same result happened. How do you think this problem? In some forums (http://forums.reprap.org/read.php?1,599640,606852), someone said the printing method, SD or host, has an influnce on filament runout function. I don`t it is real or not.
Anyway, I`ll try what you said.

yucelll, 2016-05-16:
hello. I try D4 but not working.

373996214, 2016-05-16:
@yucelll  I tried both D4 and  Z_Max_endstop, the same results I got, the printer still works. Do you try other pins and how is the result?