MarlinFirmware/Marlin #2909: LCD STUCK

CoderSquirrel, 2016-01-18:
I was using the old 1.0 firmware and just uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and changed the pins of the u8g lcd and it worked, now i uncomment and change but it get stuck in this image 
[image]

KiteLab, 2016-01-18:
Looking at the picture, that Marlin can't be that old. And of cause its not 1.0.

Marlins own device for the ST7920 has, in some aspects, a relative aggressive timing.
You could try the u8glibs original device

U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // original u8glib device

instead.

CoderSquirrel, 2016-01-18:
the picture is of the 1.1 version that i'm trying, now, for the first time to use, i have used the old one with no problem.
In fact it's SSD1306, i'm using it with
U8GLIB_SSD1306_128X64 u8g(19, 18, 16, 17);  // SW SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9
because it's a generic one, not the properly;

CoderSquirrel, 2016-01-18:
I tried using the ST7920 you said, didn't works :/

KiteLab, 2016-01-18:
If you did not use 
U8GLIB_ST7920_128X64_RRD u8g(0);
before it can not work.

I made a wrong suggestion because
>... just uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and changed the pins of the u8g lcd and it worked, ...

does not include changing the device.

CoderSquirrel, 2016-01-18:
i was talking about the 1.0 version of marlin where i just change it and worked, i was asking if it is need another change

CoderSquirrel, 2016-01-18:
what i mean is, the info screen not appears, if i comment #define SHOW_BOOTSCREEN it will be black, nothign shows up

KiteLab, 2016-01-18:
Not as far as i know. But my knowledge is limited to about one year back, and 1.0 was in use before that

Is that step in the colour of your photo visible with the naked eye or a problem introduced by taking the picture?
In the first case i'd look for changes in u8glib

CoderSquirrel, 2016-01-19:
Yeah, the oled has two colors, the seller send me the wrong one, but was ebay so...
I tried in one of those lcd ith encoder, that one everyone uses, and it works normal, but when i put the oled it freezes at the boot screen and don't goes to the info :/

KiteLab, 2016-01-19:
When you can see the bootscreen - can you communicate with Marlin via USB?
I try to find out if Marlin hangs, or if only the display content is not updated.

CoderSquirrel, 2016-01-19:
the printer responds, just isn't passing image to the display

CoderSquirrel, 2016-01-19:
i tried to change the bootscreen.
i commented 
```javascript
u8g.firstPage();
    do {
      if (show_bootscreen) {
        u8g.drawBitmapP(offx, offy, START_BMPBYTEWIDTH, START_BMPHEIGHT, start_bmp);
        lcd_setFont(FONT_MENU);
        #ifndef STRING_SPLASH_LINE2
          u8g.drawStr(txt1X, u8g.getHeight() - DOG_CHAR_HEIGHT, STRING_SPLASH_LINE1);
        #else
          int txt2X = (u8g.getWidth() - (sizeof(STRING_SPLASH_LINE2) - 1) * DOG_CHAR_WIDTH) / 2;
          u8g.drawStr(txt1X, u8g.getHeight() - DOG_CHAR_HEIGHT * 3 / 2, STRING_SPLASH_LINE1);
          u8g.drawStr(txt2X, u8g.getHeight() - DOG_CHAR_HEIGHT * 1 / 2, STRING_SPLASH_LINE2);
        #endif
        
      }
    } while (u8g.nextPage());
```

and add
lcd_implementation_status_screen ();
but it get stucks in this: 

file:///home/schirrel/Downloads/IMAG0635.jpg

KiteLab, 2016-01-19:
Ah ja.
What we see is the first 1/8 of the status-screen.
That's exactly what you __can__ see without the 
```
u8g.firstPage();
    do {
...
    } while (u8g.nextPage());
```
loop.

lcd_implementation_status_screen() is normally called from ultralcd.cpp around line 1714 (search for __firstPage()__)
Place some debug output
SERIAL_ECHO("before DOGLCD");
SERIAL_ECHO("in DOGLCD");
SERIAL_ECHO("in lcdDrawUpdate");
SERIAL_ECHO("in loop");

Hopefully we can see than where the problem is.

thinkyhead, 2016-02-07:
Any progress here?

CoderSquirrel, 2016-02-11:
HI, i'm gonna work with this today again, i havent make what kitelab purpose but i'm i'm about to

CoderSquirrel, 2016-02-11:
@ KiteLab  and @thinkyhead  it goes:

> before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
before DOGLCD
in DOGLCD
in lcdDrawUpdate
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop

i think that the right way, but keeps stucked

CoderSquirrel, 2016-02-11:
I thought that it may was repeating the bootscrean, but it only happens once, i use and serial echo too.
I think that may be something between the bootscreen and the end of it

Blue-Marlin, 2016-02-11:
If that output is repeated in a loop it's not Marlin but somewhere in the u8glib.
If it stops, remove the debug output, run it again and post the last messages you can see on the console.
Do you have the alive-dot in the lower right corner?

CoderSquirrel, 2016-02-12:
Hy @Blue-Marlin i see no alive-dot sadly.

CoderSquirrel, 2016-02-15:
@Blue-Marlin also,  about : 

> If that output is repeated in a loop it's not Marlin but somewhere in the u8glib.

The lcd worked at previously version of marlin

CoderSquirrel, 2016-02-15:
@Blue-Marlin @thinkyhead and @KiteLab  if you dont define SHOW_BOOTSCREEN in a normal screen( what i don't have to test) it work normally?
Because, the bootscreen works here, i can put whatever, basically the moment it ends that the screen stop to get info, or something like it.

closed 2016-04-23 by jbrazio