MarlinFirmware/Marlin #2910: REPRAP FULL SMART CART BUTTON NO WORKING

CoderSquirrel, 2016-01-18:
I am trying using graphic full smart cart but it doesn't work.
When i press the encoder, the button, it opens the menu and freezes, i change and put a simple button at the pin, instead of using the encoder, but in this way it doesn't even open the menu
i've done:
if (millis() > next_button_update_ms && READ(BTN_ENC) == 0){ 
         SERIAL_PROTOCOLLN("reading");
          newbutton |= EN_C;
            SERIAL_PROTOCOLLN("BT");
             SERIAL_PROTOCOLLN(newbutton);
          }

but it prints a blank space.

KiteLab, 2016-01-18:
uint8_t newbutton = 0;

SERIAL_PROTOCOLLN does not like uints. Cast to int

SERIAL_PROTOCOLLN((int) newbutton);

CoderSquirrel, 2016-01-18:
@KiteLab i was wondering now, i'm using only the arduino and ramps, nothing is pluged so far, just for trying menu ans graphic stuff, and its giving mintemp error annd then the printer is killed, maybe this is what is freeziin the screen?
How to disable this so i can work without the termistor plugged?

KiteLab, 2016-01-18:
Use fake sensors 999 or 998.

CoderSquirrel, 2016-01-18:
@KiteLab in this case i'm using a normal controller, i will try the fake sensor, i change the pins to 999? I'm more interface thant the mechanical part hehe

KiteLab, 2016-01-18:
Not the pins! The sensor type.

CoderSquirrel, 2016-01-18:
I saw now ^~^

closed 2016-02-07 by thinkyhead