MarlinFirmware/Marlin #2912: How to read analog inputs with Marlin

CoderSquirrel, 2016-01-19:
I have to read analog inputs with marlin, but when i use analogRead i figured out that it changes, for same strange reason, the temperature of the nozzle.
I tried to use READ, but this function is only Digital and i need to read it analog.
HOw to do this?

Wackerbarth, 2016-01-19:
You are mixing FastI/O pin numbers with Arduino pin numbers. If you are going to use the Arduino I/O routine (which you do need to read the analog input), you need to use the Arduino pin number.

Roxy-3DPrintBoard, 2016-01-19:
Also...  I believe only certain pins can be used as analog pins.   That is because there are only so man ADC (Analog to Digital Converters) in the chip.   (I think the number is 16 of them...)   Make sure you are on a pin that can do that kind of functionality.

CoderSquirrel, 2016-01-19:
i was trying to use only the arduino number, bu then i saw that when i use an analogRead it changed the temperature. And it may be problem,.
I'm using A3 and A4 that aren't being used with ramps

CoderSquirrel, 2016-01-19:
I try to follow the way TEMP_0 handle this, but i'm not quite understanding it

Roxy-3DPrintBoard, 2016-01-19:
> I try to follow the way TEMP_0 handle this, but i'm not quite understanding it

Post the code for TEMP_0 you are looking at and exactly what you don't understand.    But really...   This kind of stuff should be handled over at:

https://github.com/MarlinFirmware/MarlinDev/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

CoderSquirrel, 2016-01-19:
At what i look, this is were it makes the calc from digital to analog
#endif //ifndef SLOW_PWM_HEATERS
  
  switch(temp_state) {
    case 0: // Prepare TEMP_0
      #if defined(TEMP_0_PIN) && (TEMP_0_PIN > -1)
        #if TEMP_0_PIN > 7
          ADCSRB = 1<<MUX5;
        #else
          ADCSRB = 0;
        #endif
        ADMUX = ((1 << REFS0) | (TEMP_0_PIN & 0x07));
        ADCSRA |= 1<<ADSC; // Start conversion
      #endif
      lcd_buttons_update();
      temp_state = 1;
      break;
    case 1: // Measure TEMP_0
      #if defined(TEMP_0_PIN) && (TEMP_0_PIN > -1)
        raw_temp_0_value += ADC;
      #endif
      #ifdef HEATER_0_USES_MAX6675 // TODO remove the blocking
        raw_temp_0_value = read_max6675();
      #endif
      temp_state = 2;
      break;
    case 2: // Prepare TEMP_BED
      #if defined(TEMP_BED_PIN) && (TEMP_BED_PIN > -1)
        #if TEMP_BED_PIN > 7
          ADCSRB = 1<<MUX5;
        #else
          ADCSRB = 0;
        #endif
        ADMUX = ((1 << REFS0) | (TEMP_BED_PIN & 0x07));
        ADCSRA |= 1<<ADSC; // Start conversion
      #endif
      lcd_buttons_update();
      temp_state = 3;
      break;
    case 3: // Measure TEMP_BED
      #if defined(TEMP_BED_PIN) && (TEMP_BED_PIN > -1)
        raw_temp_bed_value += ADC;
      #endif
      temp_state = 4;
      break;
    case 4: // Prepare TEMP_1
      #if defined(TEMP_1_PIN) && (TEMP_1_PIN > -1)
        #if TEMP_1_PIN > 7
          ADCSRB = 1<<MUX5;
        #else
          ADCSRB = 0;
        #endif
        ADMUX = ((1 << REFS0) | (TEMP_1_PIN & 0x07));
        ADCSRA |= 1<<ADSC; // Start conversion
      #endif
      lcd_buttons_update();
      temp_state = 5;
      break;
    case 5: // Measure TEMP_1
      #if defined(TEMP_1_PIN) && (TEMP_1_PIN > -1)
        raw_temp_1_value += ADC;
      #endif
      temp_state = 6;
      break;
    case 6: // Prepare TEMP_2
      #if defined(TEMP_2_PIN) && (TEMP_2_PIN > -1)
        #if TEMP_2_PIN > 7
          ADCSRB = 1<<MUX5;
        #else
          ADCSRB = 0;
        #endif
        ADMUX = ((1 << REFS0) | (TEMP_2_PIN & 0x07));
        ADCSRA |= 1<<ADSC; // Start conversion
      #endif
      lcd_buttons_update();
      temp_state = 7;
      break;
    case 7: // Measure TEMP_2
      #if defined(TEMP_2_PIN) && (TEMP_2_PIN > -1)
        raw_temp_2_value += ADC;
      #endif
      temp_state = 8;//change so that Filament Width is also measured
      
      break;
    case 8: //Prepare FILWIDTH 
     #if defined(FILWIDTH_PIN) && (FILWIDTH_PIN> -1) 
      #if FILWIDTH_PIN>7 
         ADCSRB = 1<<MUX5;
      #else
         ADCSRB = 0; 
      #endif 
      ADMUX = ((1 << REFS0) | (FILWIDTH_PIN & 0x07)); 
      ADCSRA |= 1<<ADSC; // Start conversion 
     #endif 
     lcd_buttons_update();       
     temp_state = 9; 
     break; 
    case 9:   //Measure FILWIDTH 
     #if defined(FILWIDTH_PIN) &&(FILWIDTH_PIN > -1) 
     //raw_filwidth_value += ADC;  //remove to use an IIR filter approach 
      if(ADC>102)  //check that ADC is reading a voltage > 0.5 volts, otherwise don't take in the data.
        {
    	raw_filwidth_value= raw_filwidth_value-(raw_filwidth_value>>7);  //multipliy raw_filwidth_value by 127/128
        
        raw_filwidth_value= raw_filwidth_value + ((unsigned long)ADC<<7);  //add new ADC reading 
        }
     #endif 
     temp_state = 0;   
      
     temp_count++;
     break;

CoderSquirrel, 2016-01-19:
@Roxy-3DPrintBoard  i didn't went there because i just want to do a simple thing at my p.o.v, and i thought that MarlinDev was about dev new functionalities

Roxy-3DPrintBoard, 2016-01-19:
> i didn't went there because i just want to do a simple thing at my p.o.v, and i thought that MarlinDev was about dev new functionalities

It's not that big of deal.  But Marlin is more for vanilla straight forward uses of Marlin.   It is more for 'normal' support and feature discussions.   It is where we want people to go to get the Stable Releases.   The weird stuff is better kept on the other side of the fence.   

The code from RC-3 looks like this.   It is much different than what you posted up above.   Are you using an old release of Marlin?   This code looks simpler and easier to understand:
~~~~
  switch(temp_state) {
    case PrepareTemp_0:
      #if HAS_TEMP_0
        START_ADC(TEMP_0_PIN);
      #endif
      lcd_buttons_update();
      temp_state = MeasureTemp_0;
      break;
    case MeasureTemp_0:
      #if HAS_TEMP_0
        raw_temp_value[0] += ADC;
      #endif
      temp_state = PrepareTemp_BED;
      break;

    case PrepareTemp_BED:
      #if HAS_TEMP_BED
        START_ADC(TEMP_BED_PIN);
      #endif
      lcd_buttons_update();
      temp_state = MeasureTemp_BED;
      break;
    case MeasureTemp_BED:
      #if HAS_TEMP_BED
        raw_temp_bed_value += ADC;
      #endif
      temp_state = PrepareTemp_1;
      break;
~~~~

CoderSquirrel, 2016-01-20:
yeah, i'm using a version of 1.0;
I just looked at his piece of code to see how it gets analog values, but i don't quite believe that is what i'd use.
I'll move to marlindev

CoderSquirrel, 2016-01-20:
I'm lookin foward to solve this issue so i can PR the completed, and fully working version of joystick controlling menu.

thinkyhead, 2016-02-07:
MarlinDev is definitely the place to get into this. You will most likely add your devices and sensors to the TempState ISR, where all the other ADC code lives. The Filament Width Sensor is one example where a non-temperature-related sensor is also handled there.

closed 2016-02-07 by thinkyhead

bsculley, 2016-11-30:
Still in the wrong place, I'm sure, but: I'm trying to understand how the raw values are read from the ADC.  The syntax "+= ADC" implies that there is a definition of ADC somewhere, but I can't find it.

I changed out my thermistor for a thermocouple.  I'm advised to continue using the same pin (there not being any analog spares on mini-RAMBo), but I can't see how this is going to work since it has a 4k7 pull-up on it (real, not internal) and it is now supposed to be measuring a voltage instead of a resistance.

Anybody see how thia can work?

thinkyhead, 2016-11-30:
@bsculley The ADC is Arduino voodoo. It's a hardware register. Normally, with sketches that don't need super-fast performance, you can use the `analogRead()` function, which set a source to read, starts up the Analog-to-Digital converter, waits for a sufficient amount of time for the voltage to "charge up" the digital value from the input current, and then return the value acquired.

In Marlin we need to keep things moving, so we start the analog-to-digital converter on one interrupt cycle, and later, on the next interrupt cycle we just read the value from the ADC register, which will have had sufficient time to "charge up" and get a value.

Thermocouples tend to work differently. Instead of reading a resistance, they read a current, and then use their own internal circuitry to provide a value that we can read from a shift register digitally, one bit at a time.

What kind of thermocouple have you got? Is it the E3D described here?
```
6 : 100k EPCOS - Not as accurate as table 1 (created using a fluke thermocouple) (4.7k pullup)
```
…or maybe their other pet option?
```
20 : the PT100 circuit found in the Ultimainboard V2.x
```

bsculley, 2016-11-30:
Thanks for the clarification.  I have the AD595 flavor of thermocouple.  What I am wrestling with is whether I can just plug it into the same pin the thermistor is currently using, which has a hardware 4k7 pull-up resistor on it.  Will it work as-is or do I need to physically remove the pull-up?

Blue-Marlin, 2016-12-01:
https://github.com/MarlinFirmware/Marlin/blob/d97dc10216d9a431278052e5b60973bafeeed7e0/Marlin/temperature.cpp#L1622-L1720

Blue-Marlin, 2016-12-01:
@BillStruve 
Indeed. Looks as if the mini-RAMBo has no __accesible__ suited pin.
Instead of de-soldering a 4k7 resistor you could try to solder a wire, for the AD595s output, directly to a processor pin. That's about the same magnitude of difficulty but would leave the thermistor input intact.

BillStruve, 2016-12-01:
@Blue-Marlin just what I asked for, thanks. I would guess that 12 frames is enough time for the ADC to settle.

thinkyhead, 2016-12-01:
> Searching 234 files for "analogRead" only found my two

True. We don't like `analogRead`. Check out the `temperature.cpp` file starting at the line "`  switch (temp_state) {`" for what we _do_ like.

BillStruve, 2016-12-01:
@thinkyhead will do and change the code before making a pull request.

bsculley, 2016-12-01:
@Blue-Marlin re: mini-RAMBo  Thermocouple - can you suggest which pin to use and how to adjust the firmware to match?

bsculley, 2016-12-01:
I think the discussion got cross-threaded, so I’m replying here to make sure it gets routed properly.

 

Using another pin seems like a good compromise.  Can you suggest which pin to use and how to set up the firmware for it.

 

Thanks,

Bob

Blue-Marlin, 2016-12-01:
If you can't find out by yourself - you should not do it.
Compare with the 'extra analoge inputs' on the 'original' (not mini) RAMBo.
The look up the position on the processor at it's datasheet.

bsculley, 2016-12-02:
I totally get that, the question actually goes more to the fact that the analog pins on the Mega 2560 documentation are 82 through 97.  There are no references in this range in the firmware.  

 

What I’m looking for is the relationship between these pin references:

 

  #define TEMP_BED_PIN 2

  #define TEMP_0_PIN 0

  #define HEATER_1_PIN 7

  #define TEMP_1_PIN 1

  #define TEMP_2_PIN -1

 

And the pins on the chip.

 

Thanks,

Bob

Blue-Marlin, 2016-12-02:
https://www.youtube.com/watch?v=rfX0q1aNCos

Arduino pin numbering != processor pins.
[image]
[image]

```//
// Temperature Sensors
//
#define TEMP_0_PIN          0   // Analog Input // or 3,4,5,6
#define TEMP_1_PIN          1   // Analog Input
#define TEMP_BED_PIN        2   // Analog Input
```

thinkyhead, 2016-12-03:
> https://www.youtube.com/watch?v=rfX0q1aNCos

😆

bsculley, 2016-12-17:
Hi Scott,

 

Hopefully you can help me out.  I decided to use the E3D PT100 instead of a thermocouple.  I connected the output from the amplifier to the same inputs as the thermistor was using and left the 4k7 pullup in place.

 

I adjusted the firmware as follows:

 

#define TEMP_SENSOR_0 147

 

Based on this comment in the code:

 

// 147 is Pt100 with 4k7 pullup

 

When I start up my printer I get a temperature reading of 300.  Obviously not right, but I don’t know how to fix it.

 

What input would make the firmware think the temperature is 300?  How can I correct it?

 

Thanks,

Bob

thinkyhead, 2016-12-20:
I believe the E3D PT100 is meant to be used with thermistor ID 20, not 147.

bsculley, 2016-12-20:
I have finally figured this out:  ID 20 is intended for the PT100 Amplifier output on an analog input with no pull-up.  

 

The 147 device is for the naked PT100 (no amplifier) on an analog input with the 4k7 pull-up (hence the “47” designation I’m guessing).  I have this configuration working sort of.  The temperature table seems to be wrong.  It was providing negative readings at room temperature and the firmware will not operate the heater if measured temperature is negative.  I “fixed”  it but I don’t really trust the results.  It would be great if whoever developed the table could review it.

 

I don’t know if it is electronically feasible to use the amplifier on an analog input with the pull-up in place.  I would like to know the answer to this.

thinkyhead, 2016-12-21:
I guess the person to talk to would be @josefpavlik who created these tables…

https://github.com/MarlinFirmware/Marlin/commit/831fc2a95257d58cddd458d19e57065745fd4024

josefpavlik, 2016-12-21:
You are right, the 110, 147, 1010 and 1047 types are intended for naked Pt100 or Pt1000 connected to input with pullup 1k0 or 4k7,
I've used Pt1000 with 4k7 and works fine. Hopelessly I have not possibility to check other tables, however I believe that are ok as they are calculated by the same macro, differ only the input values.

However I not recommend to use the Pt100 with 4k7 pullup as this combination have very small change of output voltage, so very small difference between minimum and maximum value. So it's not precise.
at 0 degree it has resistance 100 ohm, so output voltage is 0.104V, a/d output is 21
at 250 degrees it has resistance 194 ohm, output 0.198V, a/d output is 40
so there are only 20 steps per 250 degrees, 12.5 degrees per step.
it's not usable

the Pt1000 with 4k7 takes the following values:
at 0 degree, resistance 1000ohm, output 0.877V, a/d 179
at 250 degree, resistance 1941ohm, output 1.46V, a/d 299
so they are about 2 degrees per step

slightly better result will be with pullup 1k0
0, 1000, 2.5, 512
250, 1941, 3.23, 676
1.5 degree per step

So I recommend to remove the tables for Pt100, I see that are unusable
and to add a comment that 1010 and 1047 devices are naked Pt1000 with pullup

bsculley, 2016-12-21:
I have actually implemented PT100 with device 147.  It works better than I imagined it would.  

 

The readings fluctuate +/- 5 degrees around the set point, but between thermal inertia in the printhead and the PID it seems ok.  I am getting good prints.

 

The main problem with the temperature table is that it was providing negative temps (-1/-10) at room temperature and the firmware wouldn’t start the heater.  I recalculated the table manually based on the PT100 resistance table.  I calculated the voltage across the divider for each temperature (assuming 5v and 4700 ohm pull-up).  Then I converted to steps (out of 1024) and finally multiplied by 16.  I don’t know where this magic number comes from, but it seems baked into the firmware somewhere.

 

Here is the revised table for your reference:

 

const short temptable_147[][2] PROGMEM = {

// only few values are needed as the curve is very flat  

{292,0},  // first two entries are bogus, just to get the heater running

{356,50},

{469,100},

{531,150},

{590,200},

{649,250},

{708,300},

{763,350},

{819,400},

{872,450},

};

 

I had to replace the first two entries with arbitrary values in order to get the heater to run, but the rest seem good.

 

I would advise not removing anything from the source but just commenting it appropriately and verifying that the temperature tables are good.,

 

Bob

josefpavlik, 2016-12-21:
> Then I converted to steps (out of 1024) and finally multiplied by 16.  I don’t know where this magic number comes from, but it seems baked into the firmware somewhere.

the magic constant 16 si OVERSAMPLENR
firmware reads the a/d 16 times to get more accuracy simulating more bits of a/d. In this case the step of a/d is not 12 degrees but slightly less. It's compatible with your experience of +/- 5 degrees.
I don't know why my table is not good, however I'll try to check and fix it.
I've closed my pull request of remove the devices 147 and 110.
Thank you for using and bugreporting my code.

Blue-Marlin, 2016-12-21:
Could make sense to recalculate the table with the resistors tolerance values (4K7 +5% and -5%). Could that be the reason for negative temperatures at room temperature.

bsculley, 2016-12-21:
I don’t think the temperature table is necessarily bad.  I also had a negative reading at room temperature using my manually calculated table.  It’s just something about the low end…  The first couple of entries (0 and 50 degrees) need to be fudged I guess.

 

I have noticed that the hotbed temp reads 24 at room temperature, even though my room is about 18.

 

Thanks for your attention to this.

 

Bob

Blue-Marlin, 2016-12-21:
5% tolerance of the pull up resistor has makes about +-1 ADC value.

But mustn't Rup in PtAdVal() be (Rup + PtRt(T,R0))?
```
#define PtA 3.9083E-3
 #define PtB -5.775E-7
 #define PtRt(T,R0) ((R0)*(1.0+(PtA)*(T)+(PtB)*(T)*(T)))
-#define PtAdVal(T,R0,Rup) (short)(1024/(Rup/PtRt(T,R0)+1))
+#define PtAdVal(T,R0,Rup) (short)((1024/(Rup + PtRt(T,R0)))*PtRt(T,R0) + 1)
 #define PtLine(T,R0,Rup) { PtAdVal(T,R0,Rup)*OVERSAMPLENR, T },
 
 #if ANY_THERMISTOR_IS(110) // Pt100 with 1k0 pullup
 const short temptable_110[][2] PROGMEM = {
   // only few values are needed as the curve is very flat
```

josefpavlik, 2016-12-22:
hi all
I've double checked the equations, the original program is right, so the table is correct.
However, I've made the patch with float calculation to increase the precision as recommended by @Blue-Marlin (thank you).

```
 #define PtA 3.9083E-3
 #define PtB -5.775E-7
 #define PtRt(T,R0) ((R0)*(1.0+(PtA)*(T)+(PtB)*(T)*(T)))
-#define PtAdVal(T,R0,Rup) (short)(1024/(Rup/PtRt(T,R0)+1))
-#define PtLine(T,R0,Rup) { PtAdVal(T,R0,Rup)*OVERSAMPLENR, T },
+#define PtAdVal(T,R0,Rup) (1024.0/(Rup/PtRt(T,R0)+1))
+#define PtLine(T,R0,Rup) { (short)(PtAdVal(T,R0,Rup)*OVERSAMPLENR+0.5), T },

```
I've also updated the comment, so now it's more clear and I've extend the tables up to 400 degrees.
This is the table for device 147, it's close to bsculley's table (the first two lines of their table are bogus)
341, 0
406, 50
469, 100
531, 150
591, 200
650, 250
707, 300
763, 350
818, 400

HOWEVER I'VE FOUND THE PROBLEM of the temperature alarm, so now should work correctly. Hopelessly, I cannot try it now as the printer is not working, so @bsculley , can you try this patch and tell me if it works?
https://github.com/josefpavlik/Marlin/tree/Pt100_fix
If it works for you, I'll raise the pull request

bsculley, 2016-12-22:
I would be happy to test this, but I am using the Prusa Research variant: https://github.com/prusa3d/Prusa-Firmware

 

Can you isolate your change so I can apply it to my version of the source?

 

Also tell me the testing procedure as I am not familiar with the problem of the temperature alarm.

 

 

https://github.com/josefpavlik/Marlin/tree/Pt100_fix

AnHardt, 2016-12-22:
Measuring temperatures with a PT100 at the normal thermocoupler inputs of RepRap boards (4k7 pull up resistors) is not that bad.
But the table has to be calibrated for the pull up resistor; they have tolerances what can make the result wrong by +-10°, or more easily.

Just put G5 in the third column in setup for table 147 and or 1047 (110. 1100)

[pt100-4k7.zip](https://github.com/MarlinFirmware/Marlin/files/669366/pt100-4k7.zip)

josefpavlik, 2016-12-23:
@bsculley 
try to add 
#  define HEATER_0_RAW_HI_TEMP 16383
#  define HEATER_0_RAW_LO_TEMP 0
and/or
#  define HEATER_BED_RAW_HI_TEMP 16383
#  define HEATER_BED_RAW_LO_TEMP 0
in your config.
and restore real values in the first two lines of the table.

When the range is not defined, the values are inverted for Pt sensor. I think that this is the problem that causes temperature error when the first two lines of your table have real values

bsculley, 2016-12-23:
Didn’t help.  I still get MIN TEMP error as soon as the firmware starts up.

 

Why would only the first entry be affected anyway?  Wouldn’t raw hi/lo settings affect interpretation of the entire table?

josefpavlik, 2016-12-23:
these values are the constant that is checked before converting the a/d values to celsius.
However seems that there is some other test that goes in error when the temperature is negative.
So please try one more thing:

const short temptable_147[][2] PROGMEM = {
  { 0, 0 },
  PtLine(  1, 100, 4700)
  PtLine( 50, 100, 4700)
  PtLine(100, 100, 4700)
  PtLine(150, 100, 4700)
  PtLine(200, 100, 4700)
  PtLine(250, 100, 4700)
  PtLine(300, 100, 4700)
  PtLine(350, 100, 4700)
  PtLine(400, 100, 4700)
};

in this case the calculated temperature should never goes below zero celsius Any temperature below 1 degree will be reported as value between 0 and 1.

andresdrb, 2019-03-19:
Hi guys,

I'm trying to connect a bare PT100 directly where my previous temp sensor was (as you are describing above), but after connecting it the printer halts saying maximum temperature reached (when It's just at room temperature).

I've seen a lot of discussions above about the temperature tables and so on, so not sure what I'm doing wrong...

Can anyone help?

Cheers

Bob-the-Kuhn, 2019-03-19:
Please open another issue for your problem & we'll be happy to help.

Please include in the new issue:
- A ZIP with the **configuration.h**, **configuration_adv.h** and **pins_YOUR_BOARD.h** files in it.
- A description of how the PT100 is connected to your board.

I expect that all you'll need to do is change the appropriate temp_sensor to 147.  This is the setting for a PT100 tied directly to the terminals on the controller (no PT100 amplifier) and the controller has a 4.7K pullup on it's thermal inputs.

andresdrb, 2019-03-19:
Ah wait, you may have given a key there.

I’m just replacing the thermocouple that comes with the Geetech A20M, so just swapped the connection cables for the PT100 ones, but you mention the thermal inputs requiring a 4.7K pullup. That maybe is what is missing (I guess the thermocouple didn’t need that pull-up). Could that be the problem?

I also tried to set it up with the amplifier, but I’m not being able to figure out which analog pin I could use on the board (GT2560 v3). I tried connecting to the ADC ones, but no idea which value I have to put in marlin for the pin (connected the signal cable to the ACD 5, but no matter if I put “5”, “A5” or anything else it doesn’t work )

Bob-the-Kuhn, 2019-03-19:
I think all you need to do is change configuration.h to read **#define TEMP_SENSOR_0 147**

I'm 99% sure the GT2560 has 4.7K pullups on it's three thermal inputs.

I like the heatsinks on the power FETs.  By any chance do the power FETs have **8743** on them?

andresdrb, 2019-03-20:
I think you lost me a bit with the FETs thing. What do you mean? Not sure what FET means.

Bob-the-Kuhn, 2019-03-21:
FETs are electronic switches.  The big power FETs for the bed and extruders are the ones with the heatsinks on them near the bed and thermistor connectors.

discip, 2020-06-15:
@josefpavlik
I know this issue is very old and closed, but I did not know, how else to contact you.
Could you please provide some help on the following:

I changed the stock thermistor of an *Artillery Sidewinder X1* with a **PT1000** from **E3D**.
But unfortunately the readings of it are off.
The measured temperature is ***25°C*** but the reading of the *PT1000* is ***32°C***.
I read in the following ***[thread](https://reprap.org/forum/read.php?219,815006,815089,quote=1#REPLY)*** about adjusting the coefficients. And after doing so it now reads ***22°C*** instead of ***25°C***.

Could you please help?

Thank you in advance
kind regards

bsculley, 2020-06-16:
Sounds like the adjustment is close.  Do you have way to measure operating temp.  I would check the reading at 300.

discip, 2020-06-16:
@bsculley
Thank you for the quick reaction! 👍
What exactly do I have to measure?
The *resistance* at **300°C**?

regards

bsculley, 2020-06-17:
I was suggesting that you set the hot end temp to  300 and see what the reading is.  You can use a laser thermometer to measure the actual temperature

discip, 2020-06-17:
@bsculley
I will report back, as soon as I have done this.

thanks

discip, 2020-06-19:
@bsculley
Sorry for being late,
but I think, first of all I have to provide more information on my setup, in order to consider every possible factor in this equation.

Before that : The heater reached only ***285°C***. I know I can change that in **Marlin**, but before that, I thought  I should share more information.

So I am running the machine using:
+ mainboard is **Big Tree Tech**'s *[SKR V1.4 Turbo](https://www.aliexpress.com/item/4000627417532.html?spm=a2g0o.productlist.0.0.65df6b0bhtumMG&algo_pvid=892c2ee2-b0fa-4356-bdf9-87a3868881ed&algo_expid=892c2ee2-b0fa-4356-bdf9-87a3868881ed-21&btsid=0b0a050b15925673554917417e68dd&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)*
+ heater cartridge is **E3D**'s *[High Precision Heater Cartridges](https://e3d-online.com/collections/printer-parts-electrical/products/high-precision-heater-cartridges)* (**`24V 40W`**)
+ thermistor is **E3D**'s *[PT1000 Temperature Sensor](https://e3d-online.com/products/pt1000-temperature-sensor?_pos=8&_sid=8a6f10fe4&_ss=r)* 
+ heater block is **E3D**'s *[Volcano Plated Copper HeaterBlock](https://e3d-online.com/products/plated-copper-volcano-block)*

I think all the other parts are not interlinked with this issue. If I am wrong please correct me.

Regarding **Marlin** please see the attached files.
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4804436/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4804437/Configuration.h.txt)

Thank you in advance
regards

oldman4U, 2020-06-28:
Hi.

Where have you got the PID values from?

   #define DEFAULT_Kp 59.04
   #define DEFAULT_Ki 10.49
   #define DEFAULT_Kd 83.04

Have you tried to do your own PID tuning with a higher value than 210ºC?
Have you tried a standard thermistor?
Have you tried same procedure but with a lower temp. like 200 to see if the heater stops?

AnHardt, 2020-06-28:
Temptable 1047 ends at 300°C. It is not applicable for higher temperatures.

```cpp
// Pt1000 with 4k7 pullup
const short temptable_1047[][2] PROGMEM = {
  // only a few values are needed as the curve is very flat
  PtLine(  0, 1000, 4700),
  PtLine( 50, 1000, 4700),
  PtLine(100, 1000, 4700),
  PtLine(150, 1000, 4700),
  PtLine(200, 1000, 4700),
  PtLine(250, 1000, 4700),
  PtLine(300, 1000, 4700)
};
```
The first and last table entry is no longer extrapolated like in former days (what in my opinion made more sense.).

thinkyhead, 2020-07-13:
> Temptable 1047 ends at 300°C.

For anyone visiting this old thread, note that it now goes up to 500°.
```cpp
  . . .
  PtLine(350, 1000, 4700),
  PtLine(400, 1000, 4700),
  PtLine(450, 1000, 4700),
  PtLine(500, 1000, 4700)
};
```