MarlinFirmware/Marlin #2913: Difference between zprobe_zoffset and Z home offset?

jminardi, 2016-01-19:
Does anyone know the difference between these two values? It seems they both serve the same purpose. If that is true, can one of them be removed? If that is not true could someone explain to me what the difference is?

brainscan, 2016-01-20:
One is for is you use a probe, the other is if you don't.

jminardi, 2016-01-20:
Why can't you use Z home offset with a probe?

Roxy-3DPrintBoard, 2016-01-21:
You can use both if you want.   Or you can just use the Z-Probe.

KiteLab, 2016-01-21:
They are logically different.
zprobe_zoffset is the distanze betrween the trigger point of the probe and the nozzles tip. Z_PROBE_OFFSET_FROM_EXTRUDER is copyd into zprobe_zoffset to make it adjustable. It's needed to get correct probes.
home_offset[Z_AXIS] is an adjustable distance to an endstop. For example for an alternate build plate, or printing on top of an other piece. Is needed if your build height varies.
endstop_adj[Z_AXIS] is only valid for delta machines. It's a difference to the endstop to adjust for not so perfect mounted endstops - to get the x/y centre in the middle of the bed.

For G92 the value is not stored, but is realized by a shift of stepper and object coordinate system. A relict from the cnc usage to get object insted of machine coordinates. In the reprap world normally exclusively used to reset the filament length at layer change - to avoid infinite large numbers. No one will hinder you to drive to the beds height and send G92 Z0, but as the position is relative fix over time there are more convenient ways to get there.

The negative side of that convenience is a certain difficulty to handle the endstops right. Here Marlin has deficits.

thinkyhead, 2016-02-07:
The `home_offset[]` is applied to the `current_position` immediately after homing with `G28`. You can set it explicitly with `M206`, or you can set it based on the current XYZ position with `M428` or the "Set Home Offsets" menu item. The intent of `home_offset[]` is to allow you to change the printer's origin-point after homing with `G28` (and to retain that offset on the next `G28`). So, for example, if after homing you find that the Z position is too high or too low, you _could_ adjust the endstop and home again, _or_ you can adjust the Z position and do `M428`. Then when you start to print, the height will be okay. The X and Y offsets are used less often, but you could use them as a way to force the printer to use a corner of the bed instead of the center, for example.

closed 2016-04-23 by jbrazio