MarlinFirmware/Marlin #2914: Help with Servo and Z Probe

hoang82, 2016-01-20:
I have configured my Prusa i3 with a BLTouch sensor. It simply replaces a servo and mechanical endstop. Configuration is the same. It works well on my Prusa. But I have recently aquired a Delta printer and started configuring the RC3 firmware. All axis and Axis endstops are triggering as normally should. Servo/Probe Pin (BLTouch) will not deploy when I issue G29 after G28 like my prusa does. So I manually deploy the pin down with M401 and move Z down while activating the probe pin. z-probe remains OPEN and does not trigger to stop Z from moving further. I tried adjusting values in the endstop section and have only been able to invert it from OPEN to TRIGGERED.  I absolutely cannot get the endstop to trigger. But on my prusa, I was able to get it to work the first time.

I am not sure if it is the board I am using or if it is my configuration in the config.h. My prusa is running Ramps 1.4 and my delta is running a MKS-Mini v1.2 which is a ramps 1.4 and gt2650 combo. The only thing I had to change was the Servo Pin to Pin 7 to activate it on the board since it does not have servo pins. I am wondering if this is the issue that causes the probe to not retract and deploy when issuing a G29 command. And also I cannot get the endstop to trigger..

I had it semiworking on Rich Cattell's Testing firmware but the m401 and m402 commands were not listed in the main. I had to issue the probe up and down commands before and after G29 which is not what I had in mind since my Prusa doesnt require that. 

Any help would be great!

hoang82, 2016-01-20:
Found it.. I should have looked further into the pins_RAMPS_13.h file. Just a few lines down was the:

 #define Z_MIN_PROBE_PIN  32

Changed to

#define Z_MIN_PROBE_PIN  18

But the firmware seems to not support servo deploy and retract code in G29. Using m401 and m402 in a script or start code would in the mean time work.

KiteLab, 2016-01-20:
Would you please be so kind to post your servo configuration?
```
/*********************************************************************\
* R/C SERVO support
* Sponsored by TrinityLabs, Reworked by codexmas
**********************************************************************/

// Number of servos
//
// If you select a configuration below, this will receive a default value and does not need to be set manually
// set it manually if you have more servos than extruders and wish to manually control some
// leaving it undefined or defining as 0 will disable the servo subsystem
// If unsure, leave commented / disabled
//
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M851 to set the Z probe vertical offset from the nozzle. Store that setting with M500.
//
//#define X_ENDSTOP_SERVO_NR 1
//#define Y_ENDSTOP_SERVO_NR 2
//#define Z_ENDSTOP_SERVO_NR 0
//#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {70,0}} // X,Y,Z Axis Extend and Retract angles

// Servo deactivation
//
// With this option servos are powered only during movement, then turned off to prevent jitter.
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
  // Delay (in microseconds) before turning the servo off. This depends on the servo speed.
  // 300ms is a good value but you can try less delay.
  // If the servo can't reach the requested position, increase it.
  #define SERVO_DEACTIVATION_DELAY 300
#endif
```

hoang82, 2016-01-20:
#define NUM_SERVOS 1
#define Z_ENDSTOP_SERVO_NR 0
#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {10,90}} // X,Y,Z Axis Extend and Retract angles

KiteLab, 2016-01-20:
That looks ok.

What kind of G29
//#define MANUAL_BED_LEVELING 
//#define MESH_BED_LEVELING 
//#define AUTO_BED_LEVELING_FEATURE
//#define AUTO_BED_LEVELING_GRID

hoang82, 2016-01-20:
Copied from the Bed Auto Leveling section under config.h

I am not sure or not even able to determine where or if servo deploy and retract is in the marlin_main. I think I read some older posts from a few months back that a person had gotten the servo to automatically deploy before G29 or similar probing commands.

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

// @section bedlevel

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
#define DEBUG_LEVELING_FEATURE
//#define Z_MIN_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#if ENABLED(AUTO_BED_LEVELING_FEATURE)

  // There are 2 different ways to specify probing locations.
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't colinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution).
  // Note: this feature generates 10KB extra code size.
  #define AUTO_BED_LEVELING_GRID  // Deltas only support grid mode.

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    // set the rectangle in which to probe
    #define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION -DELTA_PROBABLE_RADIUS
    #define RIGHT_PROBE_BED_POSITION DELTA_PROBABLE_RADIUS
    #define FRONT_PROBE_BED_POSITION -DELTA_PROBABLE_RADIUS
    #define BACK_PROBE_BED_POSITION DELTA_PROBABLE_RADIUS

    #define MIN_PROBE_EDGE 10 // The Z probe square sides can be no smaller than this.

    // Non-linear bed leveling will be used.
    // Compensate by interpolating between the nearest four Z probe values for each point.
    // Useful for deltas where the print surface may appear like a bowl or dome shape.
    // Works best with ACCURATE_BED_LEVELING_POINTS 5 or higher.
    #define AUTO_BED_LEVELING_GRID_POINTS 9

  #else  // !AUTO_BED_LEVELING_GRID

      // Arbitrary points to probe.
      // A simple cross-product is used to estimate the plane of the bed.
      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER 0     // Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 0   // Z probe to nozzle Y offset: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -0.95  // Z probe to nozzle Z offset: -below (always!)

  #define Z_RAISE_BEFORE_HOMING 5       // (in mm) Raise Z axis before homing (G28) for Z probe clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case.

[... 117 lines cut ...]
  // The Z min endstop will need to set properly as it would without a Z probe
  // to prevent head crashes and premature stopping during a print.
  //
  // To use a separate Z probe endstop, you must have a Z_MIN_PROBE_PIN
  // defined in the pins_XXXXX.h file for your control board.
  // If you are using a servo based Z probe, you will need to enable NUM_SERVOS,
  // Z_ENDSTOP_SERVO_NR and SERVO_ENDSTOP_ANGLES in the R/C SERVO support below.
  // RAMPS 1.3/1.4 boards may be able to use the 5V, Ground and the D32 pin
  // in the Aux 4 section of the RAMPS board. Use 5V for powered sensors,
  // otherwise connect to ground and D32 for normally closed configuration
  // and 5V and D32 for normally open configurations.
  // Normally closed configuration is advised and assumed.
  // The D32 pin in Aux 4 on RAMPS maps to the Arduino D32 pin.
  // Z_MIN_PROBE_PIN is setting the pin to use on the Arduino.
  // Since the D32 pin on the RAMPS maps to D32 on Arduino, this works.
  // D32 is currently selected in the RAMPS 1.3/1.4 pin file.
  // All other boards will need changes to the respective pins_XXXXX.h file.
  //
  // WARNING:
  // Setting the wrong pin may have unexpected and potentially disastrous outcomes.
  // Use with caution and do your homework.
  //
  #define Z_MIN_PROBE_ENDSTOP

#endif // AUTO_BED_LEVELING_FEATURE

Roxy-3DPrintBoard, 2016-01-20:
> But the firmware seems to not support servo deploy and retract code in G29. Using m401 and m402 in a script or start code would in the mean time work.

Check out deploy_z_probe() and stow_z_probe() in Marlin_main.cpp.   You can pretty much paste those lines in any where you want something to happen.   If you want a command to do one of those things, you can just paste the function call in at the start or end of the command.

KiteLab, 2016-01-20:
Ah DELTA

You can try:
About line 2885
```
   #endif // AUTO_BED_LEVELING_GRID

    #if ENABLED(Z_PROBE_SLED)
      dock_sled(false); // engage (un-dock) the Z probe
    #elif ENABLED(Z_PROBE_ALLEN_KEY)  || HAS_SERVO_ENDSTOPS
      #if HAS_SERVO_ENDSTOPS
         raise_z_for_servo();
      #endif
      deploy_z_probe();
    #endif

    st_synchronize();

    if (!dryrun) {

```

And around line 3150
```
    #endif // !AUTO_BED_LEVELING_GRID

    #if ENABLED(DELTA)
      // Allen Key Probe for Delta
      #if ENABLED(Z_PROBE_ALLEN_KEY) || HAS_SERVO_ENDSTOPS
        stow_z_probe();
      #elif Z_RAISE_AFTER_PROBING > 0
        raise_z_after_probing();
      #endif
    #else // !DELTA
      if (verbose_level > 0)
        plan_bed_level_matrix.debug(" \n\nBed Level Correction Matrix:");
```

And around line

hoang82, 2016-01-20:
Thanks for the info guys...

Sorry for my rookiness..

between the ( ) I need to add a value?

KiteLab, 2016-01-20:
No. 
deploy_z_probe() has no parameters and no build in lift.
stow_z_probe() has a parameter, if to lift or not - default is true - that's what you want.

hoang82, 2016-01-20:
Thanks again,,

I did a G30 command and the probe did deploy.. so I know it's working..

now I just need to figure out where to put the codes under G29

Roxy-3DPrintBoard, 2016-01-20:
> now I just need to figure out where to put the codes under G29

G29 should already be setup correctly.   Probably you don't have the 
~~~~
  #define Z_RAISE_BEFORE_HOMING 20       // (in mm) Raise Z axis before homing (G28) for Z probe clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case.
  #define Z_RAISE_BEFORE_PROBING 15   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 20 // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 20    // How much the Z axis will be raised after the last probing point.
~~~~
values set up right in Configuration.h

hoang82, 2016-01-20:
My luck I pasted it in the right spot! Now it's working.. Appreciate the help guys..

 #if ENABLED(AUTO_BED_LEVELING_GRID)

      // probe at the points of a lattice grid
      const int xGridSpacing = (right_probe_bed_position - left_probe_bed_position) / (auto_bed_leveling_grid_points - 1),
                yGridSpacing = (back_probe_bed_position - front_probe_bed_position) / (auto_bed_leveling_grid_points - 1);

      #if ENABLED(DELTA)
   deploy_z_probe();

AND 

            eqnBVector[probePointCounter] = measured_z;
            eqnAMatrix[probePointCounter + 0 * abl2] = xProbe;
            eqnAMatrix[probePointCounter + 1 * abl2] = yProbe;
            eqnAMatrix[probePointCounter + 2 * abl2] = 1;
            indexIntoAB[xCount][yCount] = probePointCounter;
          #else
            bed_level[xCount][yCount] = measured_z + z_offset;
          #endif

          probePointCounter++;

          idle();

        } //xProbe
      } //yProbe

stow_z_probe(); 

Otherwise, the Servo deploy and stow does not activate before and after G29 auto bed grid

hoang82, 2016-01-20:
Roxy, I am using the RC3 firmware. As per instructions, I moved Config.h and config_adv.h from Delta folder to the main marlin folder.

closed 2016-02-09 by thinkyhead

nkuck, 2016-07-24:
I'm having trouble getting my BLTouch to work on my Zonestar Prusa i3. Wondering if it's a configuration.h problem or wiring. As instructed, I connected to 5v and gnd, with the orange lead going to A4 at the display connection on the Melzi board. Anyone know where I might focus? When I power it up, I get two retractions from the pin, and the red led stays lighted. The probe doesn't seem to affect the endstop, crashes into the bed. Thanks for any help.

thinkyhead, 2016-07-24:
@nkuck Why don't you post this as a new issue? There is still more work to be done to deal with BLTouch error conditions, but the basic support was only just added.

nkuck, 2016-07-24:
I am hesitant to use the RCBugFix version until it is officially released.

thinkyhead, 2016-07-25:
@nkuck If you send me your BLTouch then I can use it for coding and debugging. Otherwise, I will have to rely on users who own this device and who are willing to do testing to help me develop the code.

nkuck, 2016-07-27:
After a few adjustments, I now do a G29 and get the probe to take the first reading, raise and go to the second, raise more and go to the third (by this time the probe can't touch the bed anymore), raise and go to the fourth point. The red light on the probe is flashing by then. What is happening? I can't get the probe to stay down between probe points. Suggestions??

thinkyhead, 2016-07-27:
@nkuck That sounds like a bug. What version of Marlin are you running?

hoang82, 2016-07-27:
@nkuck I have not ran into this problem. But the problem I am seeing from what you are describing is that the acceleration is too slow. The probe deploys and retracts fairly quick so you need to up your acceleration speed. It's going into test mode right about the 2nd probing location.

@thinkyhead Sorry for the late reply..  I am using Marlin firmware but an older one. Not sure what version but I configured it to work with my BLTouch and it worked pretty much the first go. And it's the basic mechanical servo settings. Ever since then I uploaded the firmware to my Prusa i3, I have not touched the firmware and it has been a very long while. I have zero issues with my BLTouch. With that being said, it is up to you if you'd stay on the decision of continuing the BLTouch support. For me honestly, it is already working with the older version of Marlin so to add more coding to the latest firmware prolly won't make a big difference IMO. But I am curious to know and see what could be done for ease of configuration for experienced and newbie BLTouch users..

nkuck, 2016-07-27:
Running version 1.0.

Live simply!
You can only take with you what you have given.

nkuck, 2016-07-27:
Hmmm, acceleration for which axis?

Live simply!
You can only take with you what you have given.

thinkyhead, 2016-07-27:
> I am curious to know and see what could be done for ease of configuration for experienced and newbie BLTouch users

Well, with the newest Marlin you just enable `BLTOUCH` and the correct "servo" settings are made for you. However, the BLTouch has extra features, such as the ability to report that it is in an "error" state, and there's a sequence of commands that can be given to it to make sure it's deployed (if the pin doesn't get stuck). So I want to add that stuff, which was discussed in another topic here.

nkuck, 2016-07-27:
What is the version that has BLTouch implemented? Is it stable? I’d like to give it a try.

thinkyhead, 2016-07-28:
> What is the version that has BLTouch implemented?

@nkuck 1.1.0-RC7 and `RCBugFix`.

> Is it stable?

We will be altering some code prior to release, but nothing major.

If you actually mean "is it bug free?" then the answer is: mostly. The BLTouch portion is bug-free, but it doesn't yet handle the extras provided by the BLTouch sensor. There may be some minor issues that we haven't encountered yet, but so far RC7 is behaving well.

nkuck, 2016-07-28:
I am trying to compile and getting:

Marlin_main.cpp:3550: error: 'DEPLOY_PROBE' was not declared in this scope
     if (DEPLOY_PROBE()) return;
                      ^
Marlin_main.cpp:3612: error: 'probe_pt' was not declared in this scope
           float measured_z = probe_pt(xProbe, yProbe, stow_probe_after_each, verbose_level);
                                                                                           ^
Marlin_main.cpp:3655: error: 'STOW_PROBE' was not declared in this scope
     if (STOW_PROBE()) return;
                    ^
exit status 1
‘DEPLOY_PROBE' was not declared in this scope

Trying to familiarize myself with this version and enabling the BLTouch probe option.

Blue-Marlin, 2016-07-28:
Try to define at lease one servo
```
#define NUM_SERVOS 1
```

nkuck, 2016-07-28:
Yes, I had 1 selected.

thinkyhead, 2016-07-28:
@nkuck Do you have either a `Z_MIN_PROBE_PIN` defined, or `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled?

nkuck, 2016-07-29:
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN ##### YES
Z_MIN_PROBE_PIN ##### NO
It compiles for me, but when I upload it and send a G28 code, the two Z-axis motors want to run in opposite directions and independent of each other???

> On Jul 28, 2016, at 7:41 PM, Scott Lahteine <notifications@github.com> wrote:
> 
> Z_MIN_PROBE_PIN defined, or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled?

thinkyhead, 2016-07-29:
>  two Z-axis motors want to run in opposite directions and independent of each other

Flip the plug on the one that moves the wrong way.

nkuck, 2016-07-29:
I don’t understand how a firmware version can independently operate the servos?
Can’t that be software-corrected?

thinkyhead, 2016-07-29:
@nkuck What the what now? I don't understand the context of your question.

Blue-Marlin, 2016-07-29:
@thinkyhead 
When one of the z-steppers runs into the other direction, this can have two reasond. Ether it's allways the case, than it may help to flip the plug. Or the stepprate and or acceleration is much to high - the torque is low, the noise is high, loosing lots of steps.

@nkuck 
> I don’t understand how a firmware version can independently operate the servos?

Shouldn't the last word be stepper?

thinkyhead, 2016-07-29:
@Blue-Marlin I was thinking perhaps the Z homing or movement speed could be too fast. I've seen that before. But I've never seen it cause the motor to move in the wrong direction. I imagine the speed must have to be very fast before that occurs.

nkuck, 2016-07-29:
Thanks, I will check it out. I have made a number of changes lately trying to get things to work. Maybe I messed with the speeds too much. 

Live simply!
You can only take with you what you have given.

Blue-Marlin, 2016-07-29:
@thinkyhead 
Fast is relative. For our Pi3 z-axis 4mm/s is to much, but 3.6 is ok when the acceleration is slow enough. Even 3mm/s can be to much when the acceleration is to high.

hornbrook, 2016-08-12:
Hello Thinkyhead,
I'd be willing to send you a BLTouch, if it will help solve some of our issues. Thanks for all your work!

thinkyhead, 2016-08-12:
@hornbrook I'm happy to take you up on your offer! The basic BLTOUCH support is there, but I haven't gotten 'round to supporting the error conditions and reset capabilities. I am anxious to get it completed.

I will be traveling for a couple of weeks, so I won't be able to work with it until I return. Or, possibly you could ship it to me while I'm in Austin, Texas. Write me an email or use [my contact form](http://www.thinkyhead.com/contact) to keep in touch.

Anastasio77, 2016-11-03:
I have the same problem with @nkuck. Did you find a solution?

I'm having trouble getting my BLTouch to work on my Zonestar Prusa i3. Wondering if it's a configuration.h problem or wiring. As instructed, I connected to 5v and gnd, with the orange lead going to A4 at the display connection on the Melzi board. Anyone know where I might focus? When I power it up, I get two retractions from the pin, and the red led stays lighted. The probe doesn't seem to affect the endstop, crashes into the bed. Thanks for any help.

Anastasio77, 2016-11-03:
?

thinkyhead, 2016-11-03:
@Anastasio77 BLTouch is pretty solid at this point. Have you tested with `RCBugFix`? The only bit that needs tweaking in some cases is to reduce the `HOMING_BUMP_DIVISOR` for Z so the probe isn't so slow that it causes the probe to error out.

nkuck, 2016-11-04:
I never did get it to work so gave up. 

Live simply!

thinkyhead, 2016-11-04:
@nkuck Did you get a new probe of a different kind and throw out the BLTouch?

Anastasio77, 2016-11-04:
My printer is Geeetech Prusa i3 pro b. The fitmware is Marlin 1.0.2 which downloaded from Geeetech website. Can i use a different Marlin firmware? Is there a possibility the sensor to be defective?

thinkyhead, 2016-11-04:
@Anastasio77 I can't say if the sensor is defective. I can say that the current `RCBugFix` supports the Geeetech i3.

Anastasio77, 2016-11-10:
Hallo again,
Here are thre videos describing the problem.
 https://youtu.be/kOeVCM9bh5Y
 https://youtu.be/zN5XBMBtOYY
https://youtu.be/gFBjkMAkEbA
Any ideas?
Thanks.

Grogyan, 2016-11-10:
Check the status of your endstop.  Ensure that you have zero effect as
zmin.
Try inverting the state of endstop

Double check that the servo commands are 10, 90
Default sometimes is 90, 0

thinkyhead, 2016-11-15:
> Double check that the servo commands are 10, 90. Default sometimes is 90, 0

It appears this probe works a lot like BLTouch, so you might try that setting, if you haven't already. The servo angles will be configured automatically for you when `BLTOUCH` is enabled.

It's not clear what version of Marlin you're using. Are you testing with `RCBugFix` now?

Anastasio77, 2016-11-15:
Hallo, it is not a firmware problem. I sent my firmware to the manufacturer and they checked it. The sensor is defective. It is an imitation of BLTOUCH. Thanks for the support.

AndKe, 2017-06-26:
I've enabled BLTOUCH and one servo, on RAMPS 1.4  - the device's LED is on.
The BLTOUCH is connected to the leftmost servo connector (to the right of the reset switch)
I am unable to make it do anything by those servo commands.
M280 P0 S10 ; pushes the pin down
M280 P0 S90 ; pulls the pin up

Could  you please confirm if I they are expected to work ? on that servo port ?

Roxy-3D, 2017-06-26:
> The BLTOUCH is connected to the leftmost servo connector (to the right of the reset switch)

This doesn't mean anything unless we know how your board is orientated.   You want to connect up the servo connector that is closest to the reset switch.

AndKe, 2017-06-26:
good point . yes, it's connected to the one closest to the reset switch. GND closest to the edge of the PCB, - the sensor gets power.
If this is supposed to work, I can check the PWM with a oscilloscope tomorrow .

Bob-the-Kuhn, 2017-06-27:
Does your BLTouch have a serial number?  If yes there should be a dim LED on if the servo signal is active & connected.  Best way to see it is to pull the pin down so the orange LED goes out.  It's on the bottom edge of the PWB.

AndKe, 2017-06-27:
at 1630us to 1930 microseconds it does the self-test dance "up'n'down"
there is no point between 800us and 2200us where it just drops the pin down.
if I pull the pin, the LED goes out.

I do not know is S120 is in the self test area of 1630...1930us , but if
even that worked, I'd know the device is defective.

AndKe, 2017-06-27:
Something is fishy about the servo output on RAMPS 1.4 
I don't see any PWM, my configuration attached.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1105918/Configuration.h.zip)

:

Bob-the-Kuhn, 2017-07-01:
I've finally had time to get back to this.

I've downloaded your configuration.h file & compiled it.

The problem is that the wrong motherboard/controller is specified.  Change the  motherboard from "BOARD_RAMPS_13_EFB" to "BOARD_RAMPS_14_EFB".  If you go to eBay today all the boards that are advertised as RAMPS (with no version specified) is 1.4.

The older RAMPS boards have different pin assignments for the servos.

AndKe, 2017-07-04:
@Bob-the-Kuhn Thank you very much, makes sense. My RAMPS board is somehow not exactly 1.4, nor did it look like 1.3 , it's an old thing, with newer stepper controllers that just worked.
I just returned from a business trip, and I am going for a 5week vacation, but I am confident you solved it, thank you.

Bob-the-Kuhn, 2017-07-04:
Have a great vacation.