MarlinFirmware/Marlin #2917: Different temp readings on Ramps 1.3/Bugfix

dmummert, 2016-01-21:
I'm getting wildly different readings on identical thermistors that used to produce identical readings.  I ohmed things out and they're ok.  The bed thermistor reads room temperature correctly, but the hotend, which should be reading 21C, thinks it is 102.  I don't know where to look.

KiteLab, 2016-01-22:
As a first step change bed and extruder thermistor with each other at the board connectors side.

Roxy-3DPrintBoard, 2016-01-22:
Wow!   That is a really good idea!    I always read everything you write!

brainscan, 2016-01-22:
Could it be one of the thermistor leads shorting out on the hotend?

dmummert, 2016-01-22:
That actually worked.  Didn't help, but it worked.  Normally I'm a lot more thorough before I start asking dumb questions but I had just come down from absorbing a little too much code in one session.  Definitely not software.

It does suggest something though.  Most of the printers have unused ADC slots and it wouldn't be out of line to write a short routine that does a room temperature comparison check on all of the temp sensors by adding an extra sensor somewhere.  It would give you a ballpark idea of whether you've got a thermistor going bad (as is apparently my case).  The hard part is deciding when to invoke it.  Definitely at POST, when  everything should be isothermic.  

On to drilling out a thermistor.  I glued it in.  I'll close this out this evening.

closed 2016-01-23 by dmummert