MarlinFirmware/Marlin #2918: SERIAL_ECHOLN show garbage from menu_eidt_type

fred2088, 2016-01-22:
it will display garbage on serial debug screen 6~8 times when try to debug this function ?

Regards

#define menu_edit_type(_type, _name, _strFunc, scale) \
  bool _menu_edit_ ## _name () { \
    bool isClicked = LCD_CLICKED; \
    sendCurrentPageId(&Ppage); \
    if ((int32_t)encoderPosition < 0) encoderPosition = 0; \
    if ((int32_t)encoderPosition > maxEditValue) encoderPosition = maxEditValue; \
    if (lcdDrawUpdate) { \
      SERIAL_ECHOLN(editLabel);\   <------------------
      lcd_implementation_drawedit(editLabel, _strFunc(((_type)((int32_t)encoderPosition + minEditValue)) / scale)); \
    } \
    if (isClicked) { \
      *((_type*)editValue) = ((_type)((int32_t)encoderPosition + minEditValue)) / scale; \
      lcd_goto_menu(prevMenu, prevEncoderPosition); \
    } \
    return isClicked; \
  } \

KiteLab, 2016-01-22:
Try SERIAL_ECHOLNPGM(

fred2088, 2016-01-22:
sketch\dogm_lcd_implementation.h:472:3: note: in expansion of macro 'SERIAL_ECHOLNPGM'

   SERIAL_ECHOLNPGM(pstr);

   ^

exit status 1
initializer fails to determine size of '__c'

KiteLab, 2016-01-22:
Ah, yes.
The pstr is already in the EEPROM we are locking for the other way around, like in.

```/* Arduino < 1.0.0 is missing a function to print PROGMEM strings, so we need to implement our own */
char lcd_printPGM(const char* str) {
  char c, n = 0;
  while ((c = pgm_read_byte(str++))) n += charset_mapper(c);
  return n;
}
```

fred2088, 2016-01-22:
is there a way to output to SERIAL_ECHOLN ?

Thanks

thinkyhead, 2016-02-07:
@fred2088 Try `serialprintPGM(editLabel); SERIAL_EOL;` and see if that does the trick.

closed 2016-04-23 by jbrazio