MarlinFirmware/Marlin #2919: Setting up Marlin on Kossel

Jan9320, 2016-01-22:
Hello,
I'm quite new to this. I was trying to set up my kossel with newest Marlin software but I can't get marlin to compile.

I have older version of Marlin running already, but auto-bed leveling doesn't work as it should.

I tried running firmware with arduino 1.06 but I get some unidentified "MSerial" reports... I already changed Configuration.h to my needs.

Can anyone suggest what to fix? I'm sure its just a silly mistake I made.

Thank you.

[Marlin in Arduino 1.0.6.txt](https://github.com/MarlinFirmware/Marlin/files/100594/Marlin.in.Arduino.1.0.6.txt)

thinkyhead, 2016-02-07:
Hi Jan, are you using the latest 1.1.0-RC3 for this? Please paste the error you're getting from the compiler. Also, if the error seems weird, give Arduino 1.6.7 a try.

Jan9320, 2016-02-09:
Hello,
Thank you for reply. Marlin firmware is already running on my printer. Thank.

closed 2016-02-09 by Jan9320