MarlinFirmware/Marlin #2920: Auto MESH_BED_LEVELING?

Poquaruse, 2016-01-22:
Hi all,

thanks for providing such a wonderful piece of software. I've been using it quite successfully for some prints.

However, my current printer has a sag in the x-axis which I can't seem to fix. It is not very large (roughly 0.1mm in the middle) but noticeable. The "classical" auto bed leveling procedure does not account for a warped bed but only for a perfectly flat and tilted one. This means that the sag is not compensated and might even reduce ABL quality when taken into account (e.g. more than 2 grid points --> higher values in the middle are measured --> edges will be too low).

Mesh bed leveling on the other hand seems to be able to compensate for that problem as well. But it doesn't work automatically using a sensor. Is there currently a way to integrate a sensor? If not, could you please point me to a starting area where I can have a look at the code and see if I'm able to fix that? (Which I doubt by the way. :D)

Thanks again and best regards!

epatel, 2016-01-22:
Hi, great to hear you find the mesh feature useful. On my own printer I have added a auto measure feature, not so much added as I added the sensor in a very simple way. I put it here and maybe there are pieces you can pull from there http://www.thingiverse.com/thing:1120142

brainscan, 2016-01-22:
I've been using mesh levelling and love it, I'm just having some problems using G29 S3 to modify one of the points. I don't have an lcd so can only move in .1 increments so need to modify one of the points but when I try using S3 it always states that X is out of range. Any ideas? Also sometimes mesh levelling deactivates even though it still has the points in eeprom. 
Thanks for the excellent feature and any help you can give.

epatel, 2016-01-22:
Ah, `S3`. Maybe I was a bit unclear about the `X` and `Y` parameters in the commit that got pulled into the project. X and Y should be indexes, and not the distances, i.e `X1 Y1` is for the point first/closes to origin. I once wrote a doc for mesh usage but that was removed. Can read it here still https://github.com/MarlinFirmware/MarlinDev/blob/9a093b2e8f/Documentation/MeshBedLeveling.md
I also tried to update the descriptive comment a while back but never got pulled into the repo.

Poquaruse, 2016-01-22:
Hi, thanks for your very fast answers. :)

One side note: you should definitely get an inductive / a capacitive sensor. It's worth the 10 bucks and you can leave it attached permanently... (For me, it replaces the z-endstop entirely.)

I've had a look at your source code. I understand parts of it but I think I need more time to figure out how to include it. From what I understood I think it should not be too hard to use another sensor attached to z-min and to perform mesh ABL in a fully automatic way. However, as I said I'll need plenty more time...

Nonetheless, I'd be very grateful if you someday found the time and motivation to create a pull request for our beloved Marlin. :+1: (I'd even be willing to beta test ;) )

Best regards

brainscan, 2016-01-22:
Thank you so much, you're a star!!! That's definitely where I'm going wrong. I've been trying to input the X and Y positions, makes more sense that it's an index. Thanks again. 

Lee

brainscan, 2016-01-22:
> Nonetheless, I'd be very grateful if you someday found the time and motivation to create a pull request for our beloved Marlin.  (I'd even be willing to beta test ;) )

If using a capacitance or inductive sensor wouldn't you still need to have the probe offsets included in the mesh levelling code?

Roxy-3DPrintBoard, 2016-01-22:
Yeah, this Mesh stuff is definitely valuable.  Right now, I'm doing a Delta version.   

@epatel   What is the reason you tell people to keep the grid down to 7 x 7?  Is that because you could not save the Mesh in EEPROM if you made it bigger than that?   The reason that is the case is because:
~~~~
void _EEPROM_writeData(int &pos, uint8_t* value, uint8_t size) 
void _EEPROM_readData(int &pos, uint8_t* value, uint8_t size) 
~~~~
have size defined as uint8_t.   Changing that to unsigned int fixes that issue if you wanted to go bigger.    I'm thinking I want to let people go to 15 x 15 or maybe even 20 x 20.

dushyantahuja, 2016-01-22:
Hi Edward,

Is there a pull request for the fully auto mesh bed leveling that you have
in your demo. Or is it already incorporated in Marlin?

I'm thinking of using it on my cobblebot with an induction sensor.

Thanks.

Dushyant Ahuja

epatel99, 2016-01-23:
@Roxy-3DPrintBoard Actually the 7x7 limit comes from the `mesh_plan_buffer_line()` parameters. To not having to rely on flawless floating point math for calculating and splitting on mesh grid boundaries that function uses two bitfields, currently 8 bits. Increasing them (uint8 -> uint16) could increase the number of points. But my gut feeling when implementing it was that a grid square say 7cmx7cm should be pretty flat, and that would give a total area of 49x49 cm...and, take quite some time to measure too.

epatel99, 2016-01-23:
@dushyantahuja Sorry, no pull request for the auto probing parts. Have been thinking about it, but since I did that the MarlinDev repo has gone through some major changes and I don't have time to get that working in that repo, yet. There was some questions about the nature of homing when using mesh levelling and I added some requested behaviour, made a PR but it never made it into the repo. I want something that works and when I looked at the MarlinDev repo (a while back) I just gave up. You are very welcome to grab any piece you want from my repo.

Roxy-3DPrintBoard, 2016-01-23:
> Maybe 8x8 points works too, haven't really tested it. I wrote 7x7 just to make sure to be on the safe side. 8x8 could maybe hit some bitfield problem due to the nature of the recursive nature of the function. Better be safe.

I can promise you, 8 x 8 does not work.  At least not if you want to save it in the EEPROM.   The write function uses an unsigned short to specify the size of the write.  That means it goes from 0 to 255.    8 x 8 is 64 and a float is 4 bytes big.  That cranks out to 256 bytes of storage.  I changed that unsigned short to an unsigned int.  But you could also go in a loop and write one mesh point at a time.  That would get around the limitation.

I doubt most people will need 20 x 20 unless they have a really uncalibrated machine with lots of defects.   But I'm pretty sure I can get that to go.   (I'm going to have to compress some of the data to get everything to fit.)

>  wrote 7x7 just to make sure to be on the safe side. 8x8 could maybe hit some bitfield problem due to the nature of the recursive nature of the function. 

I made some changes to do_blocking_move_to() so Delta's would not sag when that function was called.  If the move was too big, I would recursively call the do_blocking_move_to() function with smaller moves.  I thought I was the first example of recursion in Marlin.  It turns out your recursion was in there first!   :)    Good for you!!!!!

> There was some questions about the nature of homing when using mesh levelling and I added some requested behaviour,

The Delta version is going to home towards the bed and set zero where it finds it.   (That is important for me because I switch pieces of glass that I print on.)   I also have some other goodies planed that I am pretty sure people are going appreciate.   Right now, there are only a couple of things that I plan on doing that are not fully compatible with Cartesian machines.  But I'm going to try to write the code so it is tolerant of both machine types.  It maybe we can merge the two flavors and keep the best of each.   I'm doing a full rewrite.  But even @epatel99 is going to recognize it as 'borrowing heavily' from his work.   

On the Delta's, I think I'll be able to compensate for a Delta Radius being off by +/- 1.00 mm pretty easily.  (Even at the edge of the DELTA_PRINTABLE_RADIUS).  The trick will be to make it so somebody can get their mesh fully calibrated within 20 or 30 minutes.   But like I said, I have some goodies planned!

KiteLab, 2016-01-23:
@Roxy-3DPrintBoard 
>The Delta version is going to home towards the bed and set zero where it finds it.

I suppose you want to home x/y upward and correct z downward. Or did you find a completely new trick?

Roxy-3DPrintBoard, 2016-01-23:
> I suppose you want to home x/y upward and correct z downward. Or did you find a completely new trick?

No. It is as you say.   It will home upward and then 'Home' down until it finds the bed.   And I might be able to add a 3 point leveling to the bed independent of the higher resolution Mesh.   But I guess I'll leave that until the very end.

Most of the work is going to be getting the code setup to very quickly and easily set the Mesh exactly right.

adamfilip, 2016-02-03:
I would like to thank you all for your dedication to Marlin and specifically to work on Auto Mesh Leveling. I cant wait to try a release with it ready to go.

jfpion, 2016-02-29:
I concur that an fully automated mesh leveling would be great
why not the same than the present autolevel

call via g29 , after an x/y homing , probing the bed and the go to print ? no fuss no complication with eeprom (i really don't think it is a good idea anyway, because the main use of it is to be able to swap the glass sheet to make prints fast and you need to redoo the probing. and even without change, you never be sure youre bed didn't move when you scrap the precedent part)

thank you for your job

SeanTasker, 2016-03-07:
Nothing new to add but I just wanted to mention that this is a feature I would like to see included.

I was originally under the impression that the `AUTO_BED_LEVELING_FEATURE` with `AUTO_BED_LEVELING_GRID` enabled applied distortion compensation for each grid square but it doesn't. It isn't made clear in configuration.h.

Are there any updates on the status of this?

thinkyhead, 2016-03-16:
In the long run I would like to have both Manual and Automatic options available for all three forms of bed leveling: Mesh, Three-point, and Least-squares. This might mean overhauling all the leveling code a bit.

Roxy-3DPrintBoard, 2016-03-16:
> In the long run I would like to have both Manual and Automatic options available for all three forms of bed leveling: Mesh, Three-point, and Least-squares. This might mean overhauling all the leveling code a bit.

It the Automatic Mesh Leveling System for Delta's that I am doing, the first phase is the automatic collection of probe data for the area within DELTA_PROBABLE_RADIUS.   For areas of the bed outside this limit, a manual phase is required (if that portion of the bed is going to be used).   So, in the case of Deltas, and this code is being written to move easily to Cartesian machines, both Automatic and Manual will co-exist just fine.  

As for the 3-Point and Grid based leveling, I plan to add that to the Automatic Mesh Leveling System as a follow up.   If I get that far, the code will be able to turn on or off any sub-feature.  You just pick what you want and build the firmware.

jfpion, 2016-03-16:
great news, I will use it on my big printer as it is diffcult to have a 40*30 cm glass perfectly flat (it bend a litlle under constraint)

thank you

epatel, 2016-03-16:
@jfpion RigidBot BIG? I wrote this specially for my "bad" rigidbot big https://github.com/MarlinFirmware/Marlin/pull/1619

jfpion, 2016-03-16:
for this one :[http://www.thingiverse.com/thing:27049](url)   the thingiverse page need some update

lrpirlet, 2016-03-17:
I have a sort of an "auto mesh bed leveling"... I called it motorized mesh leveling... It is based a lot on @epatel work.

It is working for me, I now can use my "chiken" printer to build a core XY version( depend both on the state of my money and the pc failure did not help a lot...)

I saved it a few days ago on githb when my pc started to mis-behave... I now have a brand new pc and a lot of trouble to restore...

Feel free to use/examine.. 
- it  has a mofied m48 to keep low when no moves ( I use it to level in hardware before using motorized mesh), 
- it  has a few other features I cannot remember... lesson DO NOT use motorized mesh level when hot, I punched/burned a nice cone in my surface mounted switch...)

If/when I have time and resources, I'll try to upload a picture or two of the thing I build for that... (as a added file to the motorized_mesh_bed_leveling branch

see: https://github.com/lrpirlet/Marlin/tree/motorized_mesh_bed_leveling_for_LRP

Enjoy

closed 2016-05-04 by jbrazio