MarlinFirmware/Marlin #2921: babysteps dont work when pidtemp is disabled

Slavulj, 2016-01-22:
ramps1.4 with arduino mega2560

when i disable pidtemp on extruder, it wont compile code, it give error:
'babystep' was not declared in this scope
in temperature.cpp

KiteLab, 2016-01-22:
Add
```
#include "stepper.h"
```
at the top of temperature.cpp.

KiteLab, 2016-01-22:
Or - and here we have the reason why
Replace in temperature.h
```
#if ENABLED(PID_ADD_EXTRUSION_RATE)
  #include "stepper.h"
#endif
```
with
```
#if ENABLED(PID_ADD_EXTRUSION_RATE) || ENABLED(BABYSTEPPING)
  #include "stepper.h"
#endif
```

Wackerbarth, 2016-01-22:
Why the conditional include at all? Unless "stepper.h" actually generates code, (and it should not)  just #include it unconditionally.

KiteLab, 2016-01-22:
@Wackerbarth 
This is not thought as a PR. 
It's just a bit of support and analysing the code.

Do whatever you think is right. You do it anyway.

Wackerbarth, 2016-01-22:
And, likewise, I am just pointing out that your suggestion has no practical value in improving the generated code. But, IMHO, it does then to further obfuscate the source.

KiteLab, 2016-01-22:
@Wackerbarth 
Both of my suggestions let the Original Poster compile Marlin with BABYSTEPPING without PIDTEMP, what is not working with the current code. The first version is exactly what you want. The second version shows exactly what went wrong - PID_ADD_EXTRUSION_RATE is only defined when PIDTEMP is defined, but babystep() needs #include "stepper.h" - so a ` || ENABLED(BABYSTEPPING)` is a pointy fix for exactly this situation.
Both versions are a possible Workaround and are not thought to be more.
You'll have to make your own PR. Mine are, for sure, not good enough for you.

I bet including stepper.h in Marlin.h is to general for you, just because I suggested it now.
.

Wackerbarth, 2016-01-22:
Yes, I would not like to see stepper.h included in Marlin.h because that only encourages hidden side effects. It's not because you suggest it, but because it is a poor programming practice.

KiteLab, 2016-01-22:
Didn't you say:
>Why the conditional include at all? Unless "stepper.h" actually generates code, (and it should not) just #include it unconditionally.

As far as i see it is not producing code.
What could be more unconditional?
Is there any meaningful setup around not using steppers?

But I'm tired discussing this.
I prefer to find out why, the hell, the 'temparature interrupt' is sending the steps to the 'stepper interrupt'.

Wackerbarth, 2016-01-23:
> On Jan 22, 2016, at 5:57 PM, KiteLab <notifications@github.com> wrote:
> 
> Didn't you say:
> 
> Why the conditional include at all? Unless "stepper.h" actually generates code, (and it should not) just #include it unconditionally.
> 
> As far as i see it is not producing code.
> What could be more unconditional?
> 
Just
#include “stepper.h”
without any of the
#if ENABLED() conditions surrounding that include.
> Is there any meaningful setup around not using steppers?
> 
No, but there are meaningful partitions of the code which should not “talk” directly with each other.
> I prefer to find out why, the hell, the 'temparature interrupt' is sending the steps to the 'stepper interrupt'.
> 
A good question. (And, as I implied earlier, placing this include in Marlin.h would only help to hide that interaction.)

KiteLab, 2016-01-23:
The need for the inclusion of stepper.h is not obvious.
>#if ENABLED(PID_ADD_EXTRUSION_RATE) || ENABLED(BABYSTEPPING)

could be seen as a kind of documentation.
Everything has its pros and cons. ;-)

Slavulj, 2016-01-23:
Hello, thanks, its solved my problem,
but i got another problem about this,
i see babystepping in menu but i cant adjust it
picture: http://goo.gl/UJumRk
lcd is REPRAP_DISCOUNT_SMART_CONTROLLER

KiteLab, 2016-01-23:
No output is expected here. What you'd see is 0. If you turn the encoder babystepsTodo[axis] is decreased/increase - the temperature interrupt realizes a different from 0 value - sends the steps to the stepper interrupt - increases/decreases babystepsTodo[axis] until it's 0 again. That happens fast enough you would not realize a change on the display.
```
  static void _lcd_babystep(int axis, const char* msg) {
    if (encoderPosition != 0) {
      babystepsTodo[axis] += BABYSTEP_MULTIPLICATOR * (int)encoderPosition;
      encoderPosition = 0;
      lcdDrawUpdate = 1;
    }
    if (lcdDrawUpdate) lcd_implementation_drawedit(msg, "");
    if (LCD_CLICKED) lcd_goto_menu(lcd_tune_menu);
  }
```

The concept of Babystepping is to bypass all registration! Your only feedback is the change in printing. It's designed that way! If you want to know how much you changed the height use a alternate method to change the z-height.

KiteLab, 2016-01-23:
And before you ask BABYSTEP_INVERT_Z is not implemented/used any more. Use a negative BABYSTEP_MULTIPLICATOR instead.

KiteLab, 2016-01-23:
If you have a high steps/mm number you may want to increase BABYSTEP_MULTIPLICATOR to see anything.

Roxy-3DPrintBoard, 2016-01-23:
@KiteLab   Your expertise and willingness to help are much appreciated!

thinkyhead, 2016-02-07:
Let me not pile on, but I'd like to make a PR for the `stepper.h`-missing-symbol bug that makes the most sense for the current and future code-base. I agree with @Wackerbarth that it seems a weird practice to include `stepper.h` only when certain features are enabled, since it doesn't save on any resources. If the reason is: "The temperature code only needs to know about stepper symbols when it uses them," that's not a good technical reason. But I also agree with @KiteLab that these #if blocks often serve as documentation. This one tells us: "The temperature code wants to access variables in `stepper.cpp` with these features." I'm inclined to just add the second condition, because it will also stand as a to-do for the code. It reminds us: "These units are not (yet) well-encapsulated," with the ever-nagging "but maybe they could be" to torment OCD coders like myself.

thinkyhead, 2016-02-07:
Hmm. What's a good C/C++ philosophy on best-use of `#include` and `extern`? Should `extern` be used in lieu of `#include` if all you need is a symbol or two? Should `#include` be used in `.h` files when only the corresponding `.cpp` file needs it (and the header doesn't)? Should `.cpp` files always include `Marlin.h` and should all `.h` files also include that? Does the low-level nature of this kind of embedded-software project impact the use-pattern of `#include`? In other words, does the use of `#include` and `extern` —to gain access to global symbols— in some sense supplant the OOP features of C++ because we haven't used singleton classes to wrap some of these units?

I'm asking all these silly questions because I see so many styles of usage of `extern` and `#include`, and none in particular stands out as totally uncompromising. And maybe that's a weakness of C/C++. We certainly don't have this annoyance in C#, for example….

Blue-Marlin, 2016-02-20:
One of the problems seems to be the .h-files producing actual code. To get a single variable from them a  'extern' seems to be the better choice.

thinkyhead, 2016-02-21:
@Blue-Marlin I'm working on wrapping up planner, stepper, the gcode processor, and other obvious code units into singleton classes to see how it affects those externs. Hopefully the use of `extern` will be less important, as we can just use the access levels of C++ and a small set of top level objects. And now that Arduino supports subfolders we will hopefully get better file organization too. We might even end up with each feature being something more like a "module" with each having its own subfolder. We shall see…

closed 2016-04-23 by jbrazio