MarlinFirmware/Marlin #2922: Z Axis Didn't Stop

sarf2k4, 2016-01-23:
I tried to move the z axis through the menu; prepare > move axis > 1mm > z axis > increase from 0 to 170mm, which is a long travel and had to do several rotation on the encoder to reach to that point. 170mm is my max z axis that I limited on my config.h.

The z axis did pause several times during the travel probably after reaching around 120mm increment, the z axis pause a little bit and continues upward for about 20mm. Then marlin continues the same pattern afterwards ignoring the max z axis that has been set up in the config.h

I believe there's some bug in the move by 1mm on z axis option.

I got this bug since rc2 and currently using rc3. This isn't the first time I encountered such problem, could you guys check out this thing?

Thank you in advance

Roxy-3DPrintBoard, 2016-01-23:
We need more detail.  For starters, what hardware you are running.  And you Configuration.h file.

sarf2k4, 2016-01-23:
I'm using ramps 1.4 with servo, cooling fan auto bed leveling and 2004 lcd enabled on my prusa i3 printer, here is my config.h

```
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

#if ENABLED(USE_AUTOMATIC_VERSIONING)
  #include "_Version.h"
#else
  #include "Default_Version.h"
#endif

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define SHOW_BOOTSCREEN
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2

// @section machine

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
// :[0,1,2,3,4,5,6,7]
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// :[2400,9600,19200,38400,57600,115200,250000]
#define BAUDRATE 250000

[... 784 lines cut ...]
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H
```

sarf2k4, 2016-01-25:
here is the link to the video regarding the problem
https://youtu.be/3X9-qdZA_vM

ntoff, 2016-01-27:
>This video is private.

Change it to "unlisted" if you want others to see it and not show up on your channel

Is your printer homed? If not and Z is not already at it's lowest point when you power it on, it won't know it's not home and think it's already at 0 so when you tell it to move up it'll think it has 170mm worth of travel left in it. If it happens to already be at 150mm when you turn it on, it won't know that unless you home it.

Are you sure your steps/mm are correct? Perhaps it's moving physically twice as far as it thinks it is, so when the printer thinks it's moved only 50mm it's actually moved 100.

sarf2k4, 2016-01-27:
I'm sorry, I didn't notice the video was private, I changed it to public

In the video, yes, after homed. The steps value are correct, I've been printing using 2560 steps/mm for z axis using a4988 stepstick only 1/16 microstepping (I think).

thinkyhead, 2016-03-11:
@sarf2k4 This should be all fixed now in `RCBugFix`. We had a stack corruption issue associated with axis movement from the LCD. Please give the latest code a try and let us know if it's still an issue.

sarf2k4, 2016-03-14:
@thinkyhead 
I did a preliminary test before using the rcbugfix branch and using the 1mm increment on z axis would result marlin to crash/reset automatically when large travel imposed for example after homed, z at 0 moving straight to 170, marlin would crash

As for now in rcbugfix, i dont see this particular problem yet, will report back if similar thing came up

jbrazio, 2016-03-14:
Great, thanks for testing.

CONSULitAS, 2016-03-15:
@sarf2k4 If your issue is resolved, you please close it for housekeeping. Thanks.
You can reopen it at any time if needed.

closed 2016-03-16 by sarf2k4