MarlinFirmware/Marlin #2923: How to make driver of E1 for motor Z2?

releyshic, 2016-01-23:
Переводчик
Actively build 3D printer from old inkjet printers )
I have a Chinese Arduino Mega and RAMPS 1.4, DRV8825 drivers (purple, or notebook which), the firmware is Marlin
In test mode all twisted (SM) works
But the main problem is the following -> I have on the z-axis of SM c two gears of different number of steps and straps of different pitch, those physically not able to connect two of these SM's on a single driver the z axis.
RAMPS 1.4 Board has 3 channels for stepper motor 3 axes X, Y, Z and two extruders for E0 and E1. The extruder I have one, so it remains a free driver E2. To it just I want to connect a second stepper motor Z-axis different from the first parameters.
But here is just Zapara - motor control of the extruder differs from the management of SM axis, and plus and myself configuring Marlin to work with 2 extruders are not simple and little information.
The settings of the axes are stored in the header Configuration.h
#define DEFAULT_AXIS_STEPS_PER_UNIT {80,(100*1)/(1.05*40),(100*1)/(1.25*20),(3200 * 39.0)/(11.0 * 6.75 * 3.45)}// X Y Z E0 E1 100 - the number of steps per revolution stepper motor, 1.05 - step belt, 40 is the number of gear teeth 
Each Driver receives 4 parameters: MS1, MS2 - microstepping mode(1, 1/2, 1/4,1/8,1/16,1/32); STEP, DIR as I understood - step (pulse) and direction of rotation. However, I do not understand why the code is MS1, MS2 although in the diagram to the controller is not connected
I guess the easiest option would be to find where in the code the firmware sends commands to the outputs of Arduini -> driver-E1 and write down the values for Z, but given its parameters, naturally there will be several edits on the way from DEFAULT_AXIS_STEPS_PER_UNIT to the programming outputs of the mega.
Here's what I can't find where it is set to output mega for STEP, DIR
Well, maybe somebody is already doing? or well versed in the code?

sarf2k4, 2016-01-23:
you probably wanna try to look into configuration_advance.h area, I remember there's an option for dual motor mode for each axis but not sure about using assigned pins such as your case. It should be doable

KiteLab, 2016-01-23:
stepper.cpp, and several other places in the code, can't handle more than 4 independent axes.
When driving 2 z-steppers with different timings no axis for E would be left.
The double stepper configurations in Marlin do work because, the timing for the doubled steppers is identical - they are not independent.
Extending stepper.cpp to 5, or n, axis would be the first code to extend. But the stepper interrupt is already to large and slow to drive really fast. Extending it, will make things worse. At least for the AVR processors i see troubles. Try it. Good luck.

KiteLab, 2016-01-23:
Adjusting the micro stepps mode on the fly does not work with RAMPS, but with other boards. It requires special hardware.

releyshic, 2016-01-23:
"Adjusting the micro stepps mode on the fly does not work with RAMPS, but with some other boards. It requires special hardware. With RAMPS you have to set jumpers." the micro stepps mode dont need.
Marlin get from #define DEFAULT_AXIS_STEPS_PER_UNIT Configuration for all Step motor. need just to make  control SM of E1 like SM for axis and edit #define DEFAULT_AXIS_STEPS_PER_UNIT for E1
but this "to make  control SM of E1 like SM for axis " not easy without very good programming experience

releyshic, 2016-01-23:
how to commente specific user? or its impossible?

releyshic, 2016-01-23:
"sarf2k4 commented 4 hours ago
you probably wanna try to look into configuration_advance.h area, I remember there's an option for dual motor mode for each axis but not sure about using assigned pins such as your case. It should be doable"
"doable" thanks fo new word/ never dont heard =))
"option for dual motor mode for each axis" do you have description for this option? or better - work project with this, or more better - all inclusive =))

releyshic, 2016-01-23:
my mail releyshic@mail.ru

KiteLab, 2016-01-23:
DEFAULT_AXIS_STEPS_PER_UNIT can only be defined for 4 independent axis.
You can't use extruders with different steps/mm for exactly that reason.
There are extensions in Marlin to set steps/mm on the fly (M92). That enables a possibility to use extruder motors with different steps/mm - but not at the same time.
For a second z stepper with different steps/mm this is absolutely unusable!

releyshic, 2016-01-23:
"You can't use extruders with different steps/mm for exactly that reason." i dont want to use extruders, i want use driver of second extruder

thinkyhead, 2016-03-11:
Hi @releyshic – Sorry it has taken so long for me to get to your question.

Marlin now has this automated, so you don't need to do anything special. So far this code has been added to `RCBugFix` and it will be in `RC4`, but it is not in `RC3`.

So here's how it works. If your board defines additional extruder pins and you enable either `Z_DUAL_STEPPER_DRIVERS` or `Y_DUAL_STEPPER_DRIVERS` the next unused extruder pins will be automatically assigned to your `Z2` or `Y2` steppers.

So, for example, on a board that has ports for `E0` and `E1`, simply attach your `Z2` stepper to the `E1` connector. Set `#define EXTRUDERS 1` and enable `Z_DUAL_STEPPER_DRIVERS`.

I hope this helps!

sarf2k4, 2016-03-11:
@thinkyhead, does that mean the dual motor mode that currently has only accepts single steps/mm?

If the dual mode already exist, how can I enable and will it be the same like the one featured on the upcoming rc4; auto assigned?

I was thinking of using this feature to do z axis leveling using switches so that the x axis are quite flat/leveled on each side

releyshic, 2016-03-11:
@thinkyhead actually, i want to set different  numbers of steps and straps for each motors on axis Z (i have two motor on axis Z) for example in this place
`#define DEFAULT_AXIS_STEPS_PER_UNIT {80,(1001)/(1.0540),(1001)/(1.2520),(3200 * 39.0)/(11.0 * 6.75 * 3.45)}// X Y Z E0`
but i dont know how to do it, because this define dont have parametres for E1 (for second motor in my situation)
Do you developer of Marlin?

thinkyhead, 2016-03-13:
Currently the solution specifically for dual steppers operates by simply sending the same step signals to both motors. This is optimal for most cases.

So far proportional movement has not been implemented as a general feature because there hasn't been a high enough demand. Nor has a general option been added to specify different steps/mm for multiple extruder motors. These features could be patched in, but they are somewhat non-trivial and might require some thought on design, for example choosing a method to specify the steps/mm for additional stepper axes.

thinkyhead, 2016-03-19:
We have a general feature request to bring in these kinds of patches – some of which are done cleanly in other forks. So let's close this topic since #2988 has now been merged and will soon be released with 1.1.0-RC4.

closed 2016-03-19 by thinkyhead