MarlinFirmware/Marlin #2928: Connection lost / driver Issues

TheNuubi, 2016-01-25:
Hi,

I am having problems with my printer losing connection after (succesfully) printing the first layer.
This is probably not the right place for this issue but i don´t really know a better forum.
it is not the firmware (got the same problem with sprinter) and not the host. The thing is that everything was fine until it wasn´t anymore :/ I read a lot about this and the go to answer seems to be electromagnetic noise. however since the print stopps at nearly the same position without any other problems i think its software. i have the chinese arduino clone (mega 2560) with the CH 340 driver.

Sorry if this is unrelated/ bothering but i am really pissed and desperate :)

brainscan, 2016-01-25:
I might be completely wrong but do you have a heated bed? Do you have heat sinks on mosfets and a fan to cool them? Just wondering if they're tripping the thermal fuse? I think I read about that happening to someone but it was a long time ago and I might be remembering incorrectly. 

Does it disconnect from the host or does the printer reset but stay connected? Sorry for all the questions but the more info the better.

KiteLab, 2016-01-25:
Take a cloth look at the output (terminal window). Maybe there is a kill message. The output directly before that could give a hint why Marlin is unresponsive.

ntoff, 2016-01-27:
>however since the print stopps at nearly the same position without any other problems i think its software.

Do you re-slice it every time or use the same code? It could have a slicing error.

Try a different usb cable

Try printing from SD card if you can

Make sure you turn off any and all power saving features of your OS, it sounds like there's a good chance it's turning off your USB ports to save power

As for where to look for help, see if there's a google group for your particular printer or try https://www.reddit.com/r/3Dprinting/new/ and http://forums.reprap.org/ for more general stuff.

TheNuubi, 2016-02-06:
Hey,
thanks for the feedback. I don´t have a heated bed and I installed a big fan over the board as the error occured. However that was not the problem (like i said it worked fine before). Nozzle temperature looks good too. I suspect it stopps after the 60 seconds interval for lost connection that is set in repetier. sliced it with cura- same error. i will try my old laptop today (win xp) maybe that helps with driver issues. And yeah i looked through this forums and it seems like its not a common error :). I´ll probably buy a legit arduino like i had for years (switched to the chinese as i used it elsewhere).  And again thanks for the help

TheNuubi, 2016-02-06:
PS. tried with different software (sprinter and pronterface), similar error.

TheNuubi, 2016-02-06:
Ok.
It is printing right now and seems to be fine. i switched the thermistor and got a cleaner temperature curve. So it was either the temperature or windows 10 stuff. I actually tend to believe its temperature, but why didn´t it give a " cold extrusion prevented" or something like this back ?

Anyway, thanks for the help. I´m just happy i can print again :)

thinkyhead, 2016-03-11:
Also we suggest in most cases to stick with 115,200 baud connection instead of 250,000 baud. Personally, I will only print from SD after having so many USB connection issues.

closed 2016-03-11 by thinkyhead