MarlinFirmware/Marlin #2933: M600

alexkaps, 2016-01-26:
I'm having the same issue as 	user antonkopa in [#2926](https://github.com/MarlinFirmware/Marlin/pull/2926):
> I decided to test out the "Change Filament" feature, and before it continues the print, it pulls the filament completely back out again.
>When I commented out #define AUTO_FILAMENT_CHANGE, it doesn't have this issue.
>The issue here I believe is the current position of the extruder is not being restored from the remembered position.

I'm using Marlin RC3 on a daVinci 1.0 modded with mega2560/ramps/e3dV6.  Commenting the  #define AUTO_FILAMENT_CHANGE  seems to solve the retraction causing the filament to completely get out of the extruder.  It looks like E tries to go to it's previous position and reverses all the extrusion made before pressing the button on LCD.

Roxy-3DPrintBoard, 2016-01-26:
Question for Collaborators:   Should we remove the AUTO_FILAMENT_CHANGE code from RC-3 ?  It seems to be causing more confusion and problems than it is worth.

alexkaps, 2016-01-26:
In fact, a fix could be to exclude the Extruder from going back to his pre-pause position or ignore all the AUTO_FILAMENT_CHANGE translation.

KiteLab, 2016-01-26:
My guess is that 
```
    current_position[E_AXIS] = destination[E_AXIS]; //the long retract of L is compensated by manual filament feeding
    plan_set_e_position(current_position[E_AXIS]);
```
is done to early. Should be behind the return to lastpos[]. Else the host sends a next (absolute) E position potentially far far away.

alexkaps, 2016-01-26:
Thank you KiteLab! I'm running a large print (over 12hours) so i have to re-upload Marlin and check your solution tomorrow. 
I just changed the filament using the lcd command  (tune->change filament) & having commended the AUTO_FILAMENT_CHANGE in the firmware. After resuming I noticed a small offset in the z axis (approx. a layer size 0.2mm).  It was noticeable in the perimeters of the print (disconnected thin lines instead of a solid line).  
Do you think it has something to do with the Z destination or it's just caused by me while pushing the new filament in the extruder?

alexkaps, 2016-01-27:
1. Commenting:
>//destination[E_AXIS] = lastpos[E_AXIS];
>//line_to_destination();

seems to work properly. Motor E is feeding/extruding until Lcd button is pressed and then resumes printing without reversing all the extrusion made.  

2. I also noticed that sometimes due to the fast retraction when:
>FILAMENTCHANGE_FINALRETRACT -100

a piece of filament is breaking  in  the extruder resulting in a clog. This retraction speed is controlled by the firmware or it's the retraction speed from the host?

3. Is it possible to add a home X and home Y command before returning to last position in order to eliminate any offset made by mistake? Also Z axis practically missed a layer after resuming, had to manually raise the bed.

KiteLab, 2016-01-27:
As far i can see the feedrate for all the moves is untouched in M600. That means it works with the last set feedrate, Nerveless the maximum feedrate should be limited by`DEFAULT_MAX_FEEDRATE`.

Without rewriting too much, homing x/y would be possible - but would include resetting ABL.
Do the steppers switch off during filament change, for your system?

alexkaps, 2016-01-27:
1.  After uploading a new sketch containing the ultralcd.cpp from RCBugFix branch the unloading problem returned, maybe never solved since i checked it just once. Reversing back to the previous ultralcd.cpp file didn't help, checked it many times. 
In fact it's really strange... Sometimes it resumes printing without unloading and others it unloads. :| Also the Lcd button when pressed sometimes working as a resume button and sometimes it goes to the main menu.  (it looks like it depends on how quickly i press it after pausing. Quick press works, slow doesn't)  Just in case it matters,  i'm printing through SD card, actually running dry prints. 

2. When working, after pressing the LCD button the extruder keeps extruding filament and quickly returning to lastpos X & Y, carrying a line of filament.  It needs some time for cleaning and a small amount of retraction (approx 2mm) while traveling to  lastpos X & Y in order to continue smoothly. 
Reducing the DEFAULT_MAX_FEEDRATE kind of fixed the clogging issue but it would be better if this retraction speed was controlled independently in marlin_main .  

3. My steppers stay active and locked during filament change.  

I'm kind of new to this and i can't find the solutions myself but i can follow recommendations, try and error a lot.  
Thank you for your help, running very long prints and changing filament mid-print is crucial.

Roxy-3DPrintBoard, 2016-01-27:
> My steppers stay active and locked during filament change.

The filament extruder stepper is NOT supposed to stay active and locked.   I suspect you didn't mean to imply that.

alexkaps, 2016-01-27:
Only X & Y steppers are locked (didn't check Z). E is acting as expected when in change filament position.  First retracts rapidly (+100mm) and then extrudes in a loop until button pressed.

alexkaps, 2016-01-28:
Today i lost a 6 hour print at the final 20 minutes because of the unloading problem. :(

### Is there any other way of changing filament mid-print when printing from sd card? 
I tried to pause the print +  Move Axis_ 1mm _ E  + resume print, but when resumed the extruder motor reversed the same amount of movement i made and the filament unloaded again. 
Any solution?

thinkyhead, 2016-03-05:
Prusa Research has a page for adding filament change procedure between layers. On that page he describes his rewritten `M600` command:

Technical details

We completely rewrote the M600 gcode support in Marlin (available on prusa3d.com) to be _actually usable_. When M600 is triggered either from LCD panel or from a pre-prepared gcode file the printer will:

- Stop moving and retract
- Raise the Z by 2mm and move quickly outside the printbed
- Unload the current filament
- LCD panel will ask you to insert the new filament and confirm when it’s done
- After the initial loading of the filament you will get three options on the LCD:
  1. Everything is correct and printing may continue
  2. Filament didn’t load, try again
  3. Extrude a bit more if the color is still tinted by the previous one
- After confirming, printer returns to the original position and continues to print.

Presumable the code he refers to is the code posted here: https://github.com/prusa3d/Prusa-Firmware/blob/master/Firmware/Marlin_main.cpp#L3885-L4095

This could be integrated for LCD-based printing.

Host-based printing will need some extra help to be complete, because recently it's been decided that `M600` and other commands that wait for user input (or just take a very long time to complete) ought to send a "paused for input" message (or a general "keepalive" message) to the host every couple of seconds.

The soonest I will be able to work on these things is sometime next week.

jbrazio, 2016-04-27:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-27 by jbrazio

sniperxnl, 2017-02-28:
https://youtu.be/-cBtDSoMrvU

i used the m600 code and have thia problem

thinkyhead, 2017-03-01:
@sniperxnl Are you using the current `RCBugFix` branch?

sniperxnl, 2017-03-01:
@thinkyhead No,  where can i find it?  i got this printer t from a friend.

thinkyhead, 2017-03-03:
@sniperxnl https://github.com/MarlinFirmware/Marlin/tree/RCBugFix