MarlinFirmware/Marlin #2934: Selecting "tune", "nozzle" from the display during a print incorrectly displays "Flow: 100"

thatguybry, 2016-01-27:
On Marlin RC3 - during a print when you attempt to tune the nozzle temp by selecting "Tune", "Nozzle" it displays "Flow"  "100" prompting you to tune the flow rate.  I am able to consistently reproduce this issue.

I'm using a RAMPS1.4 with the 2004 display such as: http://www.amazon.com/Display-Controller-Adapter-RAMPS1-4-Printer/dp/B00JPVESXA

Attached is my configuration.h which is the only file I've changed in RC3.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/105944/Configuration.zip)

alexkaps, 2016-01-27:
Check Fix LCD Tune Menu Bug #2823.
Replace the Marlin/ultralcd.cpp.  I have the same issue and test it when my long print is over.

KiteLab, 2016-01-27:
Dear Sir's.
Would you please be so kind to read a bit in this bug-tracker.
This is at least issue nr. 3 for the same already fixed problem.

Fixed in the RCBugFix branch..

build3dparts, 2016-02-22:
I applied the #2823 fix and I still see the issue reported here.

build3dparts, 2016-02-23:
My bad ... #2823 works great.

thinkyhead, 2016-02-28:
RCBugFix has a few things patched up that were overlooked in RC3, and it's a good place to grab the most-patched code as we fix things. After a few more patches we'll put it out as RC4, hopefully the last pre-release version with all known issues resolved.

closed 2016-02-28 by thinkyhead