MarlinFirmware/Marlin #2935: FYI - Babystepping on CoreXY machines

ruggb, 2016-01-27:
Babystepping for Z works on my CoreXY machine. Apparently, contrary to information I rcvd here it has always worked for CoreXY machines at least in some versions of 1.0.x. I never tried it as I didn't know how to make it work. I finally implemented it in RC3, then found out from someone else using 1.0.x that it works for them also. - It requires commenting out the define for BABYSTEPPING_XY when BABYSTEPPING is uncommented. BABYSTEPPING_XY is what does not work for CoreXY.

KiteLab, 2016-01-27:
Think about it.
If one __stepper__, on a core-system makes one step, you get a diagonal move. Is that what you want? I don't think so.

ruggb, 2016-01-27:
@KiteLab 
Yes, think about it - then re-read my whole post.
It does not say it works for XY.

Wurstnase, 2016-01-28:
Right, I saw this some month ago and pushed that change. We could undef Babystepping_xy when corexy is active.

KiteLab, 2016-01-28:
In my opinion these existing errors:
```
#if ENABLED(BABYSTEPPING)
  #if ENABLED(COREXY) && ENABLED(BABYSTEP_XY)
    #error BABYSTEPPING only implemented for Z axis on CoreXY.
  #endif
  #if ENABLED(SCARA)
    #error BABYSTEPPING is not implemented for SCARA yet.
  #endif
  #if ENABLED(DELTA) && ENABLED(BABYSTEP_XY)
    #error BABYSTEPPING only implemented for Z axis on deltabots.
  #endif
#endif
```
are better than ignoring the config and changing silently. Maybe the text can be improved by calling BABYSTEP_XY by name.

Wurstnase, 2016-01-28:
Or in the config:
`#if ENABLED(CARTESIAN)
  #define BABYSETP_XY
#endif`

ruggb, 2016-01-28:
Maybe........from a user standpoint, it would be better to have BABYSTEPPING_Z and BABYSTEPPING_XY as separate defines and not BABYSTEPPING.
Then the comments - and future DOC file - would just list those 2 and info RE printers.
Personally, I haven't figured out a good reason to have XY, at least not for a CoreXY or cartesian.

undefining BABYSTEPPING_XY if CoreXY is defined and outputting a message as opposed to an #error might be good to - but I guess one could argue either way for that.

I have no idea what CoreXZ is, but I am thinking there must be a situation with that also.

Enabling babystepping_XY in code is probably not a good idea - and I don't remember seeing "CARTESIAN" as a parameter - So it would be adding complexity that isn't needed.

I can see having BABYSTEPPING_Z as the default.
Then the only define would be for BABYSTEPPING_XY - which is likely not to be used at all.

I can also see a reason to implement BABYSTEPPING_Z full time and not just while printing.
It would be nice to be able to test it without running a print.
It would be nice to use to fine tune the zero point. In which case it would have to remember that and not home to the endstop during that session. Disabling the Z stepper would reset it.

Also, having the menu access as a first item would be good, since it is something one would like to access quickly.

Wackerbarth, 2016-01-28:
"undefining BABYSTEPPING_XY if CoreXY" -- undefining ANYTHING is not a good solution. If something isn't always true, then it should never have been defined in the first place, Splitting BABYSTEPPING into component parts is probably more logical. (And it might even be a good idea to change the name to avoid "legacy" confusion)

Wurstnase, 2016-01-28:
Or just comment out BABYSTEP_XY.
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration_adv.h#L363

If someone need it, (i guess noone need it) can remove the commit.

ruggb, 2016-01-28:
I would go with this.........no need for #IF ........

// Babystep_Z enables the user to control the Z-axis in tiny amounts, independently from the normal printing process
// Babystep_XY enables the user to control the X&Y axis in tiny amounts,independently from the normal printing process. This is not implemented for CoreXY, or Deltabots
// NEITHER FUNCTION RESPECTS ENDSTOPS!!!

#define BABYSTEP_Z       // this is enabled by default.
//#define BABYSTEP_XY  // not implemented for CoreXY or Deltabots!
#define BABYSTEP_INVERT_Z false  // true for inverse movements in Z
#define BABYSTEP_MULTIPLICATOR 1 // faster movements

...................
Leave the #error message

I have also moved my menu item to the top of my TUNE menu - even above MAIN
Now to figure out how to make it stay on that item longer than 10 sec.
any tips???

closed 2016-03-10 by ruggb