MarlinFirmware/Marlin #2936: Display filename when printing from SD/LCD

albertolg, 2016-01-27:
Hello, is possible to display in the LCD's row status the filename that is printing without modifing the gcode? (I'm aware that with the M117 you can show anything on the display, but I want to show it when I select the file from the LCD and begin the print, and my printer is not connected to a PC). Thanks

KiteLab, 2016-01-27:
Please explain a bit more.
What do you want to display?

albertolg, 2016-01-27:
Marlin 1.1.0RC3
I select the file to print on the SD, the message on the LCD is "Heating...", then "Heating done." and the printer begins to print... but "Heating done" remain forever, I wanted to show the filename I'm printing, the filename appear just for a fraction of second before the Heating message...

albertolg, 2016-02-01:
I modified the code commenting the 2 lines in the Marlin_main.cpp because I don't need the message heating and heating done but the name of the file is more important for me

LCD_MESSAGEPGM(MSG_HEATING);

LCD_MESSAGEPGM(MSG_HEATING_COMPLETE);

bradanlane, 2016-02-01:
It sounds like you found a good solution for your situation.

This is likely one of those "personal preference" features. Since M190 and M109 are synchronous operations, having visual feedback is valueble for some users. For others, knowing the print job name has more meaning. Other users may want heating time or ETE.  There is no universal answer.

thinkyhead, 2016-03-11:
I also wouldn't mind seeing the filename I'm printing in the status line. Perhaps instead of making this a standard behavior, we could extend `M117` to display it. Something like `M117 %filename%` perhaps.

jbrazio, 2016-04-27:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-27 by jbrazio

ianobre, 2017-11-21:
@albertolg i reopen your request in new release, if you'd like to add any comment i will be happy :-)

https://github.com/MarlinFirmware/Marlin/issues/8515