MarlinFirmware/Marlin #2937: Need help

MainframeOSX, 2016-01-27:
I am working sience 3 Weeks on my 3D Printer K8200 with ramp 1.4.
The Problem is, i am using a 12DC PC power source, the power is now lower
for the initial heating.

I got always the error:

Heating failed:......

If i am use the PID Autotune i had the same Problem. After i have change the looptime in the pid_autotune to 10 Minutes it work, but the inital heating time is to short, how can i make it longer.

ntoff, 2016-01-28:
I don't know if it's the same for all k8200's but the nozzle heater cartridge on mine was absolutely useless. It is too underpowered on 12v.

CONSULitAS, 2016-01-29:
The K8200 uses a 15V DC power source.

So if you use only 12V, the power of the hotend decreases by factor 1,56 ( sqr(15V/12V) ). This will make your heating significant slower and than

```
#define THERMAL_RUNAWAY_PROTECTION_PERIOD 40 //in seconds
#define THERMAL_RUNAWAY_PROTECTION_HYSTERESIS 4 // in degree Celsius
#define THERMAL_RUNAWAY_PROTECTION_BED_PERIOD 120 //in seconds
#define THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS 4 // in degree Celsius
#define WATCH_TEMP_PERIOD 40000 //40 seconds
#define WATCH_TEMP_INCREASE 10  //Heat up at least 10 degree in 20 seconds
```

thermal runaway protection could catch you.

You could tune the parameters in config but it would be better to use a higher voltage or a different heater (with lower resistance -> higher power).

And keep in mind: 

Your power source should have enough current to drive electronics, motors, nozzle heaters and bed heaters (+10% to +30%) simultaneously or your voltage will drop significantly if a heater is switched on.

I would recommend to measure the voltage while switching on component by component. If it drops, you have a detuned pid and slower heating...

thinkyhead, 2016-03-11:
If this is still a problem, and the software seems to be unhelpful, please re-open.

closed 2016-03-11 by thinkyhead