MarlinFirmware/Marlin #2938: Custom thermistor tables

bradanlane, 2016-01-27:
Would it be sufficiently valuable to support a user-defined thermistor table?

I ran into an interesting situation while calibrating a delta printer. While the thermistor in the hotend was a standard Semitec 104GT-2, it was not behaving 100% the same. Testing the thermistor (and several others) resulting in a gently different curve. It was enough to have the hotend over reporting - aka, the hotend was colder than actual. 

There are a number of thermistor table calculators on the Internet. After using one to generate an custom table, the printer works as expected.

The problem is that the only place for a user to load a custom table is by modifying the `thermostortables.h` file. 

Ideally, users would keep their settings confined to `configuration.h`

Is this a one-off problem or worth considering?

Roxy-3DPrintBoard, 2016-01-27:
I believe it is a one-off problem.   But let's see what others think.

brainscan, 2016-01-27:
I had to make my own tables when I made my printer and it was a pain. Easier now I know what to do but it took me a while to figure it all out.

Roxy-3DPrintBoard, 2016-01-28:
@brainscan   Didn't you use off the shelf thermistors?  Couldn't you just pick a Termistor number and have things pretty much work out?

brainscan, 2016-01-28:
Not at the time no, Marlin had fewer tables and none of them worked with my thermistors. There didn't used to be so many places to get printer parts so I think it it used to be more common to make a table to fit whatever could be found. Now as most people buy their printer pre built or as a kit they don't have this problem.

bradanlane, 2016-01-28:
In my example, the performance of the off-the-shelf thermistors did not match the manufacturers' data sheet. The actual test data is the red line in the following graph. You can see that at room temperature, the thermistors were close to spec. However, when the thermistor reported 200C, the actual temperature was less. 
[image]

Roxy-3DPrintBoard, 2016-01-28:
Just for the sake of discussion, What would be the 'Right' way to compensate for this in the firmware?  Would it be acceptable to just have an offset that always gets added to number?   I realize the curve is not linear, but really, we only care about the temperature of the bed or the nozzle at one place in the curve.

bradanlane, 2016-01-29:
Providing an "adjuster" will work in some situations however I suspect we would be better served with a single solution for users who have a broader set of needs.

An option that will fit for more users would be an optional `#include thermistortables_user.h` If a user sets their thermistor to 9999, the optional file is included within `thermistortables.h`

It's not ideal but workable.

Entr04y, 2016-02-01:
I'm using RC3 and I also needed a custom table using a K-type thermocouple with an AD8495 board, and had to add it to thermistors.h.  I'm not sure that putting a thermistor table in configuration.h is the right way to go.  I'd be more comfortable with the included file that bradanlane suggested.

thinkyhead, 2016-03-16:
As with pins and languages, we ought to be splitting up `thermistortables.h` pretty soon. Then it will be easier to add a custom table in a separate file.

nomukaiki, 2016-05-04:
My [forked code](https://github.com/nomukaiki/Marlin/tree/configurations/KosselXL) from RCBugFix has exactly that additional [thermistortables_custom.h](https://github.com/nomukaiki/Marlin/blob/RC/Marlin/thermistortables_custom.h).
I'm happy to submit similar structure in the RC branch if you are ok with it.

closed 2016-05-04 by jbrazio