MarlinFirmware/Marlin #2944: Problem setting nozzle temperature from LCD

zakkos, 2016-01-29:
Hello all!

I noticed some strange behaviour from my machine when I loaded the then latest RC version (5a54204 from 30/11). 

If I try to set the nozzle temperature from the LCD menus (either from the tune or the control sub-menus) to something more than 210° it reverts automatically to 210. 
It doesn't do that every time but every "other" time! 

I try to explain better: 
I put, let's say, 240° and it goes back down to 210. 
I then go again at 240 and it accepts this value but if I then try again to tune the nozzle temperature, say I raise it to 245, it goes back to 210°!

I tried to check in the code if I could come up with a solution but hadn't any luck.

AnHardt, 2016-01-30:
If you are not on RCBugFix, it's not the same symptom, but the same bug.

zakkos, 2016-02-05:
@AnHardt Sorry for the late reply. Today I could finally check out the RCBugFix branch and indeed you were absolutely right. Different symptoms but same problem, sorry for the inconvenience!

closed 2016-02-05 by zakkos