MarlinFirmware/Marlin #2947: Feature Request: Add a endstop selftest

AnHardt, 2016-01-30:
Add a endstop selftest during startup.
Throw an error if the emdstops at both ends of an axis are triggered at the same time.
If an endsto, not probe, is triggered, back up a few mm to see if its untriggered then - if not, throw an error.

KiteLab, 2016-02-02:
#2965

bradanlane, 2016-02-02:
Thanks for linking to the existing FR. I didn't see that when I did a quick search. Feel free to close this issue as a duplicate.

thinkyhead, 2016-03-16:
I will see about integrating this concept pretty soon. There might be complications to work out, such as which direction to "back off" when you don't know which end you're at.

AnHardt, 2016-03-16:
@thinkyhead 
Have you had a look into https://github.com/AnHardt/Marlin/pull/15?
if min- and max- endstops are mounted in parallel to the same pin it's cought by a construct like
```
+   #if HAS_X_MIN && HAS_X_MAX
+      if ((READ(X_MIN_PIN)^X_MIN_ENDSTOP_INVERTING) && (READ(X_MAX_PIN)^X_MAX_ENDSTOP_INVERTING)) {
+        endstop_test_error('X');
+      }
+    #endif
```
before any movement starts.

When all endstops are connected to the same pin, but only one per axis, we don't know what axis to move to free the endsop.

In that (rare?) case, do not activate the test.

Maybe the description (https://github.com/AnHardt/Marlin/pull/15/files#diff-8270513bcf2a548cd418cdfe123f27d3R348) can be further improved.

thinkyhead, 2016-03-18:
I will look more closely at your number fifteen. But of course, after the release. (I sound just like a broken record.)

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better.
You should re-open this topic on [MarlinFirmware/MarlinDev](https://github.com/MarlinFirmware/MarlinDev) for proper followup from the development team. We suggest you to use the `MarlinFirmware/Marlin#<topic id>` markdown to link back to this original topic.

AnHardt, 2016-05-04:
@jbrazio 
Super idea to direct feature requests to MarkinDev the day it is discussed to remove the repository (#3656). Please reopen.

jbrazio, 2016-05-04:
@AnHardt we have to work with the tools in place, issues have being closed for a couple of weeks now, all of them with the same message.

AnHardt, 2016-05-04:
@jbrazio 
But the situation has clanged today (#3656). And now you are touching my FRs, so i have a right to complain.

boelle, 2016-05-04:
should this not be closed with remark that the feature should be submitted as a PR from the OP ?

thinkyhead, 2016-05-05:
I do love this idea. It should be in 1.1.1, at least. In the meantime I will do my usual thing, grab the original branch, rebase onto the latest code, clean it up a bit, and post a PR as a placeholder with "Don't Merge" tagged on it where we can discuss the Exhibit A.

jbrazio, 2016-05-05:
The robot moving around without knowing its position makes my skin crawl..

I would suggest this: When doing a `G28` the endstop trigger values are stored in ERPROM, when booting if the current endstop status is in a triggered state the robot gives a warn to the user that something maybe wrong and ask click to continue.

Because we booting we can be homed and the user clicks to confirm everything is ok.

Every time we `G28` the triggers are saved using `eeprom_update()` instead of `eeprom_write()`, `update()` only writes to the address if the byte being written is different from the one stored.

@AnHardt @thinkyhead what do you think ?

thinkyhead, 2016-05-05:
I tend to favor the simplest approach with the least storage required. That would be an endstop self-test at startup that throws an error message and calls `kill()`. It won't catch every endstop mis-configuration, necessarily. It would have been especially useful to catch instances where `DISABLE_MIN_ENDSTOPS`/`DISABLE_MAX_ENDSTOPS` were wrong, but now that those settings are gone it will mainly catch instances where the INVERTING or PULLUPS settings are wrong.

bradanlane, 2016-05-06:
The simple case I've encountered more than once is having both MIN and MAX Z report "closed". 

This happens wihh the delta printers when a wire breaks on a Z-probe mounted to the effector. With all of the motion of the effector it can be easy to break a wire. This results in the Z-min reporting "closed". G28 will move until Z-max is "closed". Then G29 will attempt to auto-level the bed and both are reporting closed and bad things happen.

thinkyhead, 2016-05-06:
@bradanlane Well with the new configuration, endstop plugs that you're not using will just be ignored and treated as open.

bradanlane, 2016-05-06:
I may not have been clear. With some of the delta printers, the Z-probe wires run right to the effector. Therefore there is a Z ends top for both min and max. If the wire breaks at the effector, then there is a condition where both min and max report "closed" for the Z axis.

thinkyhead, 2016-05-08:
@bradanlane Marlin won't treat the wires connected to "min" as an endstop if you configure it as a probe. But I get your point. If the probe and any max endstop is triggered, that's bad on a delta.

boelle, 2016-11-10:
+1 for this one

thinkyhead, 2016-11-15:
It will certainly help to abort bed probing when the probe is found to be triggered at unexpected times, such as just before starting the probe. I keep running into this in my development on the MakerArm prototype, whose probe is triggered when retracted.

AnHardt, 2016-11-15:
Just use and/or extend '_TRIGGERED_WHEN_STOWED_TEST'

thinkyhead, 2016-11-16:
Maybe. Allow me to describe. This is a case where the probe must be deployed manually. If it hasn't been unscrewed then it sits in a position where an opto endstop is triggered. Thus, when you do `G28 Z` or `G29` it immediately assumes it has hit the bed. In the case of `G29` it keeps raising the Z axis by the 'clearance' distance at every point.

Since this probe can't deploy or stow on its own, I'm not sure the triggered-when-stowed test is appropriate. There's no "deploy" or "stow" for it. And also, the test I'm proposing would apply universally — to every type of probe. If the probe is triggered before the probe move even begins the procedure should abort. So it needs to do a direct test of the appropriate pin.

AnHardt, 2016-11-16:
Prepared a test. #5229 
Maybe kill() is a bit to hard reaction? 
It's a serious configuration error it the probe is triggered then.

thinkyhead, 2016-11-17:
@AnHardt Oh, gosh, no, I don't want to see a `kill()`, but `G29` should abort.

boelle, 2017-04-26:
@thinkyhead maybe put this on the 1.1.1 milestone?

seems like a good feature

slowbro, 2021-07-03:
Is this solved by `VALIDATE_HOMING_ENDSTOPS` (https://github.com/MarlinFirmware/Marlin/pull/11912) or `DETECT_BROKEN_ENDSTOP` (https://github.com/MarlinFirmware/Marlin/pull/18424)?