MarlinFirmware/Marlin #2948: Feature Request: Don't set the software endstops with triggered hardware endstops

AnHardt, 2016-01-30:
Don't set the software endstops with triggered hardware endstops.
Else returning to a min/max position, where the endsop is, will trigger the hardware endstop.

thinkyhead, 2016-03-16:
I noticed this too. There is some conflation between the endstop plugs —usually named xmin, xmax, ymin, ymax, zmin, zmax— and the things that they're attached to. I think we're moving in the right general direction (starting with #3065, for example). But more needs to be done to make this separation clearer.

AnHardt, 2016-03-16:
We need to move a little bit away from the hardware endstops, before we set the software endstops.
?_HOME_BUMP_MM is for sure big enough, because it is enough when the hardware endstop was hit with a much higher speed.

thinkyhead, 2016-03-18:
Could the PR to "disable endstops when not in use" double as a solution to this?

AnHardt, 2016-03-18:
Simple exteriment.
Assuming a x-min-endstop and x-min = 0 and ENDSTOPS_ONLY_FOR_HOMING = off
```
G28 X
G1 X100
G1 X0
M119
```
What does M119 show?
Is a probe involved?

So, no.

thinkyhead, 2016-03-27:
Some questions, not too deep…

Should this change apply if `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED` is not turned on?

Why does it matter if the endstops are triggered at the time software endstops are set? Software endstops are not –as far as I know– meant to _prevent_ running into the hardware endstops. They are simply meant to prevent movement beyond set physical limits. (Though I see the software endstops do some special tricks for `DUAL_X_CARRIAGE` in duplication mode.)

Are you proposing to change the software endstops so that they prevent any triggering of the hardware endstops? How do we know if that was the original intention?

Thanks if you can helping me to better understand what software endstops are about, for people who are using them.

DjDemonD, 2016-03-30:
I use them on a corexy as my bed is in the center of a larger physical range of movement. So I set manual home positions to x-7 y-30 then software endstops are set to x0-185, y0-185 and z0-170mm to ensure the nozzle remains inside the bed after that. I have no max physical endstops and I will never hit my x and y min physical endstops after homing as they are not reachable as I am prevented by the software endstops.  Does this help? As an aside I am planning to have a nozzle cleaning station outside of the range set by software endstops so that it would not take up build volume and would not be a collision hazard during printing, are there any plans to enable m211 to turn software endstops on or off in gcode, it might be useful for this type of application? For now I will use g92 to enable movement outside the software endstops. Do auto nozzle cleaning and then use g92 to reposition the nozzle back within the software endstop range.

thinkyhead, 2016-04-07:
What's new with this issue? Seems it needs more feedback. Hey @Roxy-3DPrintBoard @jbrazio what do you think about this question?

Roxy-3DPrintBoard, 2016-04-07:
> what do you think about this question?

I've been quiet because I don't know the answers to your questions.   

With that said...   It would seem to me a Software End Stop should behave the same as a Hardware End Stop.  The big reason (in my mind) for not behaving this way would be because it is going to add a bunch busy work for the processor to always be checking a list of stuff.

If the #define's let us configure how much over head we incur, maybe it is OK to strive to behave identical to the Hardware End Stops ???

jbrazio, 2016-04-07:
I agree with the principle *software endstops should not be triggered with triggered hardware endstops* under some contextualization..
 - Either an endstop is hardware of software it shouldn't matter, what would really matter is that *it is an endstop* thus preventing physical movement of *something*.
 - When we have a hardware endstop we shouldn't need to have a software endstop exactly on the same position – in my view they should be mutually exclusive, either one or the other.

> Software endstops (...) are simply meant to prevent movement beyond set physical limits. 

I see software endstops under this same perspective, they are a poor's man approach not to break something on the robot, ideally you would have endstops at either ends (min,max) of all axes.

Now the million dollar question is if this *is the definition of endstop that Marlin currently has*.

thinkyhead, 2016-04-08:
#3227 is also related to this question – maintaining the software endstops even if someone uses `G92` to fool the machine into thinking the nozzle is elsewhere.

boelle, 2017-04-26:
i would agree with @jbrazio 

i would still need them as my endstop are further away from where they should be

i use to decrease the build volume to what it is so that 0,0 is precise at the edge in one corner

should be added to 1.1.1 milestone

closed 2024-02-12 by AnHardt