MarlinFirmware/Marlin #2954: Feature Request: Make negative coordinates possible on (full) Graphic LCDs

AnHardt, 2016-01-30:
Make negative coordinates possible on (full) Graphic LCDs.

thinkyhead, 2016-03-16:
Is this still not done? It should be a pretty quick fix, yes?

AnHardt, 2016-03-16:
No space. For negative numbers we have to drop a digit.
Did not make sense to me before #2989 (or one of the predecessors) was merged (changing the same lines of code).

Roxy-3DPrintBoard, 2016-03-16:
Maybe the number could be done in inverse video to indicate it has flipped to minus?   That is kind of a kludged solution, but if there is no space...  and the feature is needed, some way needs to be found to indicate the status....

AnHardt, 2016-03-16:
https://github.com/MarlinFirmware/Marlin/issues/2675#issuecomment-151672097
is already a reasonable plan i think. For that small numbers between -100 to -999 you don't need the last digit - if displaying the x/y-coordinates makes sense at all.

Roxy-3DPrintBoard, 2016-03-16:
The LCD Panels are slow enough as it is...  But with a proportional font, digits (and decimal point) would pack tighter together.   But "Yes!"   When you are in the range of -100 to -999 you can afford to lose a digit.

thinkyhead, 2016-03-18:
Yeah, I was thinking also, just drop a digit. But for a very big delta, -100 might just be far from center. Hmm....

AnHardt, 2016-03-18:
display_error = abs(position - displayed_postition)
position = -100.1
displayed_position "100.1"  > display_error = 200.2
displayed_position "-100."  > display_error = 0.1

position = -998.9
displayed_position "998.9"  > display_error = 1,997.8
displayed_position "-999."  > display_error = 0.1

Even in a bad implementation the display_error is never larger than 1.
The sign is much more important, than to know about the digits behind the dot.

If you want to know a more exact reading you could go to the move_menu, where we have infinite room to display digits.

closed 2016-05-02 by AnHardt