MarlinFirmware/Marlin #2956: Bug Report: DEFAULT_STEPPER_DEACTIVE_TIME does not work

AnHardt, 2016-01-30:
DEFAULT_STEPPER_DEACTIVE_TIME does not work as expected.

After DEFAULT_STEPPER_DEACTIVE_TIME is over only those axes are deactivated where DISABLE_? is set to true. But these axis are already deactivated when the planner buffer for that axis is empty.
Make the configuration of these different features independent from each other.

dccharacter, 2016-02-04:
Agree with AnHardt. From studying the code the axis are disabled ONLY if DISABLE_? is set:
if (stepper_inactive_time && ms > previous_cmd_ms + stepper_inactive_time
      && !ignore_stepper_queue && !blocks_queued()) {
    #if DISABLE_X == true
      disable_x();
    #endif
However, on my setup the axis get deactivated immediately (that empty planner buffer rule AnHardt is talking about?) Anyways, this is not clear and definitely doesn't work as expected.

AnHardt, 2016-02-04:
planner.cpp: check_axes_activity()
is the original place where DISABLE_? was meant for.

stepper_inactive_time used to deactivate __all__ steppers - no configuration was possible.
A to short DEFAULT_STEPPER_DEACTIVE_TIME together with not updating stepper_inactive_time during long lasting g-codes (M109, M190, G29) stitches off the steppers, what is causing the loss of axis_known_position, what is causing the loss of the display.

In effect you currently have the choice to deactivate the steppers immediately or not at all.

AnHardt, 2016-03-10:
Fixed by #2989

closed 2016-03-10 by AnHardt