MarlinFirmware/Marlin #2958: Feature Request: Speed up graphic LCDs

AnHardt, 2016-01-30:
Speed up graphic LCDs.
Screen updates for graphic LCDs take for ever and make fast move sequences unsteady.
Measurements i made talk about:

REPRAP_DISCOUNT_SMART_CONTROLLER below 30ms
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER about 80ms.
VIKI2 about 170ms

Roxy-3DPrintBoard, 2016-01-30:
Are you saying the RepRap Discount Smart Controller can do an update in under 30ms?   That seems pretty fast if it can really do that.   

But 170ms for the Viki2 does seem to suggest something needs to be done to speed it up.

AnHardt, 2016-01-30:
The test looked like https://github.com/AnHardt/Marlin/commit/714a72b704b899d7e7eca8d810bb7d8c73248f9e.

REPRAP_DISCOUNT_SMART_CONTROLLER was included as a reference for character based displays.
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is the fastest of the graphical displays.
VIKI2 is the slowest. The current driver needs 8 loops through the display routine to make one update and uses u8g.setRot180().

thinkyhead, 2016-03-11:
It's good to see this working better all the time. I can't imagine there's much more optimization that can be done at this point.

closed 2016-05-04 by jbrazio