MarlinFirmware/Marlin #2959: Stops during Print

bagwan, 2016-01-30:
Hello.

I builded a Delta Printer with 265Diameter and Z250. Not too big.
While Printing he is stopping for a very short Time. But enough to make small Blobs on the Building objekt. These Stops are not lost steps, all Motors inl. extruder stops and go at the same time
I played with Printing speed and acceleration. also i tried to print from usb and sd. theres no difference.
I changed the arduino and the ramps. I changed 13,8V power Supply to 12V Atx supply. i changed the A4988 to Drv8825 steppers.
also i added a SSR relais to the heatbed. 

The complete Mechanic runs smooth and soft. the Print looks good, but the Blobs are disturbing.

I have no idea, what the Problem is.

Thanks for help, and excuse my terrible english.

Roxy-3DPrintBoard, 2016-01-30:
You might want to try lowering this number.  That will make the processor not work as hard:

#define DELTA_SEGMENTS_PER_SECOND 160

And then you might want to try changing this number to be bigger in UltraLCD.h :

#define LCD_UPDATE_INTERVAL 160

Why don't you try:
#define DELTA_SEGMENTS_PER_SECOND 130
#define LCD_UPDATE_INTERVAL 250

And see if your printer stops stuttering.   The downside of making these changes is very fast, long curves will be made with less line segments and there is a chance you will see that in the print.  And the LCD panel will not be quite has responsive when it is updating information.

bagwan, 2016-01-30:
Okay. Thanks for your fast Answer. I will try this and give feedback.

Roxy-3DPrintBoard, 2016-01-30:
And...  If it works...   You might want to slowly bump the DELTA_SEGMENTS_PER_SECOND number back up a bit and retest.   Especially, if you can tolerate the slower LCD Display update.  More is better on the DELTA_SEGMENTS number.  But not if it is causing the processor to stutter.

bagwan, 2016-01-31:
hi.
Delta Segments from 200 to 160
and Display Interval fromm 150 to 300

Seems to Work. 

Thanks for your Help.

i have another question about set home Offset via Display.
should I make a new thread about this?

bagwan, 2016-02-09:
Hi again. 
i still have this issue with the stops during Printing.
delta segments are now at 90 and lcd update rate at 450.
both settings are not optimal for good printing and handling.

would it be help, when i change from the full graphic Display to the smart controller?  then i dont have to use the ug8 library. should this help to save processor memory?

thanks

AnHardt, 2016-02-09:
Indeed a `REPRAP_DISCOUNT_SMART_CONTROLLER` takes about 25ms per display update, while the fastest oft the full graphic controllers `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` takes about 85ms (the others i have tested are slower - up to 175ms) with a stock Marlin 1.1RC3.

But before you buy a character display, please test my `https://github.com/AnHardt/Marlin/tree/lcd-speedup` branch based on RC3+some fixex. I'm desperately looking for feedback before it dates out an other time.

Roxy-3DPrintBoard, 2016-02-09:
@AnHardt 
> But before you buy a character display, please test my  https://github.com/AnHardt/Marlin/tree/lcd-speedup  branch based on RC3+some fixex. I'm desperately looking for feedback before it dates out an other time. 

Do you have fixes in your branch that do not exist anywhere else?   We should get those crossed over to RC3 !   As for the speed ups, those are valuable too.   I'm just not sure RC-3 is the right place to put them.   (But I acknowledge, speeding up the code does fix problems, so maybe we should consider that.)

AnHardt, 2016-02-09:
@Roxy-3DPrintBoard 
The base is somewhere between RC3 and RCBugFix - nothing special.

The lcd-speedup branch is a bundle of different commits, turning on different bolts but also polishing previous commits . Most of them are pretty independent from each other, but all of them change the same 4 or 5 files. So making single PRs, based one on the other, will take forever. (Especially with the high risk someone rebases again).
All that work does not make much sense to me without some positive feedback. I'd really like to know if the sum of the speedups is enough to use Drv8825 on a Delta with a full graphical display.
If i'd know that, i could, maybe, find someone who wants to do the work and is patient enough to finish it.

You know very well i'll do nothing in the MarlinDev repository anymore.
Even today's PRs here in Marlin are done more with the feeling, i have to - no one else will do it - but it' s soooo important.

bagwan, 2016-02-10:
okay. Thanks. I will try it and give some feedback about it. I use the drv8825 steppers in my Delta.

setup und run a complete new Code will took some time.

thanks.

closed 2016-04-27 by jbrazio