MarlinFirmware/Marlin #2960: No rule to make wiring.o

dccharacter, 2016-01-30:
When making Marlin, this error is shown before stopping:

> make: *** No rule to make target '/home/andy/tmp/wiring.o', needed by '/home/andy/tmp/Marlin.elf'.  Stop.

Some suggest there might be an issue with paths - I tried almost 128 different versions of paths to the tools as well as reinstalling the tools. Arduino IDE is able to compile the project.

OS: openSUSE leap 42.3

The Makefile:
```
HARDWARE_MOTHERBOARD ?=BOARD_RAMPS_13_EFB
ARDUINO_INSTALL_DIR  ?= /usr/share/arduino-1.0.6/
ARDUINO_VERSION      ?= 106
AVR_TOOLS_PATH ?= /opt/cross/avr/bin
UPLOAD_RATE        ?= 115200
AVRDUDE_PROGRAMMER ?= wiring
UPLOAD_PORT        ?= /dev/ttyACM0
BUILD_DIR          ?= /home/andy/tmp
LIQUID_TWI2        ?= 0
WIRE               ?= 0
```

There's also a confusing direction in Makefile:
Requires a trailing slash. (ex: /usr/local/avr-gcc/bin)
As you can see, the comment says the slash is required, and the example doesn't carry a one. Multiply this with the information that the original error may have to do something with a wrong character in the end of the avr-gcc tools config line and you have an unsolvable task.

Also, the comment says the path to avr-gcc is optional... But it seems to require that.

dccharacter, 2016-01-30:
Arduino-Makefile is able to compile the project as well

thinkyhead, 2016-03-11:
Is this improved at all in the current `RCBugFix`? We don't have too many developers looking at this, but recently we have received some Makefile updates.

dccharacter, 2016-03-11:
I was actually able to find the issue - that was as far as I remember, the board name. You have to specify the numerical id as far as I remember, it wouldn't understand (logically enough) this: "BOARD_RAMPS_13_EFB". So because of this the rest of the paths in the make file were constructed incorrectly and the target couldn't build.

jbrazio, 2016-03-12:
@dccharacter So do we need to fix the existing Makefile at RCBugfix ?

dccharacter, 2016-03-12:
Well, this was my error and when I fixed that, I was able to use the makefile. This is not a critical issue. However I spent a lot of time (half a day easily) finding the root cause. Maybe it's because of the lack of knowledge, but maby it could use some better documentation. You decide. Thanks for what you do guys!!!!!

thinkyhead, 2016-03-16:
Post a new issue if the Makefile starts to act up again. In general it would be good to get as much feedback about it as we can, because we don't really know who's using it and in what way.

closed 2016-03-16 by thinkyhead