MarlinFirmware/Marlin #2961: Bug Report: sudden speed changes on delta printer

markusneg, 2016-01-31:
The suspected bug already was discussed a lot in the RepRap-forums [1]. Short-version: under some circumstances which include broad extrusion width (= slow effector speed and high extrusion rate), all movements (x, y, e) are distinctively (at least 2x) accelerated to a new constant speed level. The speed change occurs when coming closer to the center of the bed, whereas in the outer radii speed turns back to normal. During accelerated phases, the CPU seems to be overloaded since the user interface (LCD + knob) is not responsive. Reducing DELTA_SEGMENTS_PER_SECOND (to values lower than 20) restricts the speed changes to increasingly smaller bed radii but cannot completely resolve them.

In the forums, there are also some videos and gcode exposing the behavior.

[1] http://forums.reprap.org/read.php?263,590637

markusneg, 2016-02-02:
Could someone with a delta printer please confirm the bug? It should be pretty easy: move the effector into x-direction by 60 mm using 180 mm/min and extrude 7 mm of filament. And have DELTA_SEGMENTS_PER_SECOND at least at 30.

markusneg, 2016-03-06:
I did some further work and confirmed the bug: I tried 2 completely different sets of hardware (RAMPS and Arduino), disconnected all drivers and motors during tests to preclude any electrical feedback (e.g. induction / resonance), tried different Arduino IDEs each on Windows and Linux and disabled the Panel. The behavior is always the same: everything (x,y,z,e) globally speeds up under certain circumstances, even with low delta_segments_per_...

In consider this a **serious bug** of Marlin for Delta robots and - as I put a lot of time in investigation  and this bug should be very easily to reproduce - really would appreciate any feedback or comment.

Roxy-3DPrintBoard, 2016-03-07:
A lot of people are reading what you are writing...   We are trying to figure out how to check for it on our machines.

thinkyhead, 2016-05-04:
Are we still seeing this issue in the current `RCBugFix` branch? I would look into it, but I don't have a Delta to test.

markusneg, 2016-05-04:
I will test `RCBugFix` within the next 2 weeks. However, it would still be fine if someone else could reproduce the issue as well.

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

markusneg, 2016-05-29:
I had time to test the RCBugFix branch (commit dee6322) and still have the same behavior. In the video [1] I went from (84, -45) to (-57, 51) mm (~170 mm length) while extruding 17 mm of filament using a feedrate of 180 mm/min. A distinct speed change can be seen.

For the test shown on the video, the extruder motor was unplugged (waste of filament). Validation of a software bug again has been accomplished by exchanging electronics (RAMPS, Mega) and by removing all but one of the four stepper drivers in random combinations. So neither electrical feedback nor a defective uC-board cause the behaviour.

F = 180 mm/min has been chosen for the video in order to emphasize the behaviour. However, also with more "normal" feedrates, the end effector speed deviates from the planned trajectory. There seems to be a special ratio of F_extruder / F_xyz amplifying the behaviour. When the extruder is disabled, everything acts normal (i.e. the upper command almost takes one minute).

Please reopen this issue and lets tackle the bug. 

[1] https://youtu.be/K1Q4zunVzi0

jbrazio, 2016-07-23:
@MarlinFirmware/testers-delta-team do you see the same behavior with your delta robot ?

esenapaj, 2016-07-23:
I'm seeing a another? strange behaviors of E axis.
for example if I ordered as follows, second order doesn't works.
```
g1 e1
g1 e1
```
But following orders works.
```
g91
g1 e1
g1 e1
```

I think that E axis should be treated as relative coordinates always, but it looks like that it treated as both relative and absolute currently.

jbrazio, 2016-07-23:
But can you confirm this bug ? Follow the instructions in the first comment.

thinkyhead, 2016-07-24:
@esenapaj This is how E movement should behave (when `AXIS_RELATIVE_MODES` is set to all `false` values).
```gcode
G1 E1 ; Move from E0 to E1
G1 E1 ; Stay at E1 (do nothing)
```
```gcode
G91   ; Relative mode for XYZE
G1 E1 ; Move E by +1
G1 E1 ; Move E by +1
```
```gcode
G90    ; Absolute mode for XYZE
M83    ; Relative mode for E only
G1 E1  ; Move E by +1
G1 E1  ; Move E by +1
M82    ; Absolute mode for E
G1 E20 ; Move E to 20
```

boelle, 2016-08-11:
maybe close this one as it has not been confirmed??

or suggest to use RC7 and open a new thread if the issue is still there?

closed 2016-08-11 by thinkyhead

esenapaj, 2016-08-13:
@markusneg 
I've tested with newest RCBugFix, but can't reproduced the sudden speed change.

settings:
```cpp
#define DELTA_SEGMENTS_PER_SECOND 80
#define XYZ_FULL_STEPS_PER_ROTATION 400
#define XYZ_MICROSTEPS 32
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 16

#define DEFAULT_MAX_FEEDRATE          {300, 300, 300, 300}
#define DEFAULT_MAX_ACCELERATION      {3000,3000,3000,9000}

#define DEFAULT_ACCELERATION          3000
#define DEFAULT_RETRACT_ACCELERATION  9000
#define DEFAULT_TRAVEL_ACCELERATION   3000

#define DEFAULT_XYJERK                10.0
#define DEFAULT_ZJERK                 10.0
#define DEFAULT_EJERK                 5.0
```

commands:
```
m502
m500
g28
g91
g1 x-30 e7 f180
g1 x60 e7 f180
g1 x-60 e7 f180
g1 x60 e7 f180
g1 x-60 e7 f180
g1 x60 e7 f180
g1 x-60 e7 f180
g1 x60 e7 f180
g1 x-60 e7 f180
g1 x60 e7 f180
g1 x-60 e7 f180
```

@thinkyhead 
I'm thankful for your teaching.
In case of Repetier-Host, when we clicked "extrude" button, `g91` is executed before `g1` always, and after `g1`, `g90` is executed always.
[image]

Therefore, if I execute
```
g1 e1
g1 e1
```
after clicking "extrude" button, second `g1 e1` become meaningless always because it's in the absolute mode.
I feel that it's a bit problematic, but I understand that this is a specification, isn't a bug.