MarlinFirmware/Marlin #2962: Z probe servo endstop does not work unless auto leveling is turned on.

Entr04y, 2016-02-01:
I have a z probe with a servo as my z-min endstop (it is a bltouch)  I wish to use mesh leveling, but the servo code for the Z axis seems to require auto bed leveling to be turned on... which causes the head to crash into the bed because the probe doesn't deploy on homing.  I found the following in RC3 (around line 1838, which is in the function for homing a single axis):

```
#if ENABLED(Z_PROBE_SLED)
      // Get Probe
      if (axis == Z_AXIS) {
        if (axis_home_dir < 0) dock_sled(false);
      }
#endif

#if SERVO_LEVELING && DISABLED(Z_PROBE_SLED)

      // Deploy a Z probe if there is one, and homing towards the bed
      if (axis == Z_AXIS) {
        if (axis_home_dir < 0) deploy_z_probe();
      }

#endif
```

In Conditionals.h there is this:

...
#define SERVO_LEVELING (defined(AUTO_BED_LEVELING_FEATURE) && defined(Z_ENDSTOP_SERVO_NR))
...

However, for the X and Y axis, we have this:
...
#if HAS_SERVO_ENDSTOPS
      // Engage Servo endstop if enabled
      if (axis != Z_AXIS && servo_endstop_id[axis] >= 0)
        servo[servo_endstop_id[axis]].move(servo_endstop_angle[axis][0]);
    #endif
...

So I think the end result is if you are homing X or Y, go ahead and use the servo, but if you are homing Z use the sled if there is one, but if its just a servo then don't use it unless auto bed leveling is enabled... because "deploy_z_probe()" is part of the auto leveling code, rather than an independent thing.

It seems to me that the code to use servo endstops should exist outside of and independent of the auto leveling code, regardless of which axis it is.  This would allow us to use mesh leveling, or just a servo for the endstop without any kind of leveling turned on.

Entr04y, 2016-02-01:
I've created a workaround that checks to see if auto leveling and z-sled are disabled and Z has a servo on the endstop and activate the servo while z is homing.

It doesn't do anything about Z offsets, backing Z off the bed after homing is complete or anything other than the normal axis homing routine.  The diff is here:

https://github.com/Entr04y/Marlin/commit/0f117218ab6a201e273b8844550b578627d55247?diff=split

Entr04y, 2016-02-02:
I think this might fall under feature request, so I'm moving it to MarlinDev.

closed 2016-02-02 by Entr04y

samwyze, 2016-03-08:
I think z-probe variables like the offsets and the deploy and stow routines should be independent of auto bed leveling. This adds the possibility to use them in an auto mesh level routine and also to send manual deploy/stow g-code from a console...therefore I propose the way it is currently handled is a BUG and not a new feature/feature request. But that might just be me :)

thinkyhead, 2016-03-09:
Manual bed leveling and auto bed leveling should be considered as being separate from the two grid-style bed compensation systems (mesh leveling and grid-based least-squares leveling), and perhaps the three-point leveling -which still exists in the code- can also be considered another leveling option. At the moment "manual bed leveling" is tied to mesh bed leveling, and the other options are tied to auto bed leveling. It would be good to split up all these pieces, make it more pluggable, but for the moment they are interleaved in two G29 handlers.

jbrazio, 2016-03-09:
At this point making such distinction is indeed a new feature and not a bugfix, for sure is something that will not make it to RC nor 1.1.0 GA.

samwyze, 2016-03-11:
Yes, I concur. Unfortunately as much as I'd love to see this reworked it is indeed a new feature request.

thinkyhead, 2016-03-11:
I hope that after this release we can simplify the concepts around probes, endstops, homing, auto homing, and bed leveling. But for now it's a small plate of spaghetti.