MarlinFirmware/Marlin #2963: Extruder temperature problem (160 deg)

modeljjy, 2016-02-01:
Hi, this is James. I have a question about the extruder temp. I have a Lulzbot TAZ 4 model with one extruder. 
The temperature of extruder is not risen up to the desired temperature (220 deg) for ABS, just remaining within 160 deg. I already changed the extruder, heat resistor, and firmware. But same results. I do not think the mainboard (RAMBO v.1.3.) have the problem.
Today, I found some messages in Cura. 

Here are the message:
Recv: echo: Last Updated: Feb 20 2015 10:55:07 | Author: (Aleph Objects, Inc, TAZ  config)
Recv: Compiled: Feb 20 2015
Recv: echo: Free Memory: 3114  PlannerBufferBytes: 1296
Recv: echo:Stored settings retrieved
Recv: Error:0
Recv: : Extruder switched off. MINTEMP triggered !
Changing monitoring state from 'Operational' to 'Error: 0: Extruder switched off. MINTEMP t...'
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Recv: echo:SD init fail
Recv: start
Recv: echo: External Reset
Recv: Marlin 1.0.0
Recv: echo: Last Updated: Feb 20 2015 10:55:07 | Author: (Aleph Objects, Inc, TAZ  config)
Recv: Compiled: Feb 20 2015
Recv: echo: Free Memory: 3114  PlannerBufferBytes: 1296
Recv: echo:Stored settings retrieved
Recv: Error:0
Recv: : Extruder switched off. MINTEMP triggered !
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Recv: echo:SD init fail
Recv: 

Somebody help me! I really want to fix it. If you have a solution, I will give a reward for any way.

Regards,
James

brainscan, 2016-02-01:
Can you post your configuration.h etc for people to look at please? Also what temp does it show when at room temperature?

KiteLab, 2016-02-02:
Ether its extremely cold,
or your thermistor is unconnected now.

bobsalter, 2016-02-11:
Im having a similar problem. Everything looks good, bed heats but extruder never quite makes it, gets to within 10 of set temp then goes down and up forever

o3dodgesrt4, 2016-02-24:
similar issue here on my delta... target temp is 220... gets to 200-206C and then turns off and hard reboots... sometimes it just fluctuates and never stabilizes... even after a auto PID tune... iv replaced the hotend (E3dv5) with the complete E3Dv6 kit and still had the same issue..then i replaced the ramps 1.4 with a known working one from my mendel90, and same issue...iv been fighting it for weeks now trying to figure it out, but to no avail...

thinkyhead, 2016-02-29:
Do you see this problems with version `1.0.2` or just in the current RC? There have been some changes to the thermal-protection code. However, the heating itself should still be reliable.

bobsalter, 2016-02-29:
1 and 1.0.2 run perfectly. Problem for me is only in current RC

Bob

o3dodgesrt4, 2016-02-29:
current RC for me as well.. im using 1.0.2 right now and it works fine for my delta... whats weird is that the current RC seems to work just fine for my mendel90 but not my delta...

thinkyhead, 2016-03-02:
Y'all should try the `RCBugFix` branch just to confirm this hasn't already been fixed (either intentionally or as a side-effect).

I'll just post these links here for our convenience:

- [Changes between 1.0.2-1 and RC3](https://github.com/MarlinFirmware/Marlin/compare/1.0.2-1...1.1.0-RC3)
- [Changes since RC3](https://github.com/MarlinFirmware/Marlin/compare/1.1.0-RC3...RCBugFix)

jbrazio, 2016-04-28:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-28 by jbrazio

o3dodgesrt4, 2016-04-28:
Yup no prob.  Close it if it wasnt already..