MarlinFirmware/Marlin #2964: Estremly slow when browsing the SD from the LCD

albertolg, 2016-02-02:
Using the 1.1.0RC3, reprap smart controller, it's very slow when I try to select the file to print with the encoder, there is a delay of about 1 second from when I rotate/press the encoder and the reaction of the LCD, but only when browsing the SD, in the other menus it's fine, fast and responsive... is maybe a config problem or what? I couldn't find an open issue like this

thinkyhead, 2016-03-11:
Is this still a problem in `RCBugFix`? We've been doing a lot of work around the LCD menus and SD printing lately.

albertolg, 2016-03-13:
Tested now, the problem is still there... :-(

Blue-Marlin, 2016-03-13:
About how many files are in the directory?

albertolg, 2016-03-13:
2

thinkyhead, 2016-03-13:
It sounds like potentially bad wiring. Is there no change if you tighten up the connections?

jbrazio, 2016-03-13:
Have you tried with a different SD card ?

albertolg, 2016-03-14:
guys... sorry... it's not a bug.... did a low level format of my 4Gb SD, loaded the files again and now it's again responsive... I don't know what happened to the SD, always used the same, we can close this topic.. thanks again and sorry :-(

CONSULitAS, 2016-03-15:
@albertolg You are welcome.
You can (and should) close the issue yourself by just clicking "close". :smile:

closed 2016-03-15 by albertolg

JanVelleman, 2017-08-11:
This still seems to be a recurring issue for me (graphical display and SD support). Really slow scrolling behavior in some folders and not in others. When reformatted and re-used. It's seems to be ok but deteriorates rapidly when adding a file...?