MarlinFirmware/Marlin #2968: EEPROM versioning seems inconsistent

RetireeJay46, 2016-02-03:
I understand the principle that if the structure of the data stored in EEPROM is changed, then there should be a new Version number for the EEPROM.  This prevents a user from getting scrambled data from the EEPROM if they load a new firmware into their machine and the new firmware doesn't use EEPROM data the same way that the old firmware did.  The Version number serves as a "sanity check" so that if the new firmware doesn't see its own Version stored in the EEPROM, it will use "factory defaults" instead of reading from the EEPROM.

But, in the code in Configuration_Store.cpp, which manages the EEPROM, there are many conditional statements "#ifdef..." or "if ENABLED..."  So the net result is that if someone is modifying the code (for example to turn on or turn off Mesh Bed Leveling, or add an extruder) - the contents of EEPROM will be changed, but the person writing the code may be completely unaware of the need to change the Version number - particularly if he/she doesn't even have Configuration_Store.cpp open.

I'm not enough of a C programmer to suggest a bulletproof solution, but the present state of the code looks a little fragile.

KiteLab, 2016-02-03:
Because the dummy variables in the #else branch the site and positioon of the other content should not alter.

RetireeJay46, 2016-02-04:
Thanks!  Exactly right.  (I really need to learn C / C++)

closed 2016-02-04 by RetireeJay46