MarlinFirmware/Marlin #2971: Feature Request: Support for Printrboard Rev F

StephS, 2016-02-05:
The PrintrBoard Rev F utilizes a mcp4728 DAC to set motor current. Support for this DAC needs to be added to utilize the board properly, and a new pins definition. I have built a fix by merging and recoding printrbots repo with the latest 1.1.0 RC3 release. However, I am no expert with Marlin code compliance.
Printrbots implementation utilizes 2 new M-codes to set and write the DAC settings to the DAC eeprom, M909 (read DAC) and M910 (Write DAC) M907 is re-used to set value, M908 for direct control.
The Pins file is included for the RevF board.
Please see my repo for working code: https://github.com/StephS/Marlin (Tested working on a new PrintrBoard Rev F5). Needs review and cleanup to eventually merge with dev branch.

rdlaner, 2016-02-24:
Have you been able to use your fork successfully with a Rev F board? For some reason, any version of Marlin 1.1 that I've tried with a Rev F board I can compile and load fine, but I cannot connect to the board over a serial interface (namely Pronterface and Cura) afterwards.

thinkyhead, 2016-03-02:
> can compile and load fine, but I cannot connect

You might need to add `mcp4728.cpp` and some other code to make everything happy. Here are the (last set of) changes that were made to support REVF: https://github.com/MarlinFirmware/Marlin/commit/8b70f677ec3a60d2a7ef91934875d8a4f57fbb94?diff=split

thinkyhead, 2016-03-02:
Sorry this issue has been languishing for so long. We have very few coders reviewing the queue lately.

Roxy-3DPrintBoard, 2016-03-02:
Is the Baud rate set correctly?  There was a change to that (and then back again I think).  This caused some connection problems.

rdlaner, 2016-03-05:
Thanks for the input everyone. I went through and tested the changes outlined in the diff you pointed out. I am still having the same problem. I can compile both with the Arduino IDE and platformio, but for whatever reason, I cannot establish a serial connection after loading the firmware. I'm wondering if somehow the correct core pin definitions for the AT90USB1286 are not being selected. The strange part is that if I compile the Printrbot Printrboard RevF version of Marlin with either the Arduino IDE or platformio, everything works just fine.
What do you guys use for building/compiling your firmware?

As for baud rate, as I understand it, the AT90USB1286 has a direct USB connection which handles the baud rate automatically. In fact, when building for this chip, most of MarlinSerial.cpp is ignored. I have tried different buad rates for the hell of it, but to no avail.

rdlaner, 2016-03-06:
What version of Arduino are you building this with? I'm trying to determine if I still need to install teensyduino. 
I'm also going to attempt to build with the makefile.

StephS, 2016-03-10:
Did you try the branch I listed above? I still installed teensyduino for my build, just grab the latest version from pjrc.
The branch I have is working, tested all features including SD card (there is a separate bug on this). I'm using it on my custom printer on a brand new rev F5 board. stepper.cpp and dac_mcp4728.cpp are the main files that need to be pulled. it adds new gcode to control the dac (not sure how new gcode standards are set, I just copied the methodology from printrbot.)

Roxy-3DPrintBoard, 2016-03-10:
Do we know what changed from rev F5 from older revisions?  I'm using a rev D PrintRboard.   If we knew what the changes are, that would help limit the scope of what we are fighting here.

thinkyhead, 2016-03-11:
Sounds like it should be pretty straightforward to adapt. I have to chase down some kind of babystepping issue, but I can try to get this together after that. The more hardware we can support out of the box, the merrier.

rdlaner, 2016-03-12:
@StephS What are you using to compile, make? I'm still having some issues getting your branch compiled correctly.

rdlaner, 2016-03-12:
Just a random thought, I am powering the printrboard with an ATX power supply. Are there any extra jumpers I need to make to make sure the board stays powered? I ask because I saw these calls at the beginning of setup():
  setup_killpin();
  setup_filrunoutpin();
  setup_powerhold();

StephS, 2016-03-17:
Ok, I screwed some things up with my repo, had to redo the branch and I forgot a few files. I added the updated boards.h and pins.h - the update is on my git.
You need to select BOARD_PRINTRBOARD_REVF
It isn't configured for the a printrbot, you need to change the configuration to match your printer.
I am compiling on windows. Teensyduino 1.26 and Arduino 1.6.6

You don't need anything to turn on the ATX supply, it should be always on.

Roxy: RevD uses Trimmer Pots to set the current level on the board. Rev F uses a DAC. Rev D should be fully compatible with the current builds of Marlin. Rev F requires my branch to add DAC support (you can run the current marlin with the pins configured correctly, but will not be able to modify stepper motor current).
see the reprap wiki for the changes and schematics: http://reprap.org/wiki/Printrboard

Roxy-3DPrintBoard, 2016-03-17:
Oh!  I want a rev. F PrintRboard !

rdlaner, 2016-03-17:
For the life of me I cannot get anything to communicate with the printrboard after I load the latest Marlin firmware on it.
@StephS I was able to compile your branch both with platformio and Arduino/Teensyduino. That went well. However, I still have the same problem where after I load the new firmware and try to connect to the board, I am unable to make a connection.
What do you use to flash the .hex file to the printrboard? I am on a mac, so I wonder if that is part of the problem. The only thing I have that works is dfu-programmer.

StephS, 2016-03-17:
I use the atmel dfu programmer. Are you removing the boot mode jumper and resetting after programming? Does it show up as a com port? 
It could be a driver issue on the Mac. It might be looking for a device ID from printrbot.

Make sure you have the teensy2.0++ selected in arduino. The directions should be the same for printrbots firmware as well as my branch. Check to see if the device id is the same with printrbots firmware and mine. 

Might need to check with other people who have a mac, I know Printrbot does, but they're focusing on developing a new firmware.

rdlaner, 2016-03-19:
Yes, I remove the jumper and press the reset button after programming. Afterwards I do see a new com port at tty.usbmodem12341. I am also selecting teesny2.0++ in the Arduino IDE.
Additionally, I have parallels on my mac and am running Windows 7. I was able to compile with Arduino and Teensyduino and load with the FLIP programmer. After removing the jumper and resetting the board, I did have a new serial com port. I then tried to connect using Pronterface (on Windows) and I had the same exact problem where after clicking "connect" Pronterface would never be able to make a successful connection.

What do you mean exactly by "device id"? I don't actually have a printrbot, I'm just trying to use the printrboard on my custom 3D printer.

thinkyhead, 2016-03-19:
Check out #3182 and see if it has all the T's crossed and I's dotted…

jbrazio, 2016-03-25:
@thinkyhead this should be closed as RC4 is released.

closed 2016-03-26 by thinkyhead