MarlinFirmware/Marlin #2979: Home All & Auto Bed Leveling

brand0drag0, 2016-02-09:
Here's the problem I'm seeing from Marlin:

When attempting to home all axes, whether AUTO_BED_LEVELING_FEATURE is enabled or not,

on RC3:  the machine homes X, homes Y, then stops.
on RC2:  the machine homes x, homes y, then rams the bed into the head (ignores the z-sensor completely).

brand0drag0, 2016-02-09:
I am using an Azteeg X3 PRO with CORE XY Motion.

Blue-Marlin, 2016-02-09:
Please tell us where your endstops are and where your probe is connected.
Please send a M119 in mid air and post the result (with deployed/stowed probe).
Please give us a link to your Configuration.h, Configuration_adv.h and, if changed, pin file.

thinkyhead, 2016-03-16:
Where does this issue stand with the current `RCBugFix`? We've made a lot of changes for CORE machines.

closed 2016-04-27 by jbrazio