MarlinFirmware/Marlin #2980: G29 Probe position out of range - servo do not deploy

Durga000, 2016-02-09:
Hello, 

I am trying to configure the ABL for a delta printer ( ABL is z-min endstop moved by a servo) using Marlin 1.1.0-RC3 and getting the following error:

> 
11:04:42.216 : G29 Auto Bed Leveling
11:04:42.217 : ?Probe (R)ight position out of range.
11:04:42.220 : ?Probe (B)ack position out of range.
Is my endstop too far from the nozzle ?

I have made a test using lower values for the probe offset, the G29 commands execute but the servo do not deploy the probe.
G30, M401, M402 are working.

don't know if this is useful but M119 after homing:
>11:40:57.370 : Reporting endstop status
11:40:57.371 : x_max: TRIGGERED
11:40:57.372 : y_max: TRIGGERED
11:40:57.374 : z_min: open
11:40:57.374 : z_max: TRIGGERED

Don't know where the issue is...
Any help will be welcoming
Thanks

> 
>
//===========================================================================
//============================== Delta Settings =============================
//===========================================================================

#define DELTA

#if ENABLED(DELTA)

#define DELTA_SEGMENTS_PER_SECOND 200

  #define DELTA_DIAGONAL_ROD 219.5 // mm
  #define DELTA_SMOOTH_ROD_OFFSET 130.0 //145 mm
  #define DELTA_EFFECTOR_OFFSET 30.0 // 19.9mm
  #define DELTA_CARRIAGE_OFFSET 20.0 // mm
  #define DELTA_RADIUS (DELTA_SMOOTH_ROD_OFFSET-DELTA_EFFECTOR_OFFSET-DELTA_CARRIAGE_OFFSET)

  #define DELTA_PRINTABLE_RADIUS 80.0

#endif

// Enable this option for Toshiba steppers
//#define CONFIG_STEPPERS_TOSHIBA

// @section homing

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = true; // false set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
[... 555 lines cut ...]
// If unsure, leave commented / disabled
//
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M851 to set the Z probe vertical offset from the nozzle. Store that setting with M500.
//
//#define X_ENDSTOP_SERVO_NR 1
//#define Y_ENDSTOP_SERVO_NR 2
#define Z_ENDSTOP_SERVO_NR 0
#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {160,50}} // X,Y,Z Axis Extend and Retract angles

// Servo deactivation
//
// With this option servos are powered only during movement, then turned off to prevent jitter.
#define DEACTIVATE_SERVOS_AFTER_MOVE

#if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
  // Delay (in microseconds) before turning the servo off. This depends on the servo speed.
  // 300ms is a good value but you can try less delay.
  // If the servo can't reach the requested position, increase it.
  #define SERVO_DEACTIVATION_DELAY 300
#endi

Blue-Marlin, 2016-02-09:
For the not deploying servo probe on a Delta during G29 you could try:
MarlinMain.cpp ~line 2869
```
    #if ENABLED(Z_PROBE_SLED)
      dock_sled(false); // engage (un-dock) the Z probe
-    #elif ENABLED(Z_PROBE_ALLEN_KEY) //|| SERVO_LEVELING
+    #elif ENABLED(Z_PROBE_ALLEN_KEY) || (SERVO_LEVELING && ENABLED(DELTA))
      deploy_z_probe();
    #endif
```
If the after the probing the probe does not stow add:
MarlinMain.cpp ~line 3130
```
    #if ENABLED(DELTA)
      // Allen Key Probe for Delta
-      #if ENABLED(Z_PROBE_ALLEN_KEY)
+      #if ENABLED(Z_PROBE_ALLEN_KEY) || SERVO_LEVELING
        stow_z_probe();
      #elif Z_RAISE_AFTER_PROBING > 0
        raise_z_after_probing();
      #endif
    #else // !DELTA
```

Are you sure you want to probe 81 points (AUTO_BED_LEVELING_GRID_POINTS 9)?
In that case you probably have to lower MIN_PROBE_EDGE to get rid of the "?Probe ... position out of range." errors.

Durga000, 2016-02-09:
Thanks, I have made the first mod ( MarlinMain.cpp ~line 2869), and the probe now deploy well, but do not retract after probing.. should I add a custom M402 gcode something like :
#define Z_PROBE_END_SCRIPT "M402"

To get this working I had to change in config.h:

>  #define X_PROBE_OFFSET_FROM_EXTRUDER -10.00     // -26. Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -9.00   // -25. Z probe to nozzle Y offset: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -7.80  // -7.8 Z probe to nozzle Z offset: -below (always!)

I also have changed to 5 the AUTO_BED_LEVELING_GRID_POINTS but do not impact the probe position errors

closed 2016-02-09 by Durga000