MarlinFirmware/Marlin #2982: G29 Probe position out of range - servo do not retract

Durga000, 2016-02-09:
following issue #2980 I have made the first mod ( MarlinMain.cpp ~line 2869), and the probe now deploy well, but do not retract after probing.
Should I add a custom M402 gcode something like :
#define Z_PROBE_END_SCRIPT "M402"

To get this working I had to change in config.h :

    #define X_PROBE_OFFSET_FROM_EXTRUDER -10.00 // -26. Z probe to nozzle X offset: -left +right
    #define Y_PROBE_OFFSET_FROM_EXTRUDER -9.00 // -25. Z probe to nozzle Y offset: -front +behind
    #define Z_PROBE_OFFSET_FROM_EXTRUDER -7.80 // -7.8 Z probe to nozzle Z offset: -below (always!)

I also have changed to 5 the AUTO_BED_LEVELING_GRID_POINTS but do not impact the probe position errors, when using the correct value for probe position I still have :

> 13:36:01.572 : G29 Auto Bed Leveling
13:36:01.572 : ?Probe (R)ight position out of range.
13:36:01.572 : ?Probe (B)ack position out of range.

I just have tried with  #define MIN_PROBE_EDGE 5 but still out of range.

brainscan, 2016-02-09:
What numbers do you have for the bed probing area?

brainscan, 2016-02-09:
Sorry ignore me, just noticed its for a delta.

Durga000, 2016-02-09:
I have tested with 

> 
#define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 37)

as my highest value for probe offset was 37
>
#define Y_PROBE_OFFSET_FROM_EXTRUDER -37.00   // Z probe to nozzle Y offset: -front +behind

And now the printer do the probing but this is not accurate at all.
The autolevel function is unusable

Roxy-3DPrintBoard, 2016-02-10:
How flat is your movement across the bed?   Can you print without Auto Bed Leveling?   Specifically, I'm wondering how much the nozzle raises and lowers as it moves parallel to the bed.

Durga000, 2016-02-10:
the movement across the bed is almost flat everywhere.
The print is quite good without ABL.
The Printer do the probing, then the nozzle raise and moves parallel to the bed.

Changing the Z_PROBE_OFFSET_FROM_EXTRUDER value do not change anything on printing with ABL.

Roxy-3DPrintBoard, 2016-02-10:
Do you have the EEPROM storage of settings turned on?   Did you do a M502 followed by a M500 when you loaded the new firmware?

Durga000, 2016-02-11:
Yes I have the EEPROM storage turned on but I did not do a M502 and M500.
Will try it tomorrow morning as it is too late for now but will let you know.

Many thanks for your help

Roxy-3DPrintBoard, 2016-02-11:
Note to Collaborators:   

We need to get clever and figure out how to make this easier for the user.   Maybe something where the values stay in place, but they stop being used until the user explicitly says to re-use the old settings.   @Wackerbarth 's ideas for storage of values in EEPROM would help with this.

thinkyhead, 2016-02-11:
I'm not sure what new scheme is being proposed for EEPROM storage, but in the run-up to 1.1.x I took the extra time to make sure every value in EEPROM has a corresponding M-code. The idea being, you could then re-use the function which displays the current configuration settings as GCode (as with `M503 S0`) and simply write that output to EEPROM. When you want to restore from EEPROM, simply "replay" whatever GCode is stored there. Even the smallest EEPROM can easily accommodate the resulting GCode string with all the options enabled (especially if trailing zeros are removed after the decimal point).

Durga000, 2016-02-11:
Many thanks, this is now working well (just need to fine tune the Z_PROBE_OFFSET), Thanks you for your dev. and support.

Should I close the issue or leave it open for EEPROM discussions ?

Roxy-3DPrintBoard, 2016-02-11:
>  I'm not sure what new scheme is being proposed for EEPROM storage...

The proposal is just that we don't save a big structure in the EEPROM.  Instead, we save each value with a name so regardless of what values are saved, they can be accessed independently from the firmware version.   

And what I'm proposing is we find a way to defeat those values (and not use them) when a new version of the firmware is loaded.  When a new version of the firmware is loaded, I would like the EEPROM values to still be saved, but for them to not be loaded and used until the user says to do that.  This behavior of using values out of the EEPROM without the user actually saying to do that is causing way too many problems.

One easy way to do that would be to store a checksum of the current version of the firmware that saved the EEPROM settings, and if it doesn't match the image that is trying to restore the EEPROM settings, it doesn't happen.

Roxy-3DPrintBoard, 2016-02-11:
> Should I close the issue or leave it open for EEPROM discussions ?

We will close it when done...

Wackerbarth, 2016-02-12:
@thinkyhead - "The idea being, you could then re-use the function which displays the current configuration settings as GCode (as with M503 S0) and simply write that output to EEPROM. When you want to restore from EEPROM, simply "replay" whatever GCode is stored there."

I like that general approach. It is also much in keeping with the "dictionary" storage idea. Storing GCode commands as strings in the EEPROM is just one possible implementation. Another would be to "encode" the content in some form and have a decoder that "played" the unencoded version into the internal command engine. (If stored as strings, the decoder would be nothing more than a copy_string operation.)

One thing that might simplify things would be to use a single M_code for "set_parameter" and code all of the particular parameters as a "key","value" pair under that code.

But, whatever the mechanism, we would be able to deal with variable length parameters, "missing" parameters, etc. We then would be using the same engine which might also handle "macro commands" (I'm thinking of something like "deploy probe") which would provide a customization mechanism that would not require reflashing the firmware.

Roxy-3DPrintBoard, 2016-02-12:
> But, whatever the mechanism, we would be able to deal with variable length parameters, "missing" parameters, etc. We then would be using the same engine which might also handle "macro commands" 

Or...  Out of order parameters.   Right now the big problem between firmware revisions (or even different configurations of the same firmware) is not all the parameters are in the EEPROM in the same order.   

If we could be more tolerant of that fact, it would be straight forward to disable values stored in the EEPROM until they are explicitly turned on by the user.   And that would noticeably reduce support issues for the developers.

I'm not sure that the GCode commands need to be stored.  If we actually want GCodes to feed back in, we could fabricate them with a special parameter on the dump command. (After all...  We have the parameter's name and its value.)  But having the name and the value stored together would be a bet step toward being able to move seamlessly between versions.

jbrazio, 2016-04-23:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-23 by jbrazio

tuskiomi, 2018-06-25:
Link is a 404. Please re open,

thinkyhead, 2018-06-26:
@tuskiomi — Over two years have passed, and the work branches are now [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip) and [bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip).