MarlinFirmware/Marlin #2993: Duel extruder offset function fails on second and subsequent prints.

bruce356, 2016-02-11:
Marlin v1.1.0 RC3 Rambo with GLCD (REPRAP DISCOUNT FULL GRAPHIC SMART CONTROLLER with SD card) using S3D slicer.

Using the left hand hot end/extruder which in my case is tool1 (right hand is tool0) to perform a single extruder print, the part prints in the centre of the bed. (can be printed either with computer usb cable or through SD card the effect is the same).  Any subsequent prints are printed 33mm to the left of centre (33mm is the offset between the two hot ends).

If I turn off the printer and re-start between each print then the first print again is centred.
I am sure this is a Marlin bug and has nothing to do with the slicer.
Regards - bruce

bruce356, 2016-02-14:
Is no one else on planet earth having similar problems with Marlin 1.1.0 RC3?
Seems hard to believe that I am the only one
Regards - bruce

thinkyhead, 2016-02-28:
That's an interesting effect. The offset between extruders shouldn't be applied anyplace without a tool change that actually changes the tool. My guess is that the firmware is getting a `T0` command and applying the offset, even though the tool wasn't changed.

jbrazio, 2016-02-28:
Could it be related with the relative coordinate system (G90/91/92) ?

thinkyhead, 2016-02-28:
@bruce356 When you home the machine, are you setting the home XY position to the position of the right-hand nozzle tip? So, for example, if the right nozzle is at X=33mm, Y=0, when you home to an X-Min endstop and Y-Min endstop, the current position should end up being 33, 0 if you're homing Tool 1, and it should end up as 0,0 if homing Tool 0. (I am assuming that you want to be able to print on demand with the left nozzle, which, at left-home, has a position of X=0, but that is optional.) To configure for this, I believe you would need to enable `MANUAL_HOME_POSITIONS` and set `MANUAL_X_HOME_POS` to `33`. Then you would set 0 as your `X_MIN_POS` – the "farthest to the left" that your (should be leftmost) X carriage can move. With a dual nozzle setup the left nozzle can access farther to the left and the right nozzle can access farther to the right, and I'm not sure how considerate Marlin is about these possibilities. In general, it keeps only a single left limit applicable to the single "current position", but obviously this limit should differ depending on which nozzle is active. Finally, I'm not sure Marlin considers the possibility of Extruder 1 being to the left of Extruder 0.

Anyway, I don't want to get too far ahead of myself. Can you post the output of your `M503` command? And can you post the part of your `Configuration.h` where you set your movement limits and extruder offsets? And can you describe the behavior of doing `T0` and `T1` commands from Pronterface? Do they move the nozzles? Do they change the XY position? And, what is the shortest GCode you can issue that causes the position to get screwed up?

closed 2016-04-28 by jbrazio