MarlinFirmware/Marlin #2996: SDSUPPORT not workin

CoderSquirrel, 2016-02-12:
I'm using sdsupport and its not workin, i changed the pins, i'm using:

> MOSI_PIN = 51; 
> MISO_PIN = 50;  
> SCK_PIN = 52; 

but it dont worked.
SO in initsd i printed the pins and its showing 

> SCK_PIN 
> 4
> MISO_PIN 
> 2
> MOSI_PIN 
> 3

Blue-Marlin, 2016-02-12:
If that's the actual syntax of your pin file it can't work..
MOSI_PIN is not a variable.

CoderSquirrel, 2016-02-12:
I'm using a version of marlin that is am 1.0 old version and there, in the pins.h there's a MOSI_PIN;
http://www.fpaste.org/322063/

Blue-Marlin, 2016-02-14:
All i wanted to say is. MOSI_PIN is not a __variable__ but a __define__.
MOSI_PIN = 51;
is not a working syntax. I should read
#define MOSI_PIN 51

CoderSquirrel, 2016-02-14:
@Blue-Marlin okay i got it, but when i have used defines before in cpp and i try to print them, i could print the value. 
And in the pins.h i changed all the pins as

> #ifndef SDSUPPORT
> // these pins are defined in the SD library if building with SD support
> #define SCK_PIN           52
> #define MISO_PIN         50
> #define MOSI_PIN         51
> #endif

thinkyhead, 2016-02-21:
My best suggestion is to use a newer version of Marlin. The latest version is currently 1.1.0-RC3. Newer versions improve SD support and fix several other issues. Configuration is pretty much the same as it was in Marlin 1.0 through 1.0.2-1, with added warnings if something is missing.

closed 2016-04-23 by jbrazio