MarlinFirmware/Marlin #2997: Erroneous or Unclear Bed Level Correction Matrix - Multiple Marlin Versions

mirage335, 2016-02-13:
From the same probing sequence that shows excellent bed topography (<60um physical error), a totally invalid matrix seems to be calculated.

```
G29 Auto Bed Leveling
Eqn coefficients: a: -0.00009422 b: -0.00007075 d: -0.07714509
planeNormal x: 0.000094 y: 0.000071 z: 1.000000
 
Bed Height Topography: 
+-----------+
|...Back....|
|Left..Right|
|...Front...|
+-----------+
 +0.01688 -0.03562
 +0.00188 +0.01688

Bed Level Correction Matrix:
+1.000000 +0.000000 -0.000094 
-0.000000 +1.000000 -0.000071 
+0.000094 +0.000071 +1.000000 
```

Stated "+1.000000" value seems too large. Moreover, in many tests today, the same diagonal pattern was seen in the matrix, regardless of the actual physical pattern measured in the topography.

For reference, the same pattern seems to be documented under another related bug report - https://github.com/MarlinFirmware/Marlin/issues/1690 . Similar problems are also documented at the LulzBot Forums - https://forum.lulzbot.com/viewtopic.php?f=16&t=2123&start=30 . Earlier, the issue seems to have been partly blamed on large nonlinear deformations, suspected to be causing out-of-bounds calculations. Since the current topography is highly precise, this no longer seems to be the best explanation.

Additionally, the exact firmware configuration and code used for the above measurements is available as a github commit - https://github.com/mirage335/Marlin-TazStiff/tree/25fec840dda22e4a054fad463ee62c48022bb509 .

Is the bed level correction matrix always supposed to show a back-left to front-right +1 diagonal? How is this information supposed to be interpreted? Could the matrix be presented more intuitively, perhaps in absolute units of millimeters? Is there some configuration change myself and others are missing?

Blue-Marlin, 2016-02-13:
Please read a bit about vector/matrix multiplication and unity matrix. Then come back and close this issue.

mirage335, 2016-02-14:
Thanks for the helpful references. Use of an identity matrix is now obvious. If the compensation matrix does not directly reflect Z-compensation values used at various points across the computed plane, that makes sense.

However, it might still be useful to understand these values in terms of such, to more intuitively double-check how well the compensation is addressing this in a given situation. Perhaps processing the topographic data using the compensation functions would provide this. Or maybe a spreadsheet would be sufficient.

Consequently, please consider this as a feature request for a more direct/intuitive readout showing the improvement compensation is providing.

closed 2016-02-14 by mirage335

Blue-Marlin, 2016-02-14:
Have you ever tried the 'V' parameter of G29?
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L2764

mirage335, 2016-02-14:
Oh, didn't realize that provided "Corrected Bed Height vs. Bed Topology". Yes, that seems to do exactly what I want.

Thanks, and please pardon my ignorance.

Roxy-3DPrintBoard, 2016-02-14:
Besides the V parameter, the T gives a Topographical report of the bed.   It helps you see if you have the bed physically leveled as well as possible.   You want to do this before handing over the car keys to the Auto Bed leveling.