MarlinFirmware/Marlin #2999: Speed relative extrusion rates?

bradanlane, 2016-02-15:
While investigating an under extrusion issue on my printer, I found the following research:

http://www.extrudable.me/2013/04/18/exploring-extrusion-variability-and-limits/

It nearly perfectly describes my issue.

In summary, it states that "**under extrusion increases with print speed**."

This may be soothing to consider for Marlin. Then again, if this is wandering into the "over-reaching" territory, it may be better left to post processing.

Thoughts?

Sungod3000, 2016-02-15:
its an excellent test, that should be replicated more often. 
im having issues with these underextrusion patters as well. it mostly shows on top level infill when the lines are thicker towards the perimeter when the print speed slows down and the pressure builds up.

im not sure that if marlin is the place for that to be corrected especially since some slicers (slic3r, craftware) are implementing features that take this pressure build up into account.

i recently bought the bondtec extruder which reduced these patters quite a bit just by having more controll over the filament. but even with this there will always be pressure build up at some speeds.

thinkyhead, 2016-03-02:
The `ADVANCE` feature is supposed to obviate this issue, but from what I have heard it's not working right now. Fixing `ADVANCE` is a big item on my list. Currently gathering information to make sure I fully understand the concept and the code.

bradanlane, 2016-03-02:
Re: Slic3r

I had undesirable results with slic3r for a number of the prints I've had to work on so I've been bouncing from KISSlicer to Cura and now trying three more slicers. 

I was not clear on my ADVANCE would be setup so I took the kludge approach and just added a new thermistor profile. It worked well (even though I knew it would need to be re-applied with every update to Marlin).

closed 2016-05-04 by jbrazio