MarlinFirmware/Marlin #3002: Insane Acceleration and Jerk by default

dayssincethedoor, 2016-02-16:
In configuration.h the default values for acceleration and jerk are insane. This makes it very hard for someone who has never configured Marlin before to set sane values. I personally have spent the past six months hunting down extruder issues only to find out the problem was my E acceleration was way too high. The current default values are:

`#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 25}
#define DEFAULT_MAX_ACCELERATION      {3000,3000,100,10000}

#define DEFAULT_ACCELERATION          3000
#define DEFAULT_RETRACT_ACCELERATION  3000
#define DEFAULT_TRAVEL_ACCELERATION   3000

#define DEFAULT_XYJERK                20.0
#define DEFAULT_ZJERK                 0.4
#define DEFAULT_EJERK                 5.0`

I am not sure what the default values are based off of, but I don't believe any current hardware is capable of this (though I could be wrong). The values I personally using right now are:

`#define DEFAULT_MAX_FEEDRATE          {400, 400, 10, 30}
#define DEFAULT_MAX_ACCELERATION      {750,750,75,100}

#define DEFAULT_ACCELERATION          750
#define DEFAULT_RETRACT_ACCELERATION  750
#define DEFAULT_TRAVEL_ACCELERATION   75

#define DEFAULT_XYJERK                10.0
#define DEFAULT_ZJERK                 0.4
#define DEFAULT_EJERK                 0.10`

Not to say that my values are the best nor even close to proper for my machine, but they work. I purpose these values as the defaults:

`#define DEFAULT_MAX_FEEDRATE          {200, 200, 5, 20}
#define DEFAULT_MAX_ACCELERATION      {500,500,50,100}

#define DEFAULT_ACCELERATION          500
#define DEFAULT_RETRACT_ACCELERATION  500
#define DEFAULT_TRAVEL_ACCELERATION   500

#define DEFAULT_XYJERK                7.5
#define DEFAULT_ZJERK                 0.2
#define DEFAULT_EJERK                 0.10`

Based off my limited experience they seem to be on the lower end of sane values. I am not a mechanical engineer though so if any of them are still too high please adjust accordingly. The important thing is that the values should work on all printers without any changes by default. They may be slow as dirt but they should work. From there it is up to the end user to adjust tweak them to suit their equipment.

bkubicek, 2016-02-16:
this is absolutely not insane, for an ultimaker style printer its actually
quite conservative, i use 7000 very often.

Bernhard

CthulhuLabs, 2016-02-16:
Awesome. I am glad that they work on some better printers. The point is they should work on every printer by default. They should be the bare minimum. This way people who spent three hundred US dollars on a kit and do not know better don't spend months fighting with their equipment when the problem is in fact the firmware values (I am referring to myself).

brainscan, 2016-02-16:
These defaults are already lower than I use and I don't have a high end printer. The values you use seem very low and they would have caused me a lot of headaches if they were the default. They used to default to 9000.

Roxy-3DPrintBoard, 2016-02-17:
> Awesome. I am glad that they work on some better printers. The point is they should work on every printer by default. They should be the bare minimum. 

The defaults do work on 95% of the printers.   If you start spending your time helping to improve the Marlin code base, your opinion will have more of a say.   Until then...   You can live with our best guess of what makes sense.

closed 2016-02-17 by Roxy-3DPrintBoard

thinkyhead, 2016-02-21:
Historically, the default configurations were tuned for an Ultimaker printer. We've moved more towards a "reprap-level" of settings, but some of those historical defaults have remained. If you are like me, when I first started with RepRap machines I took the settings in the default Marlin configuration to be a sort of benchmark standard, which I thought my machine is supposed to reach. So I sympathize if the default settings put up an unrealistic standard. Only in recent months did I start to play with lower acceleration values, and I came to enjoy the results (and the motor sounds) a lot. Somewhere in our documentation we should definitely include a guideline on the kinds of numbers that users should expect for their machines.

porfiriosousa, 2017-05-16:
Hello .... it's true the values are usually very high .... on my corexy printer ... when printing to layer slip ... I already changed
Engines and drives and the problem kept .... so it can only be the speed of printing ....
This is a massacre .... I want the printer to work and there is always something to interrupt ...

thinkyhead, 2017-05-17:
For Marlin 1.1.0 we adopted a new Jerk algorithm, and I believe the values that applied to the old Jerk are actually 2x the values that would apply to the new Jerk. So let's start with that one setting first and see if tuning that alone is enough to make movement more reliable.