MarlinFirmware/Marlin #3003: G29 doesn't work on Delta with inductive sensor as probe

tigertooth4, 2016-02-17:
Hi all,

Has anyone tried G29 auto-bed leveling on Delta? Please help!
 
I have a DIY Kossel XL using Arduino + Ramps1.4 with a Fortek PS-05N inductive sensor mounted on my effector. The endstops for A, B and C tower were mounted at the top. The inductive sensor connected to Z-min pins on Ramps. I downloaded the RC3 version of Marlin 1.1 and modified configuration.h file using the Kossel pro's configuration as a template. 

After modified the configuration file and uploaded to Ramps, I did the following tests:

M119 endstop checking was fine, x,y,z endstops triggered when I move the carriages to the top. Inductive sensor shows "triggered" as z-min, when I moved the nozzle about 0.1mm higher above the bed. G28 was good. G1 moving was good too. My Z-height reading was correct after G28. It was about 217mm, and X, Y readings were all 0, as expected.

But when I type G29 in Pronterface (with a G28 beforehand), three carriages didn't move down towards the bed, but went up, crashed to the endstops.

Here's the G29 outputs:

SENDING:G29 V4

G29 Auto Bed Leveling

Bed X: 60.000 Y: -64.000 Z: 99.995

Bed X: 29.000 Y: -64.000 Z: 104.990

Bed X: -2.000 Y: -64.000 Z: 109.985

Bed X: -33.000 Y: -64.000 Z: 114.980

Bed X: -64.000 Y: -64.000 Z: 119.975

Bed X: -64.000 Y: -33.000 Z: 124.970

Bed X: -33.000 Y: -33.000 Z: 129.965

Bed X: -2.000 Y: -33.000 Z: 134.960

Bed X: 29.000 Y: -33.000 Z: 139.955

Bed X: 60.000 Y: -33.000 Z: 144.950

Bed X: 91.000 Y: -2.000 Z: 149.945

Bed X: 60.000 Y: -2.000 Z: 154.940

Bed X: 29.000 Y: -2.000 Z: 159.935

Bed X: -2.000 Y: -2.000 Z: 164.930

Bed X: -33.000 Y: -2.000 Z: 169.925

Bed X: -64.000 Y: -2.000 Z: 174.920

Bed X: -64.000 Y: 29.000 Z: 179.915

Bed X: -33.000 Y: 29.000 Z: 184.910

Bed X: -2.000 Y: 29.000 Z: 189.905

Bed X: 29.000 Y: 29.000 Z: 194.900

Bed X: 60.000 Y: 29.000 Z: 199.895

Bed X: 60.000 Y: 60.000 Z: 204.890

Bed X: 29.000 Y: 60.000 Z: 209.885

Bed X: -2.000 Y: 60.000 Z: 214.880

Bed X: -33.000 Y: 60.000 Z: 219.875

Bed X: -64.000 Y: 60.000 Z: 224.870

Bed X: -2.000 Y: 91.000 Z: 229.865

110.09 115.08 100.09 85.11 70.12 55.14 60.13

80.12 120.08 115.08 110.09 105.09 100.09 130.07

120.08 125.07 130.07 135.06 140.06 145.05 150.05

180.01 175.02 170.02 165.03 160.03 155.04 150.05

175.02 180.01 185.01 190.00 195.00 199.99 204.99

190.00 224.97 219.97 214.98 209.98 204.99 239.95

264.93 269.92 254.94 229.96 224.97 219.97 214.98

These data was coming out during three carriage crashing to the endstops, in about 1 second. Data should mean something, but I could not figure it out.

I'm using AUTO_BED_LEVELING_GRID mode, AUTO_BED_LEVELING_GRID_POINTS was set to 7.

`#define Z_RAISE_BEFORE_HOMING 4 `
 
`#define XY_TRAVEL_SPEED 4000 `    

`#define Z_RAISE_BETWEEN_PROBINGS 5` 

`#define Z_RAISE_AFTER_PROBING 15`  

I cannot upload my entire configuration.h file. But it can be downloaded in this post [https://groups.google.com/forum/#!topic/deltabot/IL1aZzLSCZs](url)

So my question is: Is there something wrong in my configuration ?  Does anyone have tested Marlin 1.1- RC3 auto bed leveling worked for Delta printer?  

Thanks in advance!

Best,
XJ

brainscan, 2016-02-17:
Does your sensor definitely switch between open and triggered? I think some people had this problem if it was always triggered.

brainscan, 2016-02-17:
Also cannot get that link to do anything, is there a reason you can't post your configuration.h?

tigertooth4, 2016-02-17:
@brainscan Sorry about that, I updated the previous post, the url should work now, please see my previous post, or, here: [https://groups.google.com/forum/#!topic/deltabot/IL1aZzLSCZs](https://groups.google.com/forum/#!topic/deltabot/IL1aZzLSCZs)

My sensor was ok. I defined it as z-min, it was normally open. I can check by M119. When I move the nozzle to the metal bed I can see it switch to triggered. 

I didn't know why I cannot upload, it kept saying "Something went really wrong, and we can't process that file".

brainscan, 2016-02-17:
Strange that you can't upload your file, I have no idea on that one. I don't think after homing you will have 100mm of clearance, you have raise probe 100mm before probing so it will try and go up even though it's already at the top. Sorry if that's not the problem, I'll keep looking but I haven't finished my delta yet so I'm slightly blind in this area. I'm sure it works on a delta though as Roxy uses one.

Blue-Marlin, 2016-02-17:
No unused endstop shall be triggered!
If they are, change 
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
to true or visa versa.
Even better would be to disable them at all
```
  #undef X_MIN_PIN
  #undef Y_MIN_PIN
  #define X_MIN_PIN          -1
  #define Y_MIN_PIN          -1
```

tigertooth4, 2016-02-17:
@brainscan I thought the raise probe distance was absolute z-height above the bed, therefore 100mm was in the range, since my z-max height was 217.

I remember I also tried RAISE_PROBE 0.1 (or negative 0.1), the output then changed from 100 to that number. But the motion was the same, carriage went up and crash the endstops.

tigertooth4, 2016-02-17:
@Blue-Marlin Will this cause the crashing? I'm not sure. Could you explain a bit more? Bc G28 was always fine in my case.

brainscan, 2016-02-17:
I'm pretty sure z raise is from current position but I could be wrong, try setting it to 0 and see what happens. As deltas home to max it's different from a Cartesian that normally homes to min so needs to have a raise to avoid hitting the bed/clips etc. Also X and y offsets are supposed to be integers, but I'm not sure if that still applies as certain things have changed over the last few months and I don't think that would be causing this problem. 
Will keep looking though.

Wackerbarth, 2016-02-17:
I wouldn't set it to zero. Allow a few mm to avoid triggering the sensor during movement between probe locations.

You might also try defining the pin to be a "probe" rather than a "min end stop"

Richard

brainscan, 2016-02-17:
Surely the raise is only performed once before probing starts? The raise between probes would stop accidental triggering. Or have I misunderstood this for the last couple of years?

Wackerbarth, 2016-02-17:
But you need to assure that the probe is not already triggered when you start the first probe sequence.

In the future, rather than raising, I would like to see a "minimum height" approach.

Richard

brainscan, 2016-02-17:
So given that a delta homes away from the bed at its max, how is it going to accidentally trigger the probe? Why would it need to move further away from the bed when it's already over 200mm away? I understand why when using a Cartesian machine this is needed, but this is a delta, and I had already asked about the probe being open or triggered so I'm not following your point.

Blue-Marlin, 2016-02-17:
What does M119 really show in mid air? What's your setting for `#define ENDSTOPS_ONLY_FOR_HOMING`

Roxy-3DPrintBoard, 2016-02-17:
> No unused endstop shall be triggered!

And in the case of probes, the state of an stowed probe should not be looked at either.  I think these code corrections are in RCBugFix.   If TigerTooth4 is not using RCBugFix, it might be wise to move to that.

AnHardt, 2016-02-17:
@Roxy-3DPrintBoard 
Good joke. :-(

Roxy-3DPrintBoard, 2016-02-17:
If these changes are not in RCBugFix...  They should be.

Roxy-3DPrintBoard, 2016-02-17:
@thinkyhead    Can you get the End Stop and Probe changes merged into RCBugFix?

tigertooth4, 2016-02-18:
@Roxy-3DPrintBoard I'm not using RCBugFix. I will give it a try later.

tigertooth4, 2016-02-21:
Hey Guys,

Today I tried RCBugFix. It has the same problem, no luck! @Roxy-3DPrintBoard @Blue-Marlin @Wackerbarth @brainscan @thinkyhead  
I opened debug mode, with M111 and then tried G29 again. I also added some serial output to marlin_main.cpp in order to gather more information. 

Now I believe this problem is related to the malfunctions of do_blocking_move_to() function and prepare_move_raw() function. I my case, do_blocking_move_to() does not move anything! So does prepare_move_raw(). And it seems these moving commands are only used in DELTA + autobed leveling part. G1 doesn't use these, Am I correct?

I'll check more carefully on this part of marlin_main. I have a question on the logic of code: Which function should perform the real movement? Prepare_move does not right? It only plans the move. And st_***() really do let the stepping motor move, right?

Roxy-3DPrintBoard, 2016-02-21:
do_blocking_move_to() is used as a utility function to help routines and
sub-commands move the nozzle to where it needs to be.   It is not what is
used by the G1 command.   G1's movement is controlled much tighter.

Blue-Marlin, 2016-02-21:
`do_blocking_move_to()` is indeed only defined with autoleveling - but not only for DELTAs.
The real movement is done in the stepper interrupt. See stepper.cpp.
I still suppose your X- and Y-MIN endstops are triggered.

tigertooth4, 2016-02-21:
Sorry guys! My mistake! @Blue-Marlin You are right! My X- and Y-MIN are always triggered. I inverted their settings, now G29 works fine! 

Thanks every one! I learned a lot!

closed 2016-02-22 by tigertooth4