MarlinFirmware/Marlin #3008: Depression from Autolevel? (Delta Printer)

Coltan89, 2016-02-19:
I am using a kit that I bough from Geeetech and its a Delta printer. They use the Marlin software and has been a bit of a struggle but I'm making progress. So after much trial and plenty of errors, I have noticed that the Auto-level functions creates a depression from the autolevel. 

**Illustration**
[image]

So from what I have gathered. The auto level function starts roughly around -10x, -60y to x70, 60y (rough estimations) 
I am using the provided (marlin)software from Geeetech with the modifications and general numbers. Is there a way to shift the auto level functions cover from -50x, -50y to 50x, 50y?  That may help resolve the depression that the auto level is coming up with. I should mention that I am using a proximity sensor to detect the height. Any suggestions at this point would be helpful. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/137409/Configuration.zip)

Roxy-3DPrintBoard, 2016-02-19:
I bought the same printer from Geeetech.   My suggestion is to go to the MarlinDev branch and use the G27 Delta Physical Bed Leveling command to get your bed perfectly flat and to get the Delta Radius perfectly dialed in.    And then, for right now...   Don't use the Auto Bed Leveling.   Geeetech did a really poor job on the firmware.   That is a super old branch of Marlin and the butchered it up.

I gave up on the review I was doing:   But check out:  http://3dprintboard.com/showthread.php?16024-Geeetech-G2s-Printer-Kit-Review

Coltan89, 2016-02-19:
They certainly did mess with the Marlin software, but they did come out with a recent revision back in September of last year. That is right around the time where I had to walk away from the printer or else I was going to break something. Any who, that is what I started to do last night after posting the issue on here, but a new issue appeared which was one of my stepper motors was binding up and eventually locked up completely. I think that have been one of my issues so with a bit more research tonight, I should be able to print something minus the Auto-level. 

I will look into the G27 and see what I can massage to get to work later on. I just want to see this thing in action. Thanks for your help. 

P.S. I have already heavily modified my Geeetech Delta printer. Ill upload some photos later on once It is all complete. It maybe of some use to folks if they are interested.

ihouses, 2016-02-19:
Hello, I have a chinese delta printer that uses an old marlin version. So far I can only calibrate the bed using the M666 manually. I can not find any information about the command G27 that you are talking about. Is there any good source information about delta autolevel for the lastest version of Marlin?

Thanks.

Roxy-3DPrintBoard, 2016-02-19:
It is in the Development tree:  https://github.com/MarlinFirmware/MarlinDev/tree/Feb14/gCodes

You will need to follow the instructions in the Wiki to prepare the development environment for it.

The instructions (and capability) of the G27 are documented at the top of the source code.

ihouses, 2016-02-19:
Oh thanks, I will take a look

thinkyhead, 2016-03-02:
Some stuff like this will be making it into 1.1.1, but not 1.1.0. So until this release is completed we must use `MarlinDev` to get `G27`.

closed 2016-05-04 by jbrazio