MarlinFirmware/Marlin #3009: M23 file select issue filename shortened, individual delta diagonal rod adjustments

HedronUser, 2016-02-19:
**Background:** I've resolved both issues separately but haven't found a way to resolve them both together. I want to know HOW these issues were solved so I can implement the fix(es) in a version of marlin which either has the delta geometry adjustments I want i.e. i.e. M666 U, V, W (individual diagonal rod adjustments) or proper SD card file selection protocol. For example, Marlin RC does not have the extra geometry adjustments to my knowledge, nor will it recall or fully list the M666 delta geometry parameters. If someone knows how to add this functionality please chime in.

**Problem 1:** The filename on the SD card gets truncated to XXX~1.GCO and/or the file will not select using M23.
**Observations:** It doesn't matter what filename I try.  I tried using short file names with only caps i.e. GOOD.GCODE and different extensions like *.GCO, *.GCD, *.G without success. 
I attempted adjusting the starpos code in Marlin_main.cpp according to a post I found without success. I updated to Marlin RC and this solves the problem (BUT HOW???)

**Problem 2:** The M666 delta geometry support for Marlin RC is incomplete, i.e. no individual delta diagonal rod or alpha rotation adjustment. I found [a post](https://github.com/MarlinFirmware/Marlin/issues/2773), 
on the topic but it doesn't mention individual diagonal rod (M666 U V W), and it uses M665 instead of M666.

WillSmithInChina, 2016-02-26:
you should add space key between "M23" and "XXX~1.GCO" ,but i also meet the problem of "XXX~1.GCO" and i can't resolved it , if you know that,please tell me , thanks guys.

WillSmithInChina, 2016-02-26:
i guess the answer to get correct file name  is the command of "M33", but when i enter the command of "M33" ,it  also doesn't work

WillSmithInChina, 2016-02-26:
and you should update the fireware of Marlin so that you use the command of "M33"

Blue-Marlin, 2016-02-26:
Have you tried do define LONG_FILENAME_HOST_SUPPORT in Configuration_adv.h

WillSmithInChina, 2016-02-26:
thanks for your regards

jbrazio, 2016-02-27:
Could this be related with #3039 ?

HedronUser, 2016-02-28:
Hmm, ok @jbrazio I made the mistake of publishing the problem #1 stated above as an issue in RC, its not. However, problem #2 stated above is an issue and I want it resolved.

HedronUser, 2016-02-28:
I realize this is just a repeat of issue #2773

closed 2016-02-28 by HedronUser

thinkyhead, 2016-02-28:
> I realize this is just a repeat of issue

Glad to be patching that. We'll stop seeing these as soon as RC4 is ready. Items in the Pull Requests list still need testing before we can merge them. Most are relatively obvious bug fixes. In the meantime, `RCBugFix` is the best place to get the patched code – for example, this issue is now patched there.

WillSmithInChina, 2016-03-02:
when the count of files on SD Card is few, i can get all the correct file name by M23, but when the count of files is many, i just get a part of file name ,do you meet the same problem ? in additon, which USB Serial Driver do you use ?  if you know that, please tell me

WillSmithInChina, 2016-03-02:
when the count of files on SD Card is few, i can get all the correct file name by M23, but when the count of files is many, i just get a part of file name ,do you meet the same problem ? in additon, which USB Serial Driver do you use ? if you know that, please tell me  @HedronUser