MarlinFirmware/Marlin #3011: Yikes!  M48 never got fixed for Delta's!

Roxy-3DPrintBoard, 2016-02-20:
The M48 command has never worked with Delta's.    This updated M48 code works for both Cartesian and Delta Printers.   Can we add it to RCBugFix?

~~~~
#if ENABLED(USE_PROBE)

// This is redundant since the Sanity Check already checks for a valid Z_MIN_PROBE_PIN, but here for clarity.
#if ENABLED(Z_MIN_PROBE_ENDSTOP)
  #if !HAS_Z_PROBE
    #error You must define Z_MIN_PROBE_PIN to enable Z probe repeatability calculation.
  #endif
#elif !HAS_Z_MIN
  #error You must define Z_MIN_PIN to enable Z probe repeatability calculation.
#endif

/**
 * M48: Z probe repeatability measurement function.
 *
 * Usage:
 *   M48 <P#> <X#> <Y#> <V#> <E> <L#>
 *     P = Number of sampled points (4-50, default 10)
 *     X = Sample X position
 *     Y = Sample Y position
 *     V = Verbose level (0-4, default=1)
 *     E = Engage Z probe for each reading
 *     L = Number of legs of movement before probe
 *     S = Schizoid (Or Star if you prefer)
 *
 * This function assumes the bed has been homed.  Specifically, that a G28 command
 * as been issued prior to invoking the M48 Z probe repeatability measurement function.
 * Any information generated by a prior G29 Bed leveling command will be lost and need to be
 * regenerated.
 */
void gcode_M48() {
  double sum = 0.0, mean = 0.0, sigma = 0.0, sample_set[50];
  uint8_t verbose_level = 1, n_samples = 10, n_legs = 0, schizoid_flag = 0;
  if (verbose_level > 0)
    SERIAL_PROTOCOLPGM("M48 Z-Probe Repeatability test\n");
  if (code_seen('V')) {
    verbose_level = code_value_short();
    if (verbose_level < 0 || verbose_level > 4) {
      SERIAL_PROTOCOLPGM("?Verbose Level not plausible (0-4).\n");
      return;
    }
  }
  if (verbose_level > 0)
    SERIAL_PROTOCOLPGM("M48 Z-Probe Repeatability test\n");
  if (code_seen('P')) {
    n_samples = code_value_short();
    if (n_samples < 4 || n_samples > 50) {
      SERIAL_PROTOCOLPGM("?Sample size not plausible (4-50).\n");
      return;
    }
  }
  float  X_current = current_position[X_AXIS],
         Y_current = current_position[Y_AXIS],
         Z_current = current_position[Z_AXIS],
         // E_current = current_position[E_AXIS],
         X_probe_location = X_current + X_PROBE_OFFSET_FROM_EXTRUDER,
         Y_probe_location = Y_current + Y_PROBE_OFFSET_FROM_EXTRUDER,
         Z_start_location = Z_current + Z_RAISE_BEFORE_PROBING;
  bool deploy_probe_for_each_reading = code_seen('E');
  if (code_seen('X')) {
    X_probe_location = code_value();
    if (X_probe_location < X_MIN_POS || X_probe_location > X_MAX_POS) {
      out_of_range_error(PSTR("X"));
      return;
    }
  }
  if (code_seen('Y')) {
    Y_probe_location = code_value();
[... 154 lines cut ...]
        SERIAL_PROTOCOL_F(mean, 6);
        SERIAL_PROTOCOLPGM("   sigma: ");
        SERIAL_PROTOCOL_F(sigma, 6);
      }
    }
    if (verbose_level > 0) SERIAL_EOL;
    delay(50);
    do_blocking_move_to_z(current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS);
  }  // End of probe loop code
  // raise_z_after_probing();
  if (verbose_level > 0) {
    SERIAL_PROTOCOLPGM("Mean: ");
    SERIAL_PROTOCOL_F(mean, 6);
    SERIAL_EOL;
    delay(25);
  }
  SERIAL_PROTOCOLPGM("Standard Deviation: ");
  SERIAL_PROTOCOL_F(sigma, 6);
  SERIAL_EOL; SERIAL_EOL;
  delay(25);
  clean_up_after_endstop_move();
}

#endif
~~~~

Blue-Marlin, 2016-02-20:
Hurray!

You got rid of st_get_position.
sin() and cos() work on RAD.
A standard random function.

Congratulations. :+1:

thinkyhead, 2016-02-28:
> Hurray!

Are you saying that I fixed `M48` as an extra consequence when I replaced `st_get_position_mm` with `st_get_axis_position_mm` in that CoreXY bug fix a couple weeks ago?

thinkyhead, 2016-02-28:
Should this code not still be included…?
```cpp
  if (code_seen('X')) {
      X_probe_location = code_value() - X_PROBE_OFFSET_FROM_EXTRUDER;
  ...
```
It's a question of user expectation. Up to now `M48` has used XY to set the probe position. Should it instead set the nozzle position to XY, probe, and apply the XY offset when it saves the probed point? Or, should it continue to probe to the given XY position?

thinkyhead, 2016-02-28:
Also… Is there a separate PR hanging around that introduces `DELTA_PROBEABLE_RADIUS`?

Roxy-3DPrintBoard, 2016-03-03:
> It's a question of user expectation. Up to now M48 has used XY to set the probe position. Should it instead set the nozzle position to XY, probe, and apply the XY offset when it saves the probed point? Or, should it continue to probe to the given XY position?

Yes...  This is worth discussion.   Originally the code looked like this:
~~~~
    if (code_seen('X') || code_seen('x')) {
      X_probe_location = code_value() - X_PROBE_OFFSET_FROM_EXTRUDER;
      if (X_probe_location < X_MIN_POS || X_probe_location > X_MAX_POS) {
        SERIAL_PROTOCOLPGM("?Specified X position out of range.\n");
        return;
      }
    }
~~~~
Now, it has been updated to be:
~~~~
  if (code_seen('X')) {
    X_probe_location = code_value();
    if (X_probe_location < X_MIN_POS || X_probe_location > X_MAX_POS) {
      out_of_range_error(PSTR("X"));
      return;
    }
  }
~~~~
My feeling is most users want to measure things relative to where the nozzle is.   But this is a function that is focused on the Z_Probe, so I can see where there will be some debate.   What ever makes the most sense for the user is fine by me.

> Also… Is there a separate PR hanging around that introduces DELTA_PROBEABLE_RADIUS?

I may have made a bad assumption here.   But every Configuration.h file I've seen setup for a Delta printer has #defined that value and usually as a function in terms of the DELTA_PRINTABLE_RADIUS.   Some advice on what makes sense here would be appreciated.   I'm making use of DELTA_PROBEABLE_RADIUS in the new High Resolution Delta Mesh Leveling code also.    Maybe we should consider enforcing this #define exists if DELTA is #define'd ????

What are other people's thoughts on this?

thinkyhead, 2016-03-03:
> Now, it has been updated to be:
```cpp
  if (code_seen('X')) {
    X_probe_location = code_value();
    if (X_probe_location < X_MIN_POS || X_probe_location > X_MAX_POS) {
      out_of_range_error(PSTR("X"));
      return;
    }
  }
```

So what I see here is that the position given is *still* the probe position. Because the probe offset is applied later on in the code, so that when actually moving to the points to probe the nozzle moves to a position such that the probe is at the given XY.

Given that this is the case…
```cpp
-    if (X_probe_location < X_MIN_POS || X_probe_location > X_MAX_POS) {
+   if (X_probe_location < MIN_PROBE_X || X_probe_location > MAX_PROBE_X) {
```

Roxy-3DPrintBoard, 2016-03-03:
Oh!   I agree it should be changed.   And of course...  The 'Y' version of the code right below it should get a similar change.   

Thank You for finding that!!!

thinkyhead, 2016-03-03:
Do we have any Delta machines available to test #3029? I keep begging online, but so far no one has sent me their spare Deltabot!

Roxy-3DPrintBoard, 2016-03-03:
We probably need to do something to encourage the different machine type vendors to help the developers and testers have the right hardware.   But that is going to be a very big and complicated discussion.   Let's do that in its own thread.

closed 2016-03-27 by thinkyhead