MarlinFirmware/Marlin #3022: Ability to follow cubic Bézier splines

giomasce, 2016-02-22:
Please excuse me if this has already been raised and possibly decided. I could find no traces anywhere. Currently Marlin can follow only paths shaped in form of a segment or a circle arc. I would like to have support for cubic Bézier splines as well. I use a lightly patched version of Marlin to run my laser cutter, which I often feed with Gcode generated by Inkscape. Inkscape and many other vector graphics programs naturally produce cubic splines. While approximation with arcs is possible, it is rather complicated, with a lot of annoying corner cases and a few possible numerical problems (approximation can easily change a very small arc of circle to a complete circle without a small arc). Instead Bézier splines are very stable and easy to compute.

As far as I know, Gcode does not have codes for representing splines. However, given that Marlin already uses a lot of not-completely-standard codes, I do not think that adding another would be a problem.

If it is ok for you, I can try to write a patch to add support and then submit it here. What are your views on this?

Roxy-3DPrintBoard, 2016-02-22:
In order for this to be useful, we would also need to get support in the various slicer's.   If you get a prototype working well, we can see what the slicer people think with regards to supporting (and using) it.

thinkyhead, 2016-02-22:
G2/G3 arcs are supported by some slicers, but not all. The slicer uses a simple heuristic to determine whether a series of straight lines forms a circular arc and replaces those line segments with a G2 or G3 command. Internally, these arcs are calculated on the fly by Marlin, which uses `sin` and `cos` functions to calculate the points from the current position, a given center-point-offset (`I`, `J`), and the destination position (`X`, `Y`, `Z`, `E`). A G2/G3 circle is an arc that starts and ends at the same XY position.

Bezier splines are a lot more complicated than simple arcs, and would require more calculation than you'd probably want to do with "single" precision on an 8-bit AVR processor. Moreover, there's no simple way to describe Bézier splines through GCode.

If you have a design idea and think you can come up with a good implementation, I would be interested to see it. But, as Roxy points out, even though you may be able to get Marlin to calculate Bézier curves, the hardest part will be to produce algorithms for slicer software that allow them to figure out when a series of straight line segments can be recast as a complex curve, and to get slicer software to adopt the algorithm and new GCode.

giomasce, 2016-02-23:
Quite the contrary, I would say: Bézier curves are very easy to evaluate with the [De Casteljaus's algorithm](https://en.wikipedia.org/wiki/De_Casteljau%27s_algorithm), which just requires you to perform multiplications and additions. This is far lighter than evaluating trigonometric functions, although I must admit that I do not know the exact algorithm used for `sin` and `cos` by the AVR compiler.

As for the potential users of the new feature: as I said, I am interested in it and would use it. My use case is rather atypical for Marlin, sure, because I use it for running a laser cutter. Maybe compilation flags can be set up, so that users that are not interested in this do not lose time and space on it.

As soon as it will be possible to me I will try to draft an implementation. If it is not accepted in upstream Marlin, I will keep it in my fork.

brainscan, 2016-02-23:
I'd be interested in seeing how this goes, I've been toying with adding a laser so this could be useful.

giomasce, 2016-02-28:
I have written a first working implementation, which is [here](https://github.com/giomasce/Marlin/commit/17930f965329847e48f79f970509cc8587746136). I used it to perform a few jobs with my laser cutter, which came out finely. The G-code was generated with [my Inkscape plugin](https://github.com/giomasce/thlaser-inkscape-plugin) (which, BTW, is rather messy at this point and has a few bugs, but can be used if you know the right configuration and caveats).

Remarks:
* The commit is based on a rather old version of Marlin, which is the one that I am using at the moment on my laser cutter; I could not build more the development version, because it requires a recent version of the Arduino IDE, which in turn is complicated to install on my OS; however, once the implementation is good and is accepted it should not be difficult to port it to the current version of Marlin.

* The code is not ready yet; the biggest problem at the moment is that the number of divisions of the B-spline is hardcoded; I will work on having it adapting depending on the curvature of the line.

* The G-code for Bezier-splines is actually standardized (at least in [this document](http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G5-Cubic-Spline)) with the code G5. I adhered to the specification there, except for the fact that I did not implement I and J omission and I used zero as default value for unspecified parameters.

thinkyhead, 2016-03-02:
> the [De Casteljaus's algorithm](https://en.wikipedia.org/wiki/De_Casteljau%27s_algorithm)

Ah, the _De Casteljaus's algorithm_ — of course! :dizzy_face: 

> The commit is based on a rather old version of Marlin

Not a problem. As long as the commits aren't too ugly I'm sure they can be reworked for the 47 "current versions" of Marlin. I look forward to examining the code and integrating it. But I think it will not get into 1.1.0. It can go into 1.1.1 and 1.2.0 however.

giomasce, 2016-03-13:
I produced a better revision of my algorithm, which should be relatively close to something actually usable. Here it is: https://github.com/giomasce/Marlin/commit/45e9fede688db11d5cf163878ece774c22d91b44.

The caveats today are:

* I have not tested it yet on my machine. I will do that as soon as possible, especially to tune the parameters and check that they do not an eccessive load on the board.

* However I produced a C++ snippet that can be used to test the output of the segmentation algorithm, which is here: http://pastebin.com/2XuDrdEw. In this code the assert call are uncommented, so a few formal postconditions in the algorithm can be tested properly.

* There is a known bug: the Z and E axis are interpolated linearly with the parameter of the Bézier curve; this is wrong, because the parameter is not linear in the distance on the curve. This is not a problem in my case, because I never move Z and E. In general fixing this would require pre-computing the length of the Bézier curve, which would lead to a rather long preprocessing (the whole curve must be simulated first to compute its length, then simulated again to be actually followed). It should be possible to mitigate this by using algorithms that decompose a curve in many smaller Bézier curve, but I have not thought about the CPU and memory requirements this would have.

giomasce, 2016-03-17:
I tested the code that I posted in my previous comment, trying both test shapes and real work, and it appears to work correctly.

Roxy-3DPrintBoard, 2016-03-17:
> Ah, the De Casteljaus's algorithm — of course! :dizzy_face:

I had to look it up.  It is cool!  

https://en.wikipedia.org/wiki/De_Casteljau%27s_algorithm

> It should be possible to mitigate this by using algorithms that decompose a curve in many smaller Bézier curve, but I have not thought about the CPU and memory requirements this would have.

Do you have any idea how a big Bezier curve gets broken up into smaller sections?  Can you guess how many sections your code is splitting a 5" curve into?   I started looking at the math.   My thoughts are the recursion is probably OK if you can pick which part of the curve you are calculating first.   Specifically, if you can split off and calculate the next tiny movement of the nozzle from the bigger curve, you will have time to split off the next piece (from the bigger curve) while the nozzle is moving and doing its thing on the first piece.

Conversely, if you need to calculate everything before you can get the small piece you need, I think you are going to cause Delta's to stutter.   And very possibly you will cause Cartesian machines to stuttuer.

giomasce, 2016-03-18:
You can use the program I posted via PasteBin to know exactly how many and which divisions my program will generate. However, I agree that my code might be a bit obscure, so I will post an update version with more comments ASAP.

At the moment the code does not need to compute the whole curve in advance to begin. It can compute it as long as it draws it. However, since it does not know in advance the length of the curve, it is not able to feed the Z and E axes proportionally to the length of the curve that has already been drawn. It will feed them proportionally to the parameter `t`. This is not a problem for who, like me, use Marlin just for a laser cutter, which does not require Z and E. If you want to fix this behavior, then you have to know the length of the curve in advance, which requires to simulate the whole drawing before actually starting it. Unfortunately computing the length of Bézier curve is not easy at all. Integrals are mean!

giomasce, 2016-03-20:
BTW, my commit https://github.com/MarlinFirmware/Marlin/commit/3cd597acac4524ea585cb849781e957e5ff2e2b9 adds some documentation to the code, hopefully making it clearer. Moreover https://github.com/MarlinFirmware/Marlin/commit/e0f45b65d91377b64d731c8fd5b9d73749ff6d57 adds a small optimization.

giomasce, 2016-04-03:
I pushed another commit that fixes a small bug occurring occasionally, that led to an infinite loop. The commit is https://github.com/giomasce/Marlin/commit/1533b33c2478bde56b8e059bcbd7b0884677676f.

thinkyhead, 2016-04-10:
The algorithm looks very cool. I didn't see any GCode portion, so how would a Bézier curve be expressed in GCode?

As we're starting to get more requests to make Marlin more friendly for laser cutter and CNC applications, Bézier curves begin to make more sense to support in some way. As you point out, all it takes is one Inkscape plugin and an extension to Marlin.

giomasce, 2016-04-11:
I used http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G5-Cubic-Spline as the reference for the G-code syntax (this is noted in the code comments). However, differently from that specification, I and J omission is not supported, and all control values can be omitted (and default to zero).

thinkyhead, 2016-04-12:
I don't mind keeping some of the same conditions specified for `G5` if it's workable. Such as requiring both `I` and `J` if one of them is specified, and allowing their omission if that's not too much trouble.

giomasce, 2016-04-12:
Changing any behavior with respect to that should not be difficult. I did not do that mostly because I do not know how to properly send out errors (and I also believe this may have changes with the current version of Marlin; my patch is based on a rather old version, the one that I use with my laser cutter).

thinkyhead, 2016-04-13:
@giomasce Marlin is kind of informal about how it throws errors with things like GCode arguments. The code can be as loose or uptight as makes sense in each case. So in the case of `G5`, if you specified `I` but left out `J` it might print a human-readable error to serial like `SERIAL_ERROR_START; SERIAL_ERRORLNPGM("G5 missing J arg");` but still run the command, using `0` for the missing `J` value. Or, it might print the message but not run the command. (I suppose since `G5` is well-specified, Marlin should skip the command when it's not well-formed.)

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better.
You should re-open this topic on [MarlinFirmware/MarlinDev](https://github.com/MarlinFirmware/MarlinDev) for proper followup from the development team. We suggest you to use the `MarlinFirmware/Marlin#<topic id>` markdown to link back to this original topic.

closed 2016-05-04 by jbrazio

bubnikv, 2021-07-27:
> * There is a known bug: the Z and E axis are interpolated linearly with the parameter of the Bézier curve; this is wrong, because the parameter is not linear in the distance on the curve. This is [...]

This is IMHO quite a significant issue for FDM 3D printing, which is not being documented in
https://marlinfw.org/docs/gcode/G005.html

ellensp, 2021-07-27:
@bubnikv your quoting stuff that is 5 years old!
Marlin 2 is a complete rewrite since then. It is very unlikely this issue still exists.

bubnikv, 2021-07-27:
I checked Marlin master source code.