MarlinFirmware/Marlin #3025: y axis distance/speed per unit randomly changes

dgm3333, 2016-02-23:
I've tried switching hardware (pololus, cables, motors) around with no change - whichever hardware is running the y-axis replicates the behaviour.

Error is:-
Randomly, movement codes (eg using home gcode (G28 Y0) or G1 Y1)  causes the Y axis to subsequently move approx 10x further and faster with each step (so I wonder if units per step is being incorrectly altered). 
Thereafter relative and absolute movements for the y axis are incorrect (eg using G1 Y1) with modes G90 or G91.
Randomly it sometimes seems to correct. This is not triggered by manually setting the motors to idle (M84). An example is the first movement in the code sequence at the end of this post (copied from repetier log) was incorrect, and subsequent movements reset so distance moved was correct.
This hasn't happened with other axes.

Previous firmware (which worked fine) was v1 rc2

My setup is: 
Using marlin 1.1.0 rc3
Repetier host 1.6.1 (and 1.6.0)
Windows 10
Sanguinolulu board 1.3

09:32:20.524 : N643 M105*38
09:32:23.587 : N644 M105*33
09:32:26.650 : N645 M105*32
09:32:29.712 : N646 M105*35
09:32:32.778 : N647 M105*34
09:32:35.841 : N648 M105*45
09:32:38.898 : N649 M105*44
09:32:39.181 : N650 G91*18
09:32:39.185 : N651 G1 Y0.1 F4800*22
09:32:39.201 : N652 G90*17
09:32:39.644 : N653 G91*17
09:32:39.650 : N654 G1 Y0.1 F4800*19
09:32:39.665 : N655 G90*22
09:32:39.949 : N656 G91*20
09:32:39.953 : N657 G1 Y0.1 F4800*16
09:32:39.969 : N658 G90*27
09:32:40.340 : N659 G91*27
09:32:40.353 : N660 G1 Y0.1 F4800*20
09:32:40.370 : N661 G90*17
09:32:40.565 : N662 G91*19
09:32:40.577 : N663 G1 Y0.1 F4800*23
09:32:40.593 : N664 G90*20
09:32:40.789 : N665 G91*20
09:32:40.802 : N666 G1 Y0.1 F4800*18
09:32:40.818 : N667 G90*23
09:32:40.965 : N668 G91*25
09:32:40.977 : N669 G1 Y0.1 F4800*29
09:32:40.993 : N670 G90*17
09:32:41.301 : N671 G91*17
09:32:41.313 : N672 G1 Y0.1 F4800*23
09:32:41.329 : N673 G90*18
09:32:41.501 : N674 G91*20
09:32:41.505 : N675 G1 Y0.1 F4800*16
09:32:41.521 : N676 G90*23
09:32:41.668 : N677 G91*23
09:32:41.681 : N678 G1 Y0.1 F4800*29
09:32:41.697 : N679 G90*24
09:32:41.837 : N680 G91*31
09:32:41.842 : N681 G1 Y0.1 F4800*27
09:32:41.857 : N682 G90*28

closed 2016-02-23 by dgm3333

Blue-Marlin, 2016-02-23:
Your log would look ok if the '* ' would not be interpreted as *italic* here. 
https://guides.github.com/features/mastering-markdown/ (look for Syntax highlighting)

thinkyhead, 2016-02-24:
@dgm3333 Did you solve the problem?

dgm3333, 2016-02-25:
I decided to use the problem as motivation to switch to Ramps, as I had one sitting around, so I closed the issue, as I couldn't see how to delete it, but thanks for asking.